chipweinberger/flutter_blue_plus #1057: [Help]: BluetoothAdapterState is always unknown in the first time (Android)

xOldeVx, 2024-11-17:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

v1.32.8

### Flutter Version

3.24.4

### What OS?

Android

### OS Version

14

### Bluetooth Module

Nordic nRF52

### What is your problem?

The first time always return `BluetoothAdapterState.unknown`, I tried the workaround that suggest in the read me (wait longer for the service to initialize), but it doesn't work, here's my code:

```
    BluetoothAdapterState adapterState = await FlutterBluePlus.adapterStateNow;
    if (FlutterBluePlus.adapterStateNow == BluetoothAdapterState.unknown) {
      await Future.delayed(const Duration(seconds: 1));
      adapterState = await FlutterBluePlus.adapterStateNow;
    }
    print(adapterState); // ---------> BluetoothAdapterState.unknown
```

What I missing?

### Logs

```shell
No logs
```

chipweinberger, 2024-11-17:
you need to call a function that initializes BLE, i.e. calls the platform.

`adapterState` will do it.` adapterStateNow` will not.

chipweinberger, 2024-11-17:
```
if (await FlutterBluePlus.adapterState.first == BluetoothAdapterState.unknown) {
    await Future.delayed(const Duration(seconds: 1));
}
```

this is better

closed 2024-11-17 by chipweinberger

xOldeVx, 2024-11-17:
Working! thanks!!