chipweinberger/flutter_blue_plus #1062: [Issue]: FlutterBluePlus requires android.permission.BLUETOOTH_SCAN permission

ekuleshov, 2024-11-27:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

1.34.5

### Flutter Version

3.24.5

### What OS?

Android

### OS Version

13

### Bluetooth Module

Moto G 5G (2022) phone

### What is your problem?

Getting `PlatformException(startScan, FlutterBluePlus requires android.permission.BLUETOOTH_SCAN permission, null, null)` reported from the user's Android devices and I can't reproduce it locally.

So far I've seen the following devices and Android versions in the crash reports:

* Android 14 - Pixel 6 Pro
* Android 14 - Galaxy S22
* Android 13 - Moto G 5G (2022)

With permissions being managed by the `flutter_blue_plus` plugin I haven't expected to see these errors.

The app is build with targetSdkVersion 34 and being installed from Google Play.
The following permissions are declared in the app manifest (as in the app release bundle):

```
<manifest
    xmlns:android="http://schemas.android.com/apk/res/android"
    android:compileSdkVersion="34"
    android:compileSdkVersionCodename="14"
    platformBuildVersionCode="34"
    platformBuildVersionName="14">
    <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION"/>
    <uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION"/>
    <uses-feature android:name="android.hardware.bluetooth" android:required="false"/>
    <uses-feature android:name="android.hardware.bluetooth_le" android:required="false"/>

    <uses-permission android:maxSdkVersion="30" android:name="android.permission.BLUETOOTH"/>
    <uses-permission android:maxSdkVersion="30" android:name="android.permission.BLUETOOTH_ADMIN"/>
    <uses-permission android:name="android.permission.BLUETOOTH_SCAN"/>
    <uses-permission android:name="android.permission.BLUETOOTH_CONNECT"/>
    ...
```

### Logs

```shell
Fatal Exception: io.flutter.plugins.firebase.crashlytics.FlutterError: PlatformException(startScan, FlutterBluePlus requires android.permission.BLUETOOTH_SCAN permission, null, null)
       at StandardMethodCodec.decodeEnvelope(message_codecs.dart:648)
       at MethodChannel._invokeMethod(platform_channel.dart:334)
       at FlutterBluePlus._invokeMethod(flutter_blue_plus.dart:630)
       at FlutterBluePlus.startScan(flutter_blue_plus.dart:300)
```

chipweinberger, 2024-11-27:
1. you should catch the errors and display a error message
2. they probably "denied" the permission request

ekuleshov, 2024-11-27:
@chipweinberger these errors are too generic and not something that could be shown to the user.

E.g. no error code or specific exception type that app could use to decide what error to show to the user. Especially when permissions are concerned.

chipweinberger, 2024-11-28:
You can string match and convert to any error message you want.

chipweinberger, 2024-11-28:
You can string match and convert to any error message you want.

If you want error codes, please open a PR.

ekuleshov, 2024-11-28:
> You can string match for "requires android.permission.BLUETOOTH_SCAN" and convert to any error message you want (and of course localize it to the language of your users).

Yes, but that is cumbersome and not a type safe approach. You can change message slightly and it will stop working. Not to mention potential differences between different platforms.

> If you want error codes, please open a PR.

Okay. Though it seems like it would require a lot of boilerplate and would be hard to keep things in sync between all supported platforms.

Have you considered using [pigeon](https://pub.dev/packages/pigeon)? It would allow to eliminate a lot of manually written boilerplate code and let you share common stuff, e.g. error codes enum as well as other things across Dart and platform-specific code.

> FWIW, I show all errors to the user. Better experience than crashing.

I can't show errors like `FlutterBluePlus requires android.permission.BLUETOOTH_SCAN permission` to the user. From their prospective:

* FlutterBluePlus has no meaning in context of a given app
* the BLUETOOTH_SCAN permission is not something a user could act upon or enable in the system app permissions (it is a hidden under "Nearby Devices", which I believe cover both CONNECT and SCAN)

At very least this error message need to be reworded. e.g. to something like "missing permissions required to scan for Bluetooth devices"

We really need some way to see if required permissions are granted before issuing various BLE operations, so if they aren't the app could show appropriate explanation in the app UI. E.g. something alike to [canLaunchUrl](https://pub.dev/documentation/url_launcher/latest/url_launcher/canLaunchUrl.html) in the [url_launcher](https://pub.dev/packages/url_launcher) plugin.

chipweinberger, 2024-11-29:
you can use the `permissions` package

ekuleshov, 2024-11-29:
> you can use the `permissions` package

Not really. We'd have to reimplement the same logic related to permissions and for every its API call that FBP has on every platform and every target OS version.

chipweinberger, 2024-11-30:
on iOS you can just check adapterState

on android too, actually

are you checking adapterState before you scan?

closed 2024-11-30 by chipweinberger

chipweinberger, 2024-11-30:
you can still open a PR for error codes if you want.

but in my app adapterState is all i need

ekuleshov, 2024-11-30:
Yes, I'm checking the state. I don't think scan and location permissions are checked at the time of the adapter check on Android. Only at the time BLE scan is started.

And the problem with these failures on scan and connect is that messages are not suitable for showing to the end users.

I would like to contribute a PR for error codes. Though it would make it much easier to maintain and also use outside of FBP code if something like pigeon is used to keep the common data model in sync between platforms. Are you not open to using it for FBP?

chipweinberger, 2024-11-30:
this is what my app shows 

[image]

chipweinberger, 2024-11-30:
> pidgeon

Dependencies are not allowed in this repo.

ekuleshov, 2024-11-30:
> > pidgeon
> 
> Dependencies are not allowed in this repo.
> 
> this lets users change their pidgeon version freely, without impacting FBP code.

`pigeon` is a `dev_dependency` and used only during FBP development as `dart run pigeon`.
That will generate code for all platforms that has no dependencies on `pigeon`

ekuleshov, 2024-11-30:
> this is what my app shows

Right. And that is not exactly accurate or self-explanatory error message, as permission could be called differently in the system or not visible at all.

chipweinberger, 2024-12-01:
> `pigeon` is a `dev_dependency` and used only during FBP development as `dart run pigeon`. That will generate code for all platforms that has no dependencies on `pigeon`

oh okay. good to know. Still, I don't want to change it.