chipweinberger/flutter_blue_plus #1066: [Issue]: flutter_blue_plus is not using TRANSPORT_LE when connecting from Android 5

ekuleshov, 2024-11-30:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

1.34.5

### Flutter Version

3.24.5

### What OS?

Android

### OS Version

5

### Bluetooth Module

Amazon Fire7 gen5 2015

### What is your problem?

The `flutter_blue_plus` plugin is using a default BT transport (LE + 2.0) when Android platform is below 23 (e.g. Android 5)

https://github.com/chipweinberger/flutter_blue_plus/blob/fe45ba098c76bacf31fe7ffb667d5b15352fd54f/android/src/main/java/com/lib/flutter_blue_plus/FlutterBluePlusPlugin.java#L718-L722

That often leads to 133 issues if peripheral device supports both 2.0 and BLE link.
I see a lot of that on some old Amazon Fire 7 devices running Android 5 and issue can't be worked around in the current version of the `flutter_blue_plus`.

To get around that pretty much all native Android BLE helper libraries are using reflection to call a hidden private method on BluetoothDevice class:

https://github.com/douglasjunior/AndroidBluetoothLibrary/blob/master/BluetoothLowEnergyLibrary/src/main/java/com/github/douglasjunior/bluetoothlowenergylibrary/BluetoothLeService.java#L317-L330

```
            // fall back to reflection for Android 5 to force TRANSPORT_LE
            try {
                Method method = device.getClass().getDeclaredMethod("connectGatt",
                        Context.class, Boolean.TYPE, BluetoothGattCallback.class, Integer.TYPE);
                method.setAccessible(true);
                btGatt = (BluetoothGatt) method.invoke(device, this, false, gattCallback, 2 /* BluetoothDevice.TRANSPORT_LE */);
            } catch (Exception ex) {
                btGatt = device.connectGatt(this, false, gattCallback);
            }
```

Here is a full patch that fixes these 133 connection issues in my case:

https://github.com/chipweinberger/flutter_blue_plus/pull/1065

### Logs

```shell
N/A
```

chipweinberger, 2024-11-30:
thanks for reporting. 

feel free to open a PR!

chipweinberger, 2024-11-30:
ah, thanks for the PR!

chipweinberger, 2024-11-30:
merged

closed 2024-11-30 by chipweinberger