chipweinberger/flutter_blue_plus #1070: [Help]:

maxzirilli, 2024-12-03:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

1.34.5

### Flutter Version

3.24.3

### What OS?

Android

### OS Version

8.1.0

### Bluetooth Module

Generic Beacon

### What is your problem?

I have a Bluetooth device that sends advertising data in the Eddystone format.
When I perform a scan, I can see the devices, but the manufacturerData array is empty.

What could be causing this issue? Could it be related to missing or incorrect permissions in the Manifest file?

Additional Information:

    I can see the advertisement packets using the BLE Scanner app.
    The advertisement packets are visible on other devices as well.

### Logs

```shell
There's no log for this issue
```

chipweinberger, 2024-12-03:
you need location permissions.

closed 2024-12-03 by chipweinberger

maxzirilli, 2024-12-06:
> you need location permissions.

I did it. Without location permissions I don't see these devices. Problem is I get device information except manufacturer data.

The procedure I did to allow location permission is the following:

1.  In Android Manifest (main) I inserted these rows:
     <uses-permission android:name="android.permission.BLUETOOTH_SCAN" />
     <uses-permission android:name="android.permission.BLUETOOTH_CONNECT" />
     <uses-permission android:name="android.permission.BLUETOOTH" />
     <uses-permission android:name="android.permission.BLUETOOTH_ADMIN" />
     <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />
     <uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" />
erasing the neverForLocation flag

2. In the code I ask the permission with next procedure:
Future<void> requestPermissions() async {
  // Richiedi il permesso di localizzazione (necessario per scansioni BLE)
  if (await Permission.location.isDenied) {
    await Permission.location.request();
  }

  // Richiedi il permesso di scansione Bluetooth (necessario su Android 12+)
  if (await Permission.bluetoothScan.isDenied) {
    await Permission.bluetoothScan.request();
  }

  // Richiedi il permesso di connessione Bluetooth (necessario su Android 12+)
  if (await Permission.bluetoothConnect.isDenied) {
    await Permission.bluetoothConnect.request();
  }

  // Richiedi il permesso di advertising Bluetooth (se usato su Android 12+)
  if (await Permission.bluetoothAdvertise.isDenied) {
    await Permission.bluetoothAdvertise.request();
  }
}

Could I have forgot something?