chipweinberger/flutter_blue_plus #1071: [Feature]: Add Swift Package Manager support

loic-sharma, 2024-12-03:
### FlutterBluePlus Version

1.34.5

### Flutter Version

3.24.0

### What OS?

iOS, macOS

### OS Version

N/A

### Bluetooth Module

N/A

### What is your feature request?

Hello,

Flutter is migrating to [Swift Package Manager][]. Please [add Swift Package Manager support][] to your plugin.

Flutter will eventually deprecate and then remove support for CocoaPods. Adding Swift Package Manager support future-proofs your plugins.

[Swift Package Manager]: https://www.swift.org/documentation/package-manager/
[add Swift Package Manager support]: https://docs.flutter.dev/packages-and-plugins/swift-package-manager/for-plugin-authors#how-to-add-swift-package-manager-support-to-an-existing-flutter-plugin

### How to add Swift Package Manager support to your plugins

You can find the migration guide here: https://docs.flutter.dev/packages-and-plugins/swift-package-manager/for-plugin-authors#how-to-add-swift-package-manager-support-to-an-existing-flutter-plugin

If you run into issues or have questions, please reach out to the Flutter team. You can ping me at @loic-sharma, send a message to the #hackers-ios channel on [Flutter’s discord][], or [open a GitHub issue][].

Thank you for your wonderful contributions to the Flutter ecosystem!

[Flutter’s discord]: https://github.com/flutter/flutter/blob/master/docs/contributing/Chat.md
[open a GitHub issue]: https://github.com/flutter/flutter/issues/new/choose

### Why is Flutter migrating to Swift Package Manager?

[CocoaPods is now in maintenance mode][].

Swift Package Manager support is one of Flutter’s most requested features: [flutter#33850][].

Flutter's Swift Package Manager integration has several benefits:

1. **Access to the Swift package ecosystem**. Flutter plugins can use the growing ecosystem of [Swift packages][].
2. **Simplifies Flutter installation**. Swift Package Manager is bundled with Xcode. In the future, you won’t need to install Ruby and CocoaPods to target iOS or macOS.

[CocoaPods is now in maintenance mode]: https://blog.cocoapods.org/CocoaPods-Support-Plans/
[flutter#33850]: https://github.com/flutter/flutter/issues/33850
[Swift packages]: https://swiftpackageindex.com/

### Logs

```shell
N/A
```

chipweinberger, 2024-12-04:
please open a PR thanks

chipweinberger, 2025-01-19:
merged: https://github.com/chipweinberger/flutter_blue_plus/pull/1107

closed 2025-01-19 by chipweinberger