chipweinberger/flutter_blue_plus #1072: [Help]: I have a Bluetooth device, and I use Android and iOS to get the Bluetooth address of getAddress as 00:15:83:D5:E8:B4, while using our flutter_blue_plus to get the Bluetooth address as C0:15:83:D5:E8:B4. What causes them to get different addresses

xiuzhenguo, 2024-12-04:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

1.34.5

### Flutter Version

3.19.3

### What OS?

All

### OS Version

iOS  17.7  android 11

### Bluetooth Module

Arduino HC-08

### What is your problem?

I have a Bluetooth device, and I use Android and iOS to get the Bluetooth address of getAddress as 00:15:83:D5:E8:B4, while using our flutter_blue_plus to get the Bluetooth address as C0:15:83:D5:E8:B4. What causes them to get different addresses

### Logs

```shell
The obtained Bluetooth list is fine, only the address is inconsistent with the one obtained natively
```

chipweinberger, 2024-12-04:
privacy reasons. its randomized.

closed 2024-12-04 by chipweinberger

xiuzhenguo, 2024-12-05:
its randomized.  However, the address obtained by the same device is always the same. The address obtained by flutter_blue_plus cannot connect to Bluetooth devices, while the address obtained by Java and iOS can connect to Bluetooth devices. Is there any additional processing done when obtaining the address? For example, 00 is changed to C0 after processing