chipweinberger/flutter_blue_plus #1073: [Help]:

elvysferrari, 2024-12-05:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.32.12

### Flutter Version

3.10.5

### What OS?

Android

### OS Version

Android 13

### Bluetooth Module

ESP32, libraries BLEServer, BLEClient, BLEUtils, BLEScan, BLEAddress

### What is your problem?

Hello, good afternoon,
I have a small problem. I am using

onValueReceived.listen((event) {});

to retrieve the return from an esp32.

For some reason, when I send the second command, the first one I sent is returned, plus the second one. The third time, the first one, the second one, and the third one are returned, even though I only sent a write.

I use the command: await cc.write(command, withoutResponse: false, allowLongWrite: true);

Could you please help me?

### Logs

```shell
No have logs
```

chipweinberger, 2024-12-06:
seems like a problem with your ESP32

elvysferrari, 2024-12-06:
I made some changes; my issue was in the controller. I use GetX and a singleton controller, and my communication was handled there. I solved it by placing it in a widget and destroying it.

closed 2024-12-06 by chipweinberger