chipweinberger/flutter_blue_plus #1074: [Help]: Manufacturer data

maxzirilli, 2024-12-09:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.34.5

### Flutter Version

3.24.3

### What OS?

Android

### OS Version

both 8.1.0 - 14

### Bluetooth Module

EMBC22

### What is your problem?

I have a Bluetooth device that sends advertising data in the Eddystone format.
When I perform a scan, I can see the devices, but the manufacturerData array is empty.

The packet are TLM Eddystone format. I need to get battery value but manufacturerData array is empty.

Problem isn't localization config. Here you can find manifest configurations.

I did it. Without location permissions I don't see these devices. Problem is I get device information except manufacturer data.

The procedure I did to allow location permission is the following:

    In Android Manifest (main) I inserted these rows:
    uses-permission android:name="android.permission.BLUETOOTH_SCAN"
    uses-permission android:name="android.permission.BLUETOOTH_CONNECT"
    uses-permission android:name="android.permission.BLUETOOTH"
    uses-permission android:name="android.permission.BLUETOOTH_ADMIN"
    uses-permission android:name="android.permission.ACCESS_FINE_LOCATION"
    uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION"
    erasing the neverForLocation flag

    In the code I ask the permission with next procedure:
    Future requestPermissions() async {
    // Richiedi il permesso di localizzazione (necessario per scansioni BLE)
    if (await Permission.location.isDenied) {
    await Permission.location.request();
    }

// Richiedi il permesso di scansione Bluetooth (necessario su Android 12+)
if (await Permission.bluetoothScan.isDenied) {
await Permission.bluetoothScan.request();
}

// Richiedi il permesso di connessione Bluetooth (necessario su Android 12+)
if (await Permission.bluetoothConnect.isDenied) {
await Permission.bluetoothConnect.request();
}

// Richiedi il permesso di advertising Bluetooth (se usato su Android 12+)
if (await Permission.bluetoothAdvertise.isDenied) {
await Permission.bluetoothAdvertise.request();
}
}

Could I have forgot something?

### Logs

```shell
No log
```

chipweinberger, 2024-12-09:
you need location permissions

closed 2024-12-09 by chipweinberger

maxzirilli, 2024-12-09:
> you need location permissions

no.
However I solved. Eddystone exposes a service. I thought it was a custom packet so instead to look at serviceData array I was looking on manufacturerData array. My bad