chipweinberger/flutter_blue_plus #1075: SCAN_FAILED_APPLICATION_REGISTRATION_FAILED

lwx123, 2024-12-10:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

flutter_blue_plus: ^1.34.5

### Flutter Version

Flutter 3.22.3

### What OS?

Android

### OS Version

android 14

### Bluetooth Module

esp32

### What is your problem?

`_scanResultsSubscription = FlutterBluePlus.scanResults.listen((results) {
      results = results.where((r) => r.device.platformName.isNotEmpty).toList();
      _scanResults = results;
      if (mounted) {
        setState(() {});
      }
    }, onError: (e) {
      Snackbar.show(ABC.b, prettyException("Scan Error:", e), success: false);
    });`

### Logs

```shell
When monitoring scanning events, my application typically functions smoothly during the scanning process while running. However, occasionally, I encounter the following error message in the debug console:

[ OnScanResponse ] result: {error_string: SCAN_FAILED_APPLICATION_REGISTRATION_FAILED, advertisements: [], success: 0, error_code: 2}
BluetoothLeScanner(): onScannerRegistered() - status=0 scannerId=7 mScannerId=-1

can you help me?
```

chipweinberger, 2024-12-10:
From ChatGPT

----

**Short Answer:**  
**SCAN_FAILED_APPLICATION_REGISTRATION_FAILED** typically occurs when the Android Bluetooth subsystem is unable to register the application’s BLE scan callback due to internal resource limits or conflicts. In other words, the operating system has failed to add your app’s scanner to the system’s list of active BLE scan clients, often because too many simultaneous scans are ongoing or because of internal Bluetooth stack inconsistencies.

**Detailed Explanation:**  
This error code is associated with the Android Bluetooth Low Energy (BLE) scanning APIs introduced in Android Lollipop (API level 21) and onwards. It’s most often encountered when calling methods like `BluetoothLeScanner.startScan()`.

When you initiate a BLE scan, the Android system’s Bluetooth stack attempts to register your callback as a BLE "client." A number of conditions can cause the registration to fail:

1. **Too Many Concurrent Scans:**  
   The Android BLE stack imposes limits on how many scans can run simultaneously. If multiple apps are scanning at the same time (including system apps or those running in the background), the system may refuse to register another scanning client, triggering this error.

2. **Resource Limitations:**  
   Some devices may have tighter constraints on hardware or firmware resources needed for BLE operations. If these resources are exhausted—for instance, the device’s BLE controller or system buffers are full—the registration may fail.

3. **Improper Cleanup from Previous Scans:**  
   If a previous BLE scan was not stopped properly or if the Bluetooth stack has not fully released resources from a prior operation, attempts to start a new scan may fail. This can occur due to programming oversights (e.g., not calling `stopScan()` before initiating a new scan) or due to bugs in the device’s firmware.

4. **Issues with the Bluetooth Adapter State:**  
   If the Bluetooth adapter is in a transient or error state—e.g., it’s turning off or on, or the system is resetting the Bluetooth subsystem—you might encounter this error. Ensuring that Bluetooth is fully enabled and stable before scanning can help.

**Mitigation Strategies:**

- **Reduce the Number of Simultaneous Scans:**  
  Only run a BLE scan when necessary, and always call `stopScan()` when you’re done. Avoid overlapping scan periods.

- **Retry with Backoff:**  
  If you encounter this error, consider waiting a short period before trying again. Sometimes the system needs a moment to free up resources.

- **Check Bluetooth State Before Scanning:**  
  Make sure Bluetooth is turned on and stable. You can use `BluetoothAdapter.isEnabled()` to ensure that Bluetooth is available before starting a scan.

- **Update Device Firmware or Test on Different Hardware:**  
  If the error is persistent on certain devices, it might be caused by device-specific issues or firmware bugs. Updating the device software or testing on different models can help determine if the problem is device-specific.

**In Summary:**  
`SCAN_FAILED_APPLICATION_REGISTRATION_FAILED` usually signifies a resource or registration problem within the Android BLE stack. Proper management of scanning sessions, resource checks, and handling transient Bluetooth states often helps resolve or prevent this error.

lwx123, 2024-12-11:
Run the example in flutter_blue_plus project, the first startup scan is normal, and the subsequent refresh has been a problem
[image]

closed 2024-12-16 by chipweinberger