chipweinberger/flutter_blue_plus #1078: [Help]: With large stream of data, some packets lost

thaidmfinnick, 2024-12-13:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.34.5

### Flutter Version

3.24.4

### What OS?

iOS

### OS Version

iOS 17.7.2

### Bluetooth Module

Nordic nrf52840

### What is your problem?

I use full example app, add only some lines of code below to see bytes sending from my hardware. I have sent counter at the end of this packet to make sure I cannot lost any packet. But when I see a log, many packets lost. This code I add to listen value:
```dart
// I add in characteristic_tile.dart
@override
  void initState() {
    super.initState();
    _lastValueSubscription = widget.characteristic.lastValueStream.listen((value) {
      _value = value;
      if (mounted) {
        setState(() {});
      }
    });
    final subscription = widget.characteristic.onValueReceived.listen((value) {
      print('okeee:$value');
    });
  }
```
You can see in logs, the last bytes, jump from 0 to 8, instead of 0, 1, 2, 3.

I have tried other lib like flutter_reactive_ble, it counts right.

Do I miss something in configuration? Thank you!

### Logs

```shell
flutter: okeee:[192, 0, 0, 255, 234, 55, 255, 253, 104, 253, 88, 54, 255, 178, 154, 0]
flutter: okeee:[192, 0, 0, 255, 245, 130, 255, 253, 27, 253, 112, 76, 255, 177, 125, 8]
flutter: okeee:[192, 0, 0, 255, 246, 203, 255, 253, 62, 253, 108, 38, 255, 178, 60, 9]
flutter: okeee:[192, 0, 0, 255, 250, 17, 255, 253, 66, 253, 90, 44, 255, 178, 211, 10]
flutter: okeee:[192, 0, 0, 255, 254, 28, 255, 253, 27, 253, 71, 13, 255, 179, 168, 11]
flutter: okeee:[192, 0, 0, 255, 255, 155, 255, 253, 84, 253, 57, 5, 255, 180, 24, 12]
flutter: okeee:[192, 0, 0, 255, 253, 247, 255, 253, 91, 253, 55, 78, 255, 179, 5, 13]
flutter: okeee:[192, 0, 0, 255, 250, 77, 255, 253, 91, 253, 57, 4, 255, 177, 219, 14]
flutter: okeee:[192, 0, 0, 255, 229, 241, 255, 253, 132, 253, 57, 12, 255, 177, 186, 23]
flutter: okeee:[192, 0, 0, 255, 237, 52, 255, 253, 84, 253, 58, 194, 255, 176, 94, 24]
flutter: okeee:[192, 0, 0, 255, 239, 45, 255, 252, 232, 253, 69, 52, 255, 175, 53, 25]
flutter: okeee:[192, 0, 0, 255, 233, 194, 255, 252, 212, 253, 84, 13, 255, 175, 142, 26]
flutter: okeee:[192, 0, 0, 255, 234, 173, 255, 253, 9, 253, 101, 22, 255, 175, 178, 27]
flutter: okeee:[192, 0, 0, 255, 238, 47, 255, 253, 30, 253, 113, 51, 255, 175, 186, 28]
flutter: okeee:[192, 0, 0, 255, 240, 229, 255, 253, 31, 253, 109, 132, 255, 176, 92, 29]
flutter: okeee:[192, 0, 0, 255, 254, 210, 255, 253, 40, 253, 113, 94, 255, 175, 151, 38]
flutter: okeee:[192, 0, 0, 255, 253, 152, 255, 253, 82, 253, 109, 217, 255, 175, 243, 39]
flutter: okeee:[192, 0, 0, 255, 245, 249, 255, 253, 59, 253, 92, 92, 255, 176, 131, 40]
flutter: okeee:[192, 0, 0, 255, 242, 163, 255, 253, 89, 253, 73, 133, 255, 177, 33, 41]
flutter: okeee:[192, 0, 0, 255, 246, 29, 255, 253, 130, 253, 59, 52, 255, 177, 162, 42]
flutter: okeee:[192, 0, 0, 255, 246, 228, 255, 253, 121, 253, 57, 92, 255, 176, 164, 43]
flutter: okeee:[192, 0, 0, 255, 247, 227, 255, 253, 96, 253, 59, 0, 255, 175, 100, 44]
flutter: okeee:[192, 0, 0, 255, 240, 228, 255, 253, 82, 253, 57, 54, 255, 175, 95, 53]
flutter: okeee:[192, 0, 0, 255, 235, 74, 255, 253, 77, 253, 58, 44, 255, 174, 60, 54]
flutter: okeee:[192, 0, 0, 255, 235, 105, 255, 253, 9, 253, 67, 185, 255, 172, 88, 55]
flutter: okeee:[192, 0, 0, 255, 238, 56, 255, 252, 240, 253, 82, 70, 255, 171, 12, 56]
flutter: okeee:[192, 0, 0, 255, 240, 96, 255, 253, 25, 253, 99, 97, 255, 170, 231, 57]
flutter: okeee:[192, 0, 0, 255, 244, 41, 255, 253, 22, 253, 111, 120, 255, 171, 126, 58]
flutter: okeee:[192, 0, 0, 255, 250, 96, 255, 253, 79, 253, 108, 133, 255, 172, 55, 59]
flutter: okeee:[192, 0, 0, 255, 229, 60, 255, 253, 156, 253, 53, 126, 255, 175, 138, 83]
flutter: okeee:[192, 0, 0, 255, 236, 194, 255, 253, 113, 253, 55, 29, 255, 174, 220, 84]
flutter: okeee:[192, 0, 0, 255, 237, 97, 255, 253, 18, 253, 65, 149, 255, 173, 61, 85]
flutter: okeee:[192, 0, 0, 255, 243, 13, 255, 252, 224, 253, 79, 210, 255, 172, 38, 86]
```

chipweinberger, 2024-12-13:
works find for me. no lost packets.

make sure your device uses BLE **indications** instead of **notifications**

closed 2024-12-13 by chipweinberger

thaidmfinnick, 2024-12-13:
Thank you for your response!

I only wonder with same peripheral device, same Flutter code base, and using notifications, why Android receive full packets, otherwise iOS lost periodically after 10 packets (or more), I have tested with flutter_reactive_ble, same results as flutter_blue_plus!

So maybe problems come from OS or my hardware devices? Can you give/recommend to me some ideas to check where problems come from?
(I'm only mobile developers, but I will send message to firmware people)

chipweinberger, 2024-12-14:
make sure your peripheral device is sending BLE **indications** instead of **notifications**