chipweinberger/flutter_blue_plus #1084: [Help]:

zxydream, 2024-12-19:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

1.33.5

### Flutter Version

3.3.10

### What OS?

Android

### OS Version

android 14

### Bluetooth Module

android

### What is your problem?

If I want to know all the information about Bluetooth connection status changes on the Android platform, not just the two states of successful connection and disconnection, how should I add or modify the source code to be more convenient?

[image]

### Logs

```shell
null
```

chipweinberger, 2024-12-19:
see example

closed 2024-12-19 by chipweinberger

zxydream, 2024-12-19:
Got it, thanks