chipweinberger/flutter_blue_plus #1088: [Help]: 180F or 18f0 for iOS

solomon-okolie, 2024-12-25:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.34.5

### Flutter Version

3.27.0

### What OS?

iOS

### OS Version

18.1.1

### Bluetooth Module

MP210

### What is your problem?

The docs mentions passing
````
List<Guid> withServices = [Guid("180F")];
````
for iOS but this did not work for me. passing Guid("18f0") did work.

### Logs

```shell
N/A
```

chipweinberger, 2024-12-25:
0x180F is the battery service

closed 2024-12-26 by chipweinberger

solomon-okolie, 2024-12-27:
@chipweinberger is there a place one can documentation of these numbers and what service they present?

solomon-okolie, 2024-12-27:
No worries, I found https://www.bluetooth.com/wp-content/uploads/Files/Specification/Assigned_Numbers.html