chipweinberger/flutter_blue_plus #1089: [Feature]: why not support mice?use bluetooth 5.0

zjw78123sc, 2024-12-26:
### FlutterBluePlus Version

  flutter_blue_plus: ^1.34.5   flutter_blue_plus_windows: ^1.25.0

### Flutter Version

3.27.1

### What OS?

All

### OS Version

win10

### Bluetooth Module

donnot know

### What is your feature request?

I want to find the connected mice, which use bluetooth 2.4g and 5.0, your document say that not support Bluetooth classic, but when i connect use 5.0,it alse cannect find my connected mice

### Logs

```shell
no log,  
List<BluetoothDevice> devs = await FlutterBluePlus.systemDevices(withServices);
List<BluetoothDevice> devs = FlutterBluePlus.connectedDevices;
```

chipweinberger, 2024-12-26:
ask Tim Cook.

It's not possible.

closed 2024-12-26 by chipweinberger

zjw78123sc, 2024-12-27:
hello，what is your suggestion,if I want to developer a windows software to scan the mice,and communicate with the mice through the bluetooth protocol.