chipweinberger/flutter_blue_plus #1093: [Help]: "The remoteId keeps changing."

mstfyldrmm, 2025-01-08:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.34.5

### Flutter Version

 3.27.1,

### What OS?

Android

### OS Version

Android 11

### Bluetooth Module

I don't know

### What is your problem?

"Every time I turn Bluetooth on and off and perform a scan, the remoteId parameter of the device keeps changing. In other words, the remoteId parameter of Android and iOS smartphones changes continuously. Is this due to security reasons?" "I debugged with multiple devices having different Android versions, and I encounter the same issue every time."

### Logs

```shell
my code 
import 'dart:async';
import 'package:flutter/cupertino.dart';
import 'package:flutter/foundation.dart';
import 'package:flutter/material.dart';
import 'package:flutter_blue_plus/flutter_blue_plus.dart';

class ScanPage extends StatefulWidget {
  const ScanPage({super.key});

  @override
  State<ScanPage> createState() => _ScanPageState();
}

class _ScanPageState extends State<ScanPage> {
  late bool _isScanning;
  late bool _isConnecting;
  late List<ScanResult> _scanResults;
  late BluetoothCharacteristic _selectedCharacteristic;

  @override
  void initState() {
    super.initState();
    _isScanning = false;
    _isConnecting = false;
    _scanResults = [];

    _scanDevice();
  }

  @override
  void dispose() {
    _scanResults.clear();
    super.dispose();
  }

  Future _scanDevice() async {
    _scanResults.clear();

    FlutterBluePlus.scanResults.listen(
      (results) {
        if (mounted) setState(() => _scanResults = results);
      },
      onError: (e) {
        if (kDebugMode) {
          print('Error scan result subscription: $e');
        }
      },
    );

    FlutterBluePlus.isScanning.listen((state) {
      if (mounted) setState(() => _isScanning = state);
    });

    try {
      FlutterBluePlus.startScan(
        withNames: [
          'advertisedName'
        ], // filter by advertised names (exact match)
        withRemoteIds: [
          'deviceID'
        ], // filter for known remoteIds (iOS: 128-bit guid, android: 48-bit mac address)
        withServices: [Guid('serviceID')], // filter by advertised services
        removeIfGone: const Duration(seconds: 5),
        androidUsesFineLocation: false,
        continuousUpdates: true,
      );
    } catch (e) {
      if (kDebugMode) {
        print('Error scan device method: $e');
      }
    }
  }

  @override
  Widget build(BuildContext context) {
    return Scaffold(
        floatingActionButton: FloatingActionButton(onPressed: () async {
          await _scanDevice();
          print('Scan result:${_scanResults}');
        }),
        body: ListView.builder(
            itemCount: _scanResults.length,
            itemBuilder: (BuildContext context, int index) {
              return ListTile(
                title: Text('RemoteID: ${_scanResults[index].device.id}'),
                subtitle: Text(_scanResults[index].device.platformName),
                trailing: Text('${_scanResults[index].rssi}'),
              );
            }));
  }
}
```

chipweinberger, 2025-01-10:
security

closed 2025-01-10 by chipweinberger

mstfyldrmm, 2025-01-10:
> security

"There is no solution for it to remain constant then."

mstfyldrmm, 2025-01-10:
"As far as I understand, the remoteId of the devices will constantly change."