chipweinberger/flutter_blue_plus #1094: [Help]: Missing iBeacon minor and major in advertismentData

xxxOVALxxx, 2025-01-09:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.34.5

### Flutter Version

3.24.5

### What OS?

Android

### OS Version

Android 13

### Bluetooth Module

idk

### What is your problem?

The bytes minor and major are missing in advertismentData.
In nRF Connect minor and major are present in advertismentData
[image]
[image]
From example app:
[image]

### Logs

```shell
ScanResult{device: BluetoothDevice{remoteId: 1C:34:F1:06:D7:8D, platformName: , services: null}, advertisementData: AdvertisementData{advName: , txPowerLevel: null, appearance: null, connectable: true, manufacturerData: {65535: [84, 83, 49, 51, 1, 0, 41, 91, 204, 35, 12, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]}, serviceData: {}, serviceUuids: []}, rssi: -65, timeStamp: 2025-01-09 12:25:01.723502}
```

chipweinberger, 2025-01-22:
Those two screenshots show completely different MSD bytes. What data do you expect, verse what do you see?

chipweinberger, 2025-02-17:
closing. not enough info.

closed 2025-02-17 by chipweinberger