chipweinberger/flutter_blue_plus #1096: [Help]:

akoskuczi-bw, 2025-01-10:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

1.32.12

### Flutter Version

3.24.2

### What OS?

Android

### OS Version

14

### Bluetooth Module

unknown

### What is your problem?

Hello,

I have an issue with On Motorola Edge 50 Neo Android 14 (build: U4UIS34.8-22-7-3)
Starting the scan with the withNames parameter returns no BLE devices.
e.g.,
FlutterBluePlus.startScan(
      continuousUpdates: true,
      androidScanMode: AndroidScanMode(2),
      withNames: _listOfNames,
    );

Without the withNames parameter it finds the BLE beacon device. 

### Logs

```shell
From logcat I see that line:

"Ignoring scan response received without advertising event"
```

chipweinberger, 2025-01-22:
logs needed. reopen with logs

akoskuczi-bw, 2025-02-28:
Hi,

please find the logs below, when scan started with withNames parameter:

2025-02-28 23:15:02.303  3173-3300  bluetooth               com.android.bluetooth                I  le_scanning_reassembler.cc:59 ProcessAdvertisingReport: Ignoring scan response received without advertising event
2025-02-28 23:15:05.543  2841-7257  NearbyMediums           com.google.android.gms.persistent    I  No BLE Fast/GATT advertisements found in the latest cycle.
2025-02-28 23:15:07.861  3173-3300  bluetooth               com.android.bluetooth                I  le_scanning_reassembler.cc:59 ProcessAdvertisingReport: Ignoring scan response received without advertising event
2025-02-28 23:15:09.329  3173-3300  bluetooth               com.android.bluetooth                I  le_scanning_reassembler.cc:59 ProcessAdvertisingReport: Ignoring scan response received without advertising event
2025-02-28 23:15:14.895  3173-3300  bluetooth               com.android.bluetooth                I  le_scanning_reassembler.cc:59 ProcessAdvertisingReport: Ignoring scan response received without advertising event
2025-02-28 23:15:15.547  2841-6105  NearbyMediums           com.google.android.gms.persistent    I  No BLE Fast/GATT advertisements found in the latest cycle.
2025-02-28 23:15:21.937  3173-3300  bluetooth               com.android.bluetooth                I  le_scanning_reassembler.cc:59 ProcessAdvertisingReport: Ignoring scan response received without advertising event
2025-02-28 23:15:25.550  2841-7257  NearbyMediums           com.google.android.gms.persistent    I  No BLE Fast/GATT advertisements found in the latest cycle.
2025-02-28 23:15:34.440  3173-3300  bluetooth               com.android.bluetooth                I  le_scanning_reassembler.cc:59 ProcessAdvertisingReport: Ignoring scan response received without advertising event
2025-02-28 23:15:35.554  2841-6105  NearbyMediums           com.google.android.gms.persistent    I  No BLE Fast/GATT advertisements found in the latest cycle.
2025-02-28 23:15:40.028  3173-3300  bluetooth               com.android.bluetooth                I  le_scanning_reassembler.cc:59 ProcessAdvertisingReport: Ignoring scan response received without advertising event
2025-02-28 23:15:41.475  3173-3300  bluetooth               com.android.bluetooth                I  le_scanning_reassembler.cc:59 ProcessAdvertisingReport: Ignoring scan response received without advertising event
2025-02-28 23:15:45.557  2841-7257  NearbyMediums           com.google.android.gms.persistent    I  No BLE Fast/GATT advertisements found in the latest cycle.
----------
Started the scan without the withNames parameter, and with the withRemoteIds parameter, where withRemoteIds contains the MAC addresses of the BLE beacons. The logs look like this:

2025-02-28 23:41:39.839  3173-3289  bt_btif_dm              com.android.bluetooth                I  btif_dm.cc:537 get_cod: remote_cod=0x00001f00
2025-02-28 23:41:39.839  3173-3380  BtGatt.ScanHelper       com.android.bluetooth                I  Skipping client for location deny list
2025-02-28 23:41:39.840  3173-3380  BtGatt.ScanHelper       com.android.bluetooth                I  Skipping client for location deny list
2025-02-28 23:41:39.841  3173-3380  BtGatt.ScanHelper       com.android.bluetooth                I  Skipping client for location deny list
2025-02-28 23:41:46.871  3173-3289  bt_btif_dm              com.android.bluetooth                I  btif_dm.cc:537 get_cod: remote_cod=0x00001f00
2025-02-28 23:41:46.872  3173-3380  BtGatt.ScanHelper       com.android.bluetooth                I  Skipping client for location deny list
2025-02-28 23:41:46.873  3173-3380  BtGatt.ScanHelper       com.android.bluetooth                I  Skipping client for location deny list
2025-02-28 23:41:46.875  3173-3380  BtGatt.ScanHelper       com.android.bluetooth                I  Skipping client for location deny list

akoskuczi-bw, 2025-02-28:
One more thing: if I use withKeywords instead of withNames, the beacons are found.

chipweinberger, 2025-03-01:
those are not FBP logs

akoskuczi-bw, 2025-03-01:
Started with 
    await FlutterBluePlus.startScan(
      continuousUpdates: true,
      androidScanMode: AndroidScanMode(2),
      androidUsesFineLocation: true,
      withNames: _beaconNames,
    );
No result:

2025-03-01 11:34:21.625 26389-26389 [FBP-Android]           io.bitwitch.dev.seekretbox           D  [FBP] onAttachedToEngine
2025-03-01 11:34:21.627 26389-26389 [FBP-Android]           io.bitwitch.dev.seekretbox           D  [FBP] onAttachedToActivity
2025-03-01 11:34:24.344 26389-26389 [FBP-Android]           io.bitwitch.dev.seekretbox           D  [FBP] onAttachedToEngine
2025-03-01 11:34:25.285 26389-26389 [FBP-Android]           io.bitwitch.dev.seekretbox           D  [FBP] onMethodCall: flutterRestart
2025-03-01 11:34:25.285 26389-26389 [FBP-Android]           io.bitwitch.dev.seekretbox           D  [FBP] initializing BluetoothAdapter
2025-03-01 11:34:25.288 26389-26389 [FBP-Android]           io.bitwitch.dev.seekretbox           D  [FBP] disconnectAllDevices(flutterRestart)
2025-03-01 11:34:25.288 26389-26389 [FBP-Android]           io.bitwitch.dev.seekretbox           D  [FBP] connectedPeripherals: 0
2025-03-01 11:34:25.334 26389-26541 flutter                 io.bitwitch.dev.seekretbox           I  [FBP] <setLogLevel> args: 5
2025-03-01 11:34:25.334 26389-26389 [FBP-Android]           io.bitwitch.dev.seekretbox           D  [FBP] onMethodCall: setLogLevel
2025-03-01 11:34:25.341 26389-26541 flutter                 io.bitwitch.dev.seekretbox           I  [FBP] (setLogLevel) result: true
2025-03-01 11:34:25.348 26389-26541 flutter                 io.bitwitch.dev.seekretbox           I  [FBP] <getAdapterState> args: null
2025-03-01 11:34:25.368 26389-26389 [FBP-Android]           io.bitwitch.dev.seekretbox           D  [FBP] onMethodCall: getAdapterState
2025-03-01 11:34:25.383 26389-26541 flutter                 io.bitwitch.dev.seekretbox           I  [FBP] (getAdapterState) result: {adapter_state: 4}
2025-03-01 11:34:26.700 26389-26541 flutter                 io.bitwitch.dev.seekretbox           I  [FBP] <startScan> args: {with_services: [], with_remote_ids: [], with_names: [ps3kfmdito3j_001, snkjhfz5r3bc_001], with_keywords: [], with_msd: [], with_service_data: [], continuous_updates: true, continuous_divisor: 1, android_legacy: false, android_scan_mode: 2, android_uses_fine_location: true}
2025-03-01 11:34:26.700 26389-26389 [FBP-Android]           io.bitwitch.dev.seekretbox           D  [FBP] onMethodCall: startScan
2025-03-01 11:34:26.708 26389-26541 flutter                 io.bitwitch.dev.seekretbox           I  [FBP] (startScan) result: true

----------------

Started with:
    await FlutterBluePlus.startScan(
      continuousUpdates: true,
      androidScanMode: AndroidScanMode(2),
      androidUsesFineLocation: true,
      withNames: _beaconNames,
      withRemoteIds: ["BC:57:11:00:32:A7"]
    );

Finds the beacon:

2025-03-01 11:32:59.926 25599-25599 [FBP-Android]           io.bitwitch.dev.seekretbox           D  [FBP] onAttachedToEngine
2025-03-01 11:32:59.927 25599-25599 [FBP-Android]           io.bitwitch.dev.seekretbox           D  [FBP] onAttachedToActivity
2025-03-01 11:33:02.549 25599-25599 [FBP-Android]           io.bitwitch.dev.seekretbox           D  [FBP] onAttachedToEngine
2025-03-01 11:33:03.626 25599-25599 [FBP-Android]           io.bitwitch.dev.seekretbox           D  [FBP] onMethodCall: flutterRestart
2025-03-01 11:33:03.626 25599-25599 [FBP-Android]           io.bitwitch.dev.seekretbox           D  [FBP] initializing BluetoothAdapter
2025-03-01 11:33:03.630 25599-25599 [FBP-Android]           io.bitwitch.dev.seekretbox           D  [FBP] disconnectAllDevices(flutterRestart)
2025-03-01 11:33:03.630 25599-25599 [FBP-Android]           io.bitwitch.dev.seekretbox           D  [FBP] connectedPeripherals: 0
2025-03-01 11:33:03.671 25599-25924 flutter                 io.bitwitch.dev.seekretbox           I  [FBP] <setLogLevel> args: 5
2025-03-01 11:33:03.672 25599-25599 [FBP-Android]           io.bitwitch.dev.seekretbox           D  [FBP] onMethodCall: setLogLevel
2025-03-01 11:33:03.680 25599-25924 flutter                 io.bitwitch.dev.seekretbox           I  [FBP] (setLogLevel) result: true
2025-03-01 11:33:03.691 25599-25924 flutter                 io.bitwitch.dev.seekretbox           I  [FBP] <getAdapterState> args: null
2025-03-01 11:33:03.701 25599-25599 [FBP-Android]           io.bitwitch.dev.seekretbox           D  [FBP] onMethodCall: getAdapterState
2025-03-01 11:33:03.746 25599-25924 flutter                 io.bitwitch.dev.seekretbox           I  [FBP] (getAdapterState) result: {adapter_state: 4}
2025-03-01 11:33:05.112 25599-25924 flutter                 io.bitwitch.dev.seekretbox           I  [FBP] <startScan> args: {with_services: [], with_remote_ids: [BC:57:11:00:32:A7], with_names: [ps3kfmdito3j_001, snkjhfz5r3bc_001], with_keywords: [], with_msd: [], with_service_data: [], continuous_updates: true, continuous_divisor: 1, android_legacy: false, android_scan_mode: 2, android_uses_fine_location: true}
2025-03-01 11:33:05.112 25599-25599 [FBP-Android]           io.bitwitch.dev.seekretbox           D  [FBP] onMethodCall: startScan
2025-03-01 11:33:05.337 25599-25924 flutter                 io.bitwitch.dev.seekretbox           I  [FBP] (startScan) result: true
2025-03-01 11:33:11.036 25599-25599 [FBP-Android]           io.bitwitch.dev.seekretbox           D  [FBP] onScanResult
[... cut ...]
2025-03-01 11:33:16.043 25599-25599 [FBP-Android]           io.bitwitch.dev.seekretbox           D  [FBP] onScanResult
2025-03-01 11:33:16.066 25599-25924 flutter                 io.bitwitch.dev.seekretbox           I  [FBP] [[ OnScanResponse ]] result: {advertisements: [{rssi: -96, connectable: 1, manufacturer_data: {76: 02157777772e6b6b6d636e2e636f6d00000c000525fab1}, remote_id: BC:57:11:00:32:A7, platform_name: snkjhfz5r3bc_001, service_data: {2080: 5504000525fa}, adv_name: snkjhfz5r3bc_001}]}

chipweinberger, 2025-03-01:
thanks. I don't have time to look into it right now.

but feel free to open a PR.