chipweinberger/flutter_blue_plus #1100: [Help]: readCharacteristic always fail

tree1891, 2025-01-16:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

1.35.0

### Flutter Version

3.27.2

### What OS?

Android

### OS Version

Android 10

### Bluetooth Module

Nordic

### What is your problem?

readCharacteristic always fail, but not in 1.34.5

### Logs

```shell
D/[FBP-Android](16804): [FBP] onMethodCall: flutterRestart
D/[FBP-Android](16804): [FBP] initializing BluetoothAdapter
D/[FBP-Android](16804): [FBP] disconnectAllDevices(flutterRestart)
D/[FBP-Android](16804): [FBP] connectedPeripherals: 0
D/[FBP-Android](16804): [FBP] onMethodCall: isSupported
D/[FBP-Android](16804): [FBP] onMethodCall: getAdapterState
I/flutter (16804): [FBP] stopScan: already stopped
D/[FBP-Android](16804): [FBP] onMethodCall: startScan
D/BluetoothAdapter(16804): isLeEnabled(): ON
D/BluetoothLeScanner(16804): onScannerRegistered() - status=0 scannerId=7 mScannerId=0
D/[FBP-Android](16804): [FBP] onMethodCall: stopScan
D/BluetoothAdapter(16804): isLeEnabled(): ON
D/[FBP-Android](16804): [FBP] onMethodCall: connect
D/BluetoothGatt(16804): connect() - device: D8:3B:DA:4D:B1:CA, auto: false
D/BluetoothGatt(16804): registerApp()
D/BluetoothGatt(16804): registerApp() - UUID=a46690e8-094b-4d0e-80ec-9b6af66cf9f3
D/BluetoothGatt(16804): onClientRegistered() - status=0 clientIf=7
D/BluetoothGatt(16804): onClientConnectionState() - status=0 clientIf=7 device=D8:3B:DA:4D:B1:CA
D/[FBP-Android](16804): [FBP] onConnectionStateChange:connected
D/[FBP-Android](16804): [FBP]   status: SUCCESS
D/[FBP-Android](16804): [FBP] onMethodCall: requestMtu
D/BluetoothGatt(16804): configureMTU() - device: D8:3B:DA:4D:B1:CA mtu: 512
D/BluetoothGatt(16804): onConnectionUpdated() - Device=D8:3B:DA:4D:B1:CA interval=36 latency=3 timeout=500 status=0
D/BluetoothGatt(16804): onConnectionUpdated() - Device=D8:3B:DA:4D:B1:CA interval=6 latency=0 timeout=500 status=0
D/BluetoothGatt(16804): onConfigureMTU() - Device=D8:3B:DA:4D:B1:CA mtu=256 status=0
D/[FBP-Android](16804): [FBP] onMtuChanged:
D/[FBP-Android](16804): [FBP]   mtu: 256
D/[FBP-Android](16804): [FBP]   status: GATT_SUCCESS (0)
D/[FBP-Android](16804): [FBP] onMethodCall: discoverServices
D/BluetoothGatt(16804): discoverServices() - device: D8:3B:DA:4D:B1:CA
D/BluetoothGatt(16804): onSearchComplete() = Device=D8:3B:DA:4D:B1:CA Status=0
D/[FBP-Android](16804): [FBP] onServicesDiscovered:
D/[FBP-Android](16804): [FBP]   count: 6
D/[FBP-Android](16804): [FBP]   status: 0GATT_SUCCESS
D/BluetoothGatt(16804): onConnectionUpdated() - Device=D8:3B:DA:4D:B1:CA interval=36 latency=3 timeout=500 status=0
D/[FBP-Android](16804): [FBP] onMethodCall: setNotifyValue
D/BluetoothGatt(16804): setCharacteristicNotification() - uuid: 00002a05-0000-1000-8000-00805f9b34fb enable: true
D/[FBP-Android](16804): [FBP] onMethodCall: readCharacteristic
[LinkPage] Exception in connectDevice
D/[FBP-Android](16804): [FBP] onMethodCall: disconnect
D/BluetoothGatt(16804): cancelOpen() - device: D8:3B:DA:4D:B1:CA
D/BluetoothGatt(16804): onClientConnectionState() - status=0 clientIf=7 device=D8:3B:DA:4D:B1:CA
D/[FBP-Android](16804): [FBP] onConnectionStateChange:disconnected
D/[FBP-Android](16804): [FBP]   status: SUCCESS
D/BluetoothGatt(16804): close()
D/BluetoothGatt(16804): unregisterApp() - mClientIf=7
           PlatformException(readCharacteristic, gatt.readCharacteristic() returned false, null, null)
           #0      StandardMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:646:7)
           #1      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:334:18)
           <asynchronous suspension>
           #2      FlutterBluePlusAndroid._invokeMethod (package:flutter_blue_plus_android/flutter_blue_plus_android.dart:400:20)
           <asynchronous suspension>
           #3      FlutterBluePlusAndroid.readCharacteristic (package:flutter_blue_plus_android/flutter_blue_plus_android.dart:232:5)
           <asynchronous suspension>
           #4      FlutterBluePlus._invokeMethod (package:flutter_blue_plus/src/flutter_blue_plus.dart:568:14)
           <asynchronous suspension>
           #5      BluetoothCharacteristic.read (package:flutter_blue_plus/src/bluetooth_characteristic.dart:124:7)
           <asynchronous suspension>
           #6      _LinkPageState.connectDevice (package:geolink/pages/link.dart:904:15)
           <asynchronous suspension>
```

tree1891, 2025-01-16:
No, it is disconnected after exception occurs.

tnc1997, 2025-01-16:
> No, it is disconnected after exception occurs.

Ah, I see, the logs are not ordered in chronological order, so I was reading the stack trace after the disconnection.

tree1891, 2025-01-16:
```
    // in order to match iOS behavior on all platforms,
    // we always listen to the Services Changed characteristic if it exists.
```
It seems to readCharacteristic before onDescriptorWrite finish for setNotifyValue.

tnc1997, 2025-01-16:
Does the same exception get thrown if `subscribeToServicesChanged` is `false` when calling `discoverServices`?

tree1891, 2025-01-16:
no problem if `subscribeToServicesChanged` is `false`

tnc1997, 2025-01-16:
Thank you very much for confirming that, it sounds like there may be a synchronisation issue.

tnc1997, 2025-01-16:
This issue may be related to #1102 whereby the previous operation has not completed before the next one starts.

tnc1997, 2025-01-16:
@tree1891  would I be able to get you to confirm if you still experience this issue with the [fix/1102](https://github.com/chipweinberger/flutter_blue_plus/tree/fix/1102) branch?

```yaml
dependency_overrides:
  flutter_blue_plus_android:
    git:
      url: https://github.com/chipweinberger/flutter_blue_plus.git
      ref: fix/1102
      path: packages/flutter_blue_plus_android
```

tree1891, 2025-01-16:
It does work well.

tnc1997, 2025-01-16:
> It does work well.

Thank you for confirming that change fixes this issue, we will publish some patched versions to pub.dev.

chipweinberger, 2025-01-22:
update to 1.35.1

closed 2025-01-22 by chipweinberger