chipweinberger/flutter_blue_plus #1101: [Help]:

freemansoft, 2025-01-16:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

1.34.5

### Flutter Version

3.22.4

### What OS?

Android

### OS Version

n/a

### Bluetooth Module

n/a

### What is your problem?

The example won't compile 

- Flutter: 3.22.4
- Android Studio: 2024.2.1
- Java: 21
- OS: Windows 11

If I delete the `android` directory and recreate it with `flutter create  --template=app --platforms android example` then I get a version of the example that builds and runs without issues.  Regenerating the example has the advantage of upgrading the Android example app migrates from `build.gradle` to `build.gradle.kts`. I can change the `sourceCompatability` and `targetCompatability` and `jvmTarget` from version from the generated `11` to `1.8` and the example app still compiles and builds.

I can open a PR with the changes or we can just say my environment is weird.

### Logs

```shell
Launching lib\main.dart on SM S156V in debug mode...

FAILURE: Build failed with an exception.

* Where:
Script 'c:\dev\flutter\packages\flutter_tools\gradle\app_plugin_loader.gradle' line: 9

* What went wrong:
A problem occurred evaluating script.
> You are applying Flutter's app_plugin_loader Gradle plugin imperatively using the apply script method, which is not possible anymore. Migrate to applying Gradle plugins with the declarative plugins block: https://flutter.dev/to/flutter-gradle-plugin-apply

* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.

* Get more help at https://help.gradle.org
```

MrCsabaToth, 2025-01-19:
I don't see 3.22.4 Flutter in the releases https://docs.flutter.dev/release/archive
3.22.3 was released last July on the stable channel and then 3.24.0 in August.

ekuleshov-idexx, 2025-01-19:
@freemansoft could you please post the output of your `flutter doctor -v`

Though I may agree that we need to regenerate the sample app for Android.

ekuleshov, 2025-01-19:
@freemansoft could you please post the output of your `flutter doctor -v`

Though I may agree that we need to regenerate the sample app for Android.

freemansoft, 2025-01-20:
> I don't see 3.22.4 Flutter in the releases https://docs.flutter.dev/release/archive 3.22.3 was released last July on the stable channel and then 3.24.0 in August.

- I thought I was on `3.24.2` and swapped the two characters. The example complies with some deprecation warnings under `3.24.2`, 
- I am actually on the latest from main `3.28.0`.   Compiling the Android example fails with the messages above. 

flutter doctor with 3.28.0
```
PS C:\dev\flutter> flutter doctor -v
[✓] Flutter (Channel main, 3.28.0-2.0.pre.38786, on Microsoft Windows [Version 10.0.26100.2894], locale en-US) [1,331ms]
    • Flutter version 3.28.0-2.0.pre.38786 on channel main at c:\dev\flutter
    • Upstream repository https://github.com/flutter/flutter.git
    • Framework revision 63110570da (4 hours ago), 2025-01-20 12:23:19 +1300
    • Engine revision 63110570da
    • Dart version 3.8.0 (build 3.8.0-1.0.dev)
    • DevTools version 2.42.0

[✓] Windows Version (11 Pro 64-bit, 24H2, 2009) [1,816ms]

[✓] Android toolchain - develop for Android devices (Android SDK version 34.0.0) [1,444ms]
    • Android SDK at C:\Users\joe\AppData\Local\Android\Sdk
    • Platform android-35, build-tools 34.0.0
    • ANDROID_HOME = C:\Users\joe\AppData\Local\Android\Sdk
    • Java binary at: C:\Program Files\Android\Android Studio\jbr\bin\java
      This is the JDK bundled with the latest Android Studio installation on this machine.
      To manually set the JDK path, use: `flutter config --jdk-dir="path/to/jdk"`.
    • Java version OpenJDK Runtime Environment (build 21.0.3+-12282718-b509.11)
    • All Android licenses accepted.

[✓] Chrome - develop for the web [204ms]
    • Chrome at C:\Program Files\Google\Chrome\Application\chrome.exe

[✗] Visual Studio - develop Windows apps [203ms]
    ✗ Visual Studio not installed; this is necessary to develop Windows apps.
      Download at https://visualstudio.microsoft.com/downloads/.
      Please install the "Desktop development with C++" workload, including all of its default components

[✓] Android Studio (version 2024.2) [18ms]
    • Android Studio at C:\Program Files\Android\Android Studio
    • Flutter plugin can be installed from:
      🔨 https://plugins.jetbrains.com/plugin/9212-flutter
    • Dart plugin can be installed from:
      🔨 https://plugins.jetbrains.com/plugin/6351-dart
    • Java version OpenJDK Runtime Environment (build 21.0.3+-12282718-b509.11)

[✓] VS Code (version 1.96.4) [16ms]
    • VS Code at C:\Users\joe\AppData\Local\Programs\Microsoft VS Code
    • Flutter extension version 3.102.0

[✓] Connected device (3 available) [137ms]
    • Windows (desktop) • windows • windows-x64    • Microsoft Windows [Version 10.0.26100.2894]
    • Chrome (web)      • chrome  • web-javascript • Google Chrome 131.0.6778.265
    • Edge (web)        • edge    • web-javascript • Microsoft Edge 131.0.2903.112

[✓] Network resources [138ms]
    • All expected network resources are available.

```

The deprecation warnings under 3.24.2

```
Try `flutter pub outdated` for more information.

You are applying Flutter's app_plugin_loader Gradle plugin imperatively using the apply script method, which is deprecated and will be removed in a future release. Migrate to applying Gradle plugins with the declarative plugins block: https://flutter.dev/to/flutter-gradle-plugin-apply

You are applying Flutter's main Gradle plugin imperatively using the apply script method, which is deprecated and will be removed in a future release. Migrate to applying Gradle plugins with the declarative plugins block: https://flutter.dev/to/flutter-gradle-plugin-apply

Warning: unable to detect project KGP version. Skipping version checking.
This may be because you have applied KGP after the Flutter Gradle Plugin.
Font asset "MaterialIcons-Regular.otf" was tree-shaken, reducing it from 1645184 to 1748 bytes (99.9% reduction). Tree-shaking can be disabled by providing the --no-tree-shake-icons flag when building your app.
warning: [options] source value 8 is obsolete and will be removed in a future release
warning: [options] target value 8 is obsolete and will be removed in a future release
warning: [options] To suppress warnings about obsolete options, use -Xlint:-options.
3 warnings
warning: [options] source value 8 is obsolete and will be removed in a future release
warning: [options] target value 8 is obsolete and will be removed in a future release
warning: [options] To suppress warnings about obsolete options, use -Xlint:-options.
3 warnings
Running Gradle task 'bundleRelease'...                             51.4s
√ Built build\app\outputs\bundle\release\app-release.aab (20.4MB)
```

ekuleshov, 2025-01-20:
@freemansoft these deprecation warnings are due FBP using Java 8 source/target and the recent android gradle plugin seem to be deprecating support for Java 8.

```
warning: [options] source value 8 is obsolete and will be removed in a future release
warning: [options] target value 8 is obsolete and will be removed in a future release
```

@chipweinberger @tnc1997 do we need to support older android gradle plugin versions?
Those may not support Java 17 or 21 as used in the recent android gradle plugin.

chipweinberger, 2025-01-20:
FBP aims to be backwards compatible with older android phones

And aims to be compatible with Flutter versions from the last ~2 years

ekuleshov, 2025-01-20:
> FBP aims to be backwards compatible with older android phones. Same minSdk as flutter.
> And aims to be compatible with Flutter versions from the last ~2 years

Compatibility with Android target is not an issue.
But consumers will have to bump the android gradle plugin in their app build for Android.

freemansoft, 2025-01-20:
https://developer.android.com/google/play/requirements/target-sdk

> Starting August 31 2024:
>
> New apps and app updates must target Android 14 (API level 34) or higher to be submitted to Google Play; except for Wear OS and Android TV apps, which must target Android 13 (API level 33) or higher.
>
> Existing apps must target Android 13 (API level 33) or higher to remain available to new users on devices running Android OS higher than your app's target API level. Apps that target Android 12 (API level 31) or lower (Android 10 (API level 29) or lower for Wear OS and Android 11 (API level 30) or lower for Android TV), will only be available on devices running Android OS that are the same or lower than your app's target API level.

ekuleshov, 2025-01-20:
> Am on a corporate project where we just updated our Android app to be Java 17 and are moving to gradle 8. This was forced by the store policy and by some 3rd party libraries Flutter libraries that we wanted to use that required 17.

@freemansoft same here. Though I can still use AGP 7 and older gradle to publish Android API 34 targeted app.
I always thought that gradle and AGP versions are independent from the target platform for the app.

freemansoft, 2025-01-20:
Yes, we are still using the older gradle with Flutter 3.24.2. We just ignore the deprecation warnings like the "java 8" warnings.  The example compiles fine ignoring the warnings.

I thought that Google is driving people to use AGP 8 and the later gradle as shown by my failure to compile the example with Flutter 3.28.  

This isn't my area of expertise so I reserve the right to be wrong!

chipweinberger, 2025-02-17:
closing. i dont think we need to change anything.