chipweinberger/flutter_blue_plus #1104: [Help]: Facing issue in Web while using bluetooth write function.

Narayanhegde14, 2025-01-17:
### Requirements

- [x] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.35.0

### Flutter Version

3.27.2

### What OS?

All

### OS Version

macOS 15.2

### Bluetooth Module

Expressif Esp32

### What is your problem?

adapterState is not supported for Flutter web for this package but when calling write function from the package its giving UnimplementedError for getAdapterState please fix this or let me know if I'm doing something wrong.

### Logs

```shell
DartError: UnimplementedError
dart-sdk/lib/_internal/js_dev_runtime/private/ddc_runtime/errors.dart 288:3                    throw_
packages/flutter_blue_plus_platform_interface/flutter_blue_plus_platform_interface.dart 135:5  getAdapterState
packages/flutter_blue_plus/src/flutter_blue_plus.dart 127:79                                   <fn>
packages/flutter_blue_plus/src/flutter_blue_plus.dart 568:26                                   <fn>
dart-sdk/lib/_internal/js_dev_runtime/patch/async_patch.dart 610:19                            <fn>
dart-sdk/lib/_internal/js_dev_runtime/patch/async_patch.dart 634:23                            <fn>
dart-sdk/lib/_internal/js_dev_runtime/patch/async_patch.dart 581:31                            <fn>
dart-sdk/lib/async/zone.dart 1676:54                                                           runUnary
dart-sdk/lib/async/future_impl.dart 204:18                                                     handleValue
dart-sdk/lib/async/future_impl.dart 902:44                                                     handleValueCallback
dart-sdk/lib/async/future_impl.dart 931:13                                                     _propagateToListeners
dart-sdk/lib/async/future_impl.dart 707:5                                                      [_completeWithValue]
dart-sdk/lib/async/future_impl.dart 777:7                                                      callback
dart-sdk/lib/async/schedule_microtask.dart 40:11                                               _microtaskLoop
dart-sdk/lib/async/schedule_microtask.dart 49:5                                                _startMicrotaskLoop
dart-sdk/lib/_internal/js_dev_runtime/patch/async_patch.dart 186:7                             <fn>
```

chipweinberger, 2025-01-17:
@tnc1997 just an idea, maybe have `adapterState` always return 'on'? thats my first thought, so users can have one code path

tnc1997, 2025-01-17:
> maybe have adapterState always return 'on'?

The issue with that is that the adapter state is technically "unknown" as it is not supported on the Web platform currently. It would perhaps be better to have an "unknown" state so that we could implement the correct state if/when it becomes supported on the Web platform.

chipweinberger, 2025-01-17:
that technicality seems unimportant.

users can detect web and do whatever they want,

but at least their existing code will work without modification, which is the whole idea of flutter.

tnc1997, 2025-01-17:
> that technicality seems unimportant.

That is perfectly reasonable and of course it is ultimately your decision if that is your preferred approach.

chipweinberger, 2025-01-17:
as long as there are no obvious downsides, yes, it seems better for users.

ekuleshov, 2025-01-18:
@tnc1997 @chipweinberger there are a few more platform API methods that are needed:

* getAdapterState
* onAdapterStateChanged (an empty stream may be enough for this and others below?)
* onBondStateChanged
* onMtuChanged
* onNameChanged
* onReadRssi
* onServicesReset
* getAdapterName (perhaps better to return something that fail on web?)
* getSystemDevices (an empty list? currently the example app is failing on this call)

ekuleshov, 2025-01-18:
@tnc1997 while you are on this, could you please also make all calls to `Platform.is*` guarded with `kIsWeb` (and maybe also with `kIsWasm`)?

There is one unguarded call in `flutter_blue_plus.dart` and some more in the example app.

chipweinberger, 2025-01-18:
@ekuleshov, are you interested in being a maintainer?

chipweinberger, 2025-01-19:
and yes, I agree. It's better to return sensible default then to require everyone to specially handle these methods on web.

ekuleshov, 2025-01-19:
> [@ekuleshov](https://github.com/ekuleshov), are you interested in being a maintainer?
> you've been following FBP for awhile.

Probably not. Can't claim enough expertise for all target platforms and also have other commitments...

Though I will submit PRs when I can.

> I see your comments in other BLE libs too sometimes, do you use multiple?

Unfortunately yes. The FBP doesn't support peripherals and there are a few other issues.

chipweinberger, 2025-01-22:
update to 1.35.1

closed 2025-01-22 by chipweinberger