chipweinberger/flutter_blue_plus #1105: [Feature]: Support writing to multiple devices at the same time

chipweinberger, 2025-01-17:
### FlutterBluePlus Version

1.35.0

### Flutter Version

3.27.2

### What OS?

All

### OS Version

iOS 18.2

### Bluetooth Module

esp32-s2

### What is your feature request?

Currently, we use a global mutex to do a single operation at a time in FBP. i.e. `_Mutex mtx = _MutexFactory.getMutexForKey("global")`

It would make sense for all functions to switch to a device mutex. i.e. `_Mutex mtx = _MutexFactory.getMutexForKey(remoteId)`

Or potentially a command queue with an explicit list of pending operations.

**Current Code:**
```
  /// Writes a characteristic.
  Future<void> write(List<int> value,
      {bool withoutResponse = false, bool allowLongWrite = false, int timeout = 15}) async {
    //  check args
    if (withoutResponse && allowLongWrite) {
      throw ArgumentError("cannot longWrite withoutResponse, not allowed on iOS or Android");
    }

    // check connected
    if (device.isDisconnected) {
      throw FlutterBluePlusException(
          ErrorPlatform.fbp, "writeCharacteristic", FbpErrorCode.deviceIsDisconnected.index, "device is not connected");
    }

    // Only allow a single ble operation to be underway at a time
    _Mutex mtx = _MutexFactory.getMutexForKey("global");
    await mtx.take();

    try {
      final writeType = withoutResponse ? BmWriteType.withoutResponse : BmWriteType.withResponse;

      var request = BmWriteCharacteristicRequest(
        remoteId: remoteId,
        characteristicUuid: characteristicUuid,
        serviceUuid: serviceUuid,
        writeType: writeType,
        allowLongWrite: allowLongWrite,
        value: value,
        primaryServiceUuid: primaryServiceUuid,
      );

      var responseStream = FlutterBluePlusPlatform.instance.onCharacteristicWritten
          .where((p) => p.remoteId == request.remoteId)
          .where((p) => p.serviceUuid == request.serviceUuid)
          .where((p) => p.characteristicUuid == request.characteristicUuid)
          .where((p) => p.primaryServiceUuid == request.primaryServiceUuid);

      // Start listening now, before invokeMethod, to ensure we don't miss the response
      Future<BmCharacteristicData> futureResponse = responseStream.first;

      // invoke
      await FlutterBluePlus._invokeMethod(() => FlutterBluePlusPlatform.instance.writeCharacteristic(request));

      // wait for response so that we can:
      //  1. check for success (writeWithResponse)
      //  2. wait until the packet has been sent, to prevent iOS & Android dropping packets (writeWithoutResponse)
      BmCharacteristicData response = await futureResponse
          .fbpEnsureAdapterIsOn("writeCharacteristic")
          .fbpEnsureDeviceIsConnected(device, "writeCharacteristic")
          .fbpTimeout(timeout, "writeCharacteristic");

      // failed?
      if (!response.success) {
        throw FlutterBluePlusException(_nativeError, "writeCharacteristic", response.errorCode, response.errorString);
      }

      return Future.value();
    } finally {
      mtx.give();
    }
  }
```

### Logs

```shell
na
```

shingohu, 2025-05-09:
Is there any progress

chipweinberger, 2025-05-09:
I wont personally work on this, but someone else can