chipweinberger/flutter_blue_plus #1108: FBP web SecurityError in requestDevice()

mgeilich, 2025-01-19:
### Requirements

- [x] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

1.34.5

### Flutter Version

3.27.0

### What OS?

macOS

### OS Version

MacOS 15.2

### Bluetooth Module

custom

### What is your problem?

With FBP web, when I try to get services for a device, it fails with this:

DartError: FlutterBluePlusException | discoverServices | web-code: 0 | SecurityError: Origin is not allowed to access any service. Tip: Add the service UUID to 'optionalServices' in requestDevice() options. https://goo.gl/HxfxSQ

I am running with flutter run -d chrome

I have tried to specify optionalServices in flutter_blue_plus_web.dart but either I don't have the syntax right or it isn't the right solution.  Can you help?
      options = RequestDeviceOptions(
            filters: filters.toJS,
            optionalServices:
                js.JsArray.from([Guid.fromString('00006459-0000-1000-8000-00805F9B34FB'), Guid.fromString('6459')])
                    as JSArray<BluetoothServiceUUID>);

### Logs

```shell
HA: BluetoothDevice{remoteId: nJB1Hqfeohd6OwQ05FlxUg==, platformName: CHA-E0040258, services: null} is connected
DartError: FlutterBluePlusException | discoverServices | web-code: 0 | SecurityError: Origin is not allowed to access any service. Tip: Add the service UUID to 'optionalServices' in requestDevice() options. https://goo.gl/HxfxSQ
dart-sdk/lib/_internal/js_dev_runtime/private/ddc_runtime/errors.dart 288:3  throw_
errors.dart:288
packages/flutter_blue_plus/src/bluetooth_device.dart 296:9                   <fn>
bluetooth_device.dart:296
dart-sdk/lib/_internal/js_dev_runtime/patch/async_patch.dart 610:19          <fn>
async_patch.dart:610
dart-sdk/lib/_internal/js_dev_runtime/patch/async_patch.dart 634:23          <fn>
async_patch.dart:634
dart-sdk/lib/_internal/js_dev_runtime/patch/async_patch.dart 581:31          <fn>
async_patch.dart:581
dart-sdk/lib/async/zone.dart 1676:54                                         runUnary
zone.dart:1676
dart-sdk/lib/async/future_impl.dart 204:18                                   handleValue
future_impl.dart:204
dart-sdk/lib/async/future_impl.dart 902:44                                   handleValueCallback
future_impl.dart:902
dart-sdk/lib/async/future_impl.dart 931:13                                   _propagateToListeners
future_impl.dart:931
dart-sdk/lib/async/future_impl.dart 707:5                                    [_completeWithValue]
future_impl.dart:707
dart-sdk/lib/async/future_impl.dart 1007:16                                  <fn>
future_impl.dart:1007
dart-sdk/lib/async/zone.dart 1676:54                                         runUnary
zone.dart:1676
dart-sdk/lib/async/future_impl.dart 204:18                                   handleValue
future_impl.dart:204
dart-sdk/lib/async/future_impl.dart 902:44                                   handleValueCallback
future_impl.dart:902
dart-sdk/lib/async/future_impl.dart 931:13                                   _propagateToListeners
future_impl.dart:931
dart-sdk/lib/async/future_impl.dart 707:5                                    [_completeWithValue]
future_impl.dart:707
dart-sdk/lib/async/future_impl.dart 777:7                                    callback
future_impl.dart:777
dart-sdk/lib/async/schedule_microtask.dart 40:11                             _microtaskLoop
schedule_microtask.dart:40
dart-sdk/lib/async/schedule_microtask.dart 49:5                              _startMicrotaskLoop
schedule_microtask.dart:49
dart-sdk/lib/_internal/js_dev_runtime/patch/async_patch.dart 186:7           <fn>
async_patch.dart:186
Done scanning for headsets
```

ekuleshov, 2025-01-19:
@mgeilich first of all you need to use `flutter_blue_plup: ^1.35.0` in your `pubspec.yaml`.
The `flutter_blue_plus_web` is an endorsed package and it will be automatically included.

The error you seeing "Origin is not allowed to access any service. Tip: Add the service UUID to 'optionalServices' in requestDevice() options" is somewhat misleading in the context of Flutter/Dart. It is the web bluetooth error, which is basically telling you that you have to specify service UUIDs you want to connect to when doing the device scan.

So, you have to use the `FlutterBluePlus.startScan(..., withServices: services)` and specify your `services` there.

mgeilich, 2025-01-20:
@ekuleshov Thank you!

chipweinberger, 2025-01-22:
we throw a dart exception if `withServices` is empty

mgeilich, 2025-02-07:
Adding withServices: to the scan() does not appear to work, it doesn't return anything.  There is no requestDevice() for setting optionalServices in the API.  requestDevice() is specific to flutter_web_bluetooth().

   await FlutterBluePlus.startScan(withServices: [
      Guid('1800'),
      Guid('1801'),
      Guid('2A05')
    ], timeout: Duration(seconds: timeout));

ekuleshov, 2025-02-23:
> Adding withServices: to the scan() does not appear to work, it doesn't return anything. There is no requestDevice() for setting optionalServices in the API. requestDevice() is specific to [...]

@mgeilich these services also has to be in the advertisement data. If BLE device does not advertise them, "withServices" won't work and you have to use recently added parameter "webOptionalServices" in the `startScan()`. See parameter's doc for the references to the web APIs.

ekuleshov, 2025-02-26:
@mgeilich what error do you get? Are you running on the last release version of FBP? Try with the latest from master.

I pushed a change to ignore some odd errors in the `discoverServices()` call on web and I don't think this change has been released yet. https://github.com/chipweinberger/flutter_blue_plus/commit/307accaa8f50de67b4530ad010eb451783e0e6a8

mgeilich, 2025-02-26:
@ekuleshov I am seeing this error on  flutter_blue_plus: ^1.35.3:
FlutterBluePlusException | discoverServices | web-code: 0 | NotFoundError: No Descriptors found in Characteristic.

ekuleshov, 2025-02-26:
Try the FBP from the master

mgeilich, 2025-02-26:
Same error using:
  flutter_blue_plus:
    git:
      url: https://github.com/chipweinberger/flutter_blue_plus.git
      ref: master

chipweinberger, 2025-04-03:
and updates? lets fix this

ekuleshov, 2025-04-03:
I haven't had a chance to look at it again. Will try to find time later this week.

mgeilich, 2025-04-11:
Any progress on this?  I am looking forward to retesting.

ekuleshov, 2025-04-11:
> Any progress on this? I am looking forward to retesting.

I looked at it and using the `master` branch my own app works out of the box with this change to enable connection to the custom service:

```
await FlutterBluePlus.startScan(
    timeout: Duration(seconds: timeout),
    webOptionalServices: [Guid(serviceUUID)]); // <--- added Nordic uart service here
```

The result of the scan is used to connect to a device and then discover services on device. Pretty much the same how the example app does it.

You should not need to use `FlutterWebBluetooth.instance.requestDevice()` API.

Also note, the `master` branch has mismatched versions of the platform plugins. So, you may need to add explicit `dependency_overrides` for all of them.

This is my override for a local relative path copy, similar to the example app:

```
dependency_overrides:
  flutter_blue_plus_platform_interface:
    path: ../flutter_blue_plus/packages/flutter_blue_plus_platform_interface
  flutter_blue_plus:
    path: ../flutter_blue_plus/packages/flutter_blue_plus
  flutter_blue_plus_android:
    path: ../flutter_blue_plus/packages/flutter_blue_plus_android
  flutter_blue_plus_darwin:
    path: ../flutter_blue_plus/packages/flutter_blue_plus_darwin
  flutter_blue_plus_web:
    path: ../flutter_blue_plus/packages/flutter_blue_plus_web
```

mgeilich, 2025-04-12:
@ekuleshov -
This latest version is working for we as well.
Thanks!

closed 2025-04-12 by mgeilich