chipweinberger/flutter_blue_plus #1110: [Help]: upgraded flutter_blue_plus to 1.35.0 from 1.34.5 now im getting UnimplementedError

DarkTemplar03, 2025-01-19:
### Requirements

- [x] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.35.0

### Flutter Version

3.27.2

### What OS?

macOS, iOS

### OS Version

macOS 14.7.1, Xcode 16.1, iOS 17.6.1

### Bluetooth Module

Arduino HC-08

### What is your problem?

I upgraded flutter_blue_plus to 1.35.0 from 1.34.5 (using flutter pub upgrade) now everytime i try to run my app it throws an UnimplementedError from whatever the first FlutterBluePlus function is called originally that was setLogLevel() however after i commented that out isBluetoothSupported() threw the same error.

[image]

[image]

[image]

when i looked into the example app i noticed it was running flutter_blue_plus 1.35.1 however it was running successfully, also whenever i rolled back to 1.34.5 my app is running successfully.

### Logs

```shell
UnimplementedError (UnimplementedError)
```

tnc1997, 2025-01-21:
Hi @DarkTemplar03, thank you very much for opening this issue.

Bonding is only supported on Android and Linux, and will throw an `UnimplementedError` on other platforms.

https://github.com/chipweinberger/flutter_blue_plus#compatibility

In the next version of `flutter_blue_plus` a [sensible default value](https://github.com/chipweinberger/flutter_blue_plus/issues/1104#issuecomment-2600327153) will be returned instead of throwing an error.

> when i looked into the example app i noticed it was running flutter_blue_plus 1.35.1 however it was running successfully

The example app should run successfully because it [checks the platform](https://github.com/chipweinberger/flutter_blue_plus/blob/a201aad40b48a6e7fbb034b7fa6a3bde76845b8d/packages/flutter_blue_plus/example/lib/screens/bluetooth_off_screen.dart#L61C15-L61C77) before calling platform specific methods.

chipweinberger, 2025-01-22:
update to 1.35.1

closed 2025-01-22 by chipweinberger