chipweinberger/flutter_blue_plus #1112: [Help]: error PlatformException(setNotifyValue, gatt.writeDescriptor() returned false, nu I/flutter (24657): ║ ll, null)

Johnjson, 2025-01-20:
### Requirements

- [x] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

^1.34.5

### Flutter Version

3.19.5

### What OS?

Android

### OS Version

macOS 14.5

### Bluetooth Module

error PlatformException(setNotifyValue, gatt.writeDescriptor() returned false, nu I/flutter (24657): ║ ll, null)

### What is your problem?

error PlatformException(setNotifyValue, gatt.writeDescriptor() returned false, nu
I/flutter (24657): ║ ll, null)

### Logs

```shell
error PlatformException(setNotifyValue, gatt.writeDescriptor() returned false, nu
I/flutter (24657): ║ ll, null)
```

Johnjson, 2025-01-20:
Exception writeCommand :  PlatformException(writeCharacteristic, gatt.writeCharacteristic() returned false, null, null)
I/flutter (13763): [FBP] <writeCharacteristic> args: {remote_id: 30:30:F9:6F:2F:46, service_uuid: 55535343-fe7d-4ae5-8fa9-9fafd205e455, characteristic_uuid: 16962447-c623-61ba-d94b-4d1e43535349, write_type: 1, allow_long_write: 0, value: 5aa50125000000001000000004000000}
D/[FBP-Android](13763): [FBP] onMethodCall: writeCharacteristic

Johnjson, 2025-01-20:
Return to 1.34.4 to resolve the problem

MrCsabaToth, 2025-01-26:
That's not a resolution