chipweinberger/flutter_blue_plus #1113: [Help]: After I've connected to multiple BLE devices, if one of them takes a really long time to connect or if it times out later, the others just have to wait around for it.

z1737029714, 2025-01-22:
### Requirements

- [x] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.35.0

### Flutter Version

3.27.1

### What OS?

All

### OS Version

ios17

### Bluetooth Module

unknow

### What is your problem?

as soon as I find a device, I try to connect to it right away.   And sometimes I've got a bunch of devices around me, but if one of them is way further away, it takes longer to connect.   And while that's happening, the other BLE devices just sit there waiting.   It's like they can only move on after that one device either connection completed  or connected fails.   Is this some kind of blockage, or do I need something to make their connections run independently without affecting each other?

### Logs

```shell
unknow
```

z1737029714, 2025-01-22:
I want to make sure that if any device fails to connect or times out, it shouldn't affect the connection speed of the other BLE devices. Is there a solution for that?

chipweinberger, 2025-01-22:
there is an existing issue for this

https://github.com/chipweinberger/flutter_blue_plus/issues/1105

closed 2025-01-22 by chipweinberger