chipweinberger/flutter_blue_plus #1114: [Help]: Error discovering services: PlatformException(iosException, Client Characteristic Configuration descriptors

hezhipengzipp, 2025-01-23:
### Requirements

- [x] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

1.35.2

### Flutter Version

3.24.5

### What OS?

iOS

### OS Version

18.1.1

### Bluetooth Module

discovering services:

### What is your problem?

OtaUpgradeManager Error discovering services: PlatformException(iosException, Client Characteristic Configuration descriptors must be configured using setNotifyValue:forCharacteristic:, {stackTrace: 0   CoreFoundation                      0x000000018e5d47d8 1532D3D8-9B3B-3F2F-B35F-55A20DDF411B + 534488<…>
flutter: \^[[38;5;196m│ ⛔ 1   libobjc.A.dylib                     0x000000018b8a72e4 objc_exception_throw + 88<…>
flutter: \^[[38;5;196m│ ⛔ 2   Foundation                          0x000000018d9a48d8 6D0212CC-3B9E-32C9-BE20-72989CE3ACB8 + 8452312<…>
flutter: \^[[38;5;196m│ ⛔ 3   CoreBluetooth                       0x00000001b6aa3f78 630933C1-A870-38CB-9760-F519FD0752AD + 307064<…>
flutter: \^[[38;5;196m│ ⛔ 4   flutter_blue_plus_darwin            0x00000001099b1230 -[FlutterBluePlusPlugin handleMethodCall:result:] + 20100<…>
flutter: \^[[38;5;196m│ ⛔ 5   Flutter                             0x000000010f43ecc4 __45-[FlutterMethodChannel setMethodCallHandler:]_block_invoke + 164<…>
flutter: \^[[38;5;196m│ ⛔ 6   Flutter                             0x000000010eec0d88 ___ZN7flutter25PlatformMessageHandlerIos21HandlePlatformMessageENSt3_fl10unique_ptrINS_15PlatformMessageENS1_14default_deleteIS3_EEEE_block_invoke + 116<…>
flutter: \^[[38;5;196m│ ⛔ 7   libdispatch.dylib                   0x000000010c180a30 _dispatch_call_block_and_release + 32<…>
flutter: \^[[38;5;196m│ ⛔ 8   libdispatch.dylib                   0x000000010c18271c _dispatch_client_callout + 20<…>
flutter: \^[[38;5;196m│ ⛔ 9   libdispatch.dylib                   0x000000010c192de8 _dispatch_main_queue_drain + 1076<…>
flutter: \^[[38;5;196m│ ⛔ 10  libdispatch.dylib                   0x000000010c1929a4 _dispatch_main_queue_callback_4CF + 44<…>
flutter: \^[[38;5;196m│ ⛔ 11  CoreFoundation                      0x000000018e5a8204 1532D3D8-9B3B-3F2F-B35F-55A20DDF411B + 352772<…>
flutter: \^[[38;5;196m│ ⛔ 12  CoreFoundation                      0x000000018e5a5440 1532D3D8-9B3B-3F2F-B35F-55A20DDF411B + 341056<…>
flutter: \^[[38;5;196m│ ⛔ 13  CoreFoundation                      0x000000018e5a4830 CFRunLoopRunSpecific + 588<…>
flutter: \^[[38;5;196m│ ⛔ 14  GraphicsServices                    0x00000001da5841c4 GSEventRunModal + 164<…>
flutter: \^[[38;5;196m│ ⛔ 15  UIKitCore                           0x000000019110aeb0 575E5140-FA6A-37C2-B00B-A4EACEDFDA53 + 4009648<…>
flutter: \^[[38;5;196m│ ⛔ 16  UIKitCore                           0x00000001911b95b4 UIApplicationMain + 340<…>
flutter: \^[[38;5;196m│ ⛔ 17  UIKitCore                           0x00000001914f3fa8 575E5140-FA6A-37C2-B00B-A4EACEDFDA53 + 8109992<…>
flutter: \^[[38;5;196m│ ⛔ 18  Runner                              0x0000000104fe8cd4 $sSo21UIApplicationDelegateP5UIKitE4mainyyFZ + 120<…>
flutter: \^[[38;5;196m│ ⛔ 19  Runner                              0x0000000104fe8c4c $s6Runner11AppDelegateC5$mainyyFZ + 44<…>
flutter: \^[[38;5;196m│ ⛔ 20  Runner                              0x0000000104fe8d50 main + 28<…>
flutter: \^[[38;5;196m│ ⛔ 21  dyld                                0x00000001b3f92ec8 3060D36A-16CE-3C3A-9258-3881459F5714 + 212680}, null)<

### Logs

```shell
flutter: [FBP] \^[[1;30m<setNotifyValue>\^[[0m args: \^[[1;35m{remote_id: 9C27C3BF-5C87-1C9E-5858-A44B44F763D9, service_uuid: ff12, characteristic_uuid: ff14, force_indications: false, enable: tru<…>
flutter: [FBP] \^[[1;30m(setNotifyValue)\^[[0m result: \^[[1;33mtr<…>
flutter: [FBP] \^[[1;30m[[ OnDescriptorWritten ]]\^[[0m result: \^[[1;33m{descriptor_uuid: 2902, remote_id: 9C27C3BF-5C87-1C9E-5858-A44B44F763D9, characteristic_uuid: ff14, success: 1, value: 01000000, error_code: 0, service_uuid: ff12, error_string: succes<…>
flutter: ┌───────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
flutter: │ #0   OtaUpgradeManager.discoverServices (package:flutter_lib_parasols/manager/ota_upgrade_manager.dart:143:16)
flutter: │ #1   <asynchronous suspension>
flutter: ├┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄
flutter: │ 🐛 OtaUpgradeManager descriptor: write 0
flutter: └───────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
flutter: ┌───────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
flutter: │ #0   OtaUpgradeManager.discoverServices.<anonymous closure> (package:flutter_lib_parasols/manager/ota_upgrade_manager.dart:132:18)
flutter: │ #1   _RootZone.runUnaryGuarded (dart:async/zone.dart:1594:10)
flutter: ├┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄
flutter: │ 🐛 OtaUpgradeManager in lastValueStream BLE_READ:
[... 120 lines cut ...]
flutter: \^[[38;5;196m└───────────────────────────────────────────────────────────────────────────────────────────────────────────────────────<…>
flutter: \^[[38;5;196m┌───────────────────────────────────────────────────────────────────────────────────────────────────────────────────────<…>
flutter: \^[[38;5;196m│ #0   OtaUpgradeManager.discoverServices (package:flutter_lib_parasols/manager/ota_upgrade_manager.dart:151:14)<…>
flutter: \^[[38;5;196m│ #1   <asynchronous suspension><…>
flutter: \^[[38;5;196m├┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄<…>
flutter: \^[[38;5;196m│ ⛔ OtaUpgradeManager Error discovering stackTrace: #0      StandardMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:648:7)<…>
flutter: \^[[38;5;196m│ ⛔ #1      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:334:18)<…>
flutter: \^[[38;5;196m│ ⛔ <asynchronous suspension><…>
flutter: \^[[38;5;196m│ ⛔ #2      FlutterBluePlusDarwin._invokeMethod (package:flutter_blue_plus_darwin/flutter_blue_plus_darwin.dart:285:17)<…>
flutter: \^[[38;5;196m│ ⛔ <asynchronous suspension><…>
flutter: \^[[38;5;196m│ ⛔ #3      FlutterBluePlusDarwin.writeDescriptor (package:flutter_blue_plus_darwin/flutter_blue_plus_darwin.dart:262:12)<…>
flutter: \^[[38;5;196m│ ⛔ <asynchronous suspension><…>
flutter: \^[[38;5;196m│ ⛔ #4      FlutterBluePlus._invokeMethod (package:flutter_blue_plus/src/flutter_blue_plus.dart:578:14)<…>
flutter: \^[[38;5;196m│ ⛔ <asynchronous suspension><…>
flutter: \^[[38;5;196m│ ⛔ #5      BluetoothDescriptor.write (package:flutter_blue_plus/src/bluetooth_descriptor.dart:158:7)<…>
flutter: \^[[38;5;196m│ ⛔ <asynchronous suspension><…>
flutter: \^[[38;5;196m│ ⛔ #6      OtaUpgradeManager.discoverServices (package:flutter_lib_parasols/manager/ota_upgrade_manager.dart:144:9)<…>
flutter: \^[[38;5;196m│ ⛔ <asynchronous suspension><…>
flutter: \^[[38;5;196m│ ⛔ <…>
flutter: \^[[38;5;196m└───────────────────────────────────────────────────────────────────────────────────────────────────────────────────────<…>
flutter: [FBP] \^[[1;30m<connect>\^[[0m args: \^[[1;35m{remote_id: 9C27C3BF-5C87-1C9E-5858-A44B44F763D9, auto_connect: <…>
flutter: [FBP] \^[[1;30m(connect)\^[[0m result: \^[[1;33mfal<…>
flutter: [FBP] \^[[1;30m<connect>\^[[0m args: \^[[1;35m{remote_id: 9C27C3BF-5C87-1C9E-5858-A44B44F763D9, auto_connect: <…>
flutter: [FBP] \^[[1;30m(connect)\^[[0m result: \^[[1;33mfal<…>
```

hezhipengzipp, 2025-01-23:
flutter: [FBP] \^[[1;30m[[ OnDiscoveredServices ]]\^[[0m result: \^[[1;33m{
  "remote_id": "9C27C3BF-5C87-1C9E-5858-A44B44F763D9",
  "success": 1,
  "services": [
    {
      "service_uuid": "ff12",
      "remote_id": "9C27C3BF-5C87-1C9E-5858-A44B44F763D9",
      "characteristics": [
        {
          "remote_id": "9C27C3BF-5C87-1C9E-5858-A44B44F763D9",
          "descriptors": [],
          "characteristic_uuid": "ff15",
          "properties": {
            "notify": 0,
            "indicate_encryption_required": 0,
            "write": 0,
            "read": 1,
            "write_without_response": 1,
            "notify_encryption_required": 0,
            "broadcast": 0,
            "extended_properties": 0,
            "indicate": 0,
            "authenticated_signed_writes": 0
          },
          "service_uuid": "ff12"
        },
        {
          "remote_id": "9C27C3BF-5C87-1C9E-5858-A44B44F763D9",
          "descriptors": [
            {
              "remote_id": "9C27C3BF-5C87-1C9E-5858-A44B44F763D9",
              "characteristic_uuid": "ff14",
              "descriptor_uuid": "2902",
              "service_uuid": "ff12"
            }
          ],
          "characteristic_uuid": "ff14",
          "properties": {
            "notify": 1,
            "indicate_encryption_required": 0,
            "write": 0,
            "read": 0,
            "write_without_response": 0,
            "notify_encryption_required": 0,
            "broadcast": 0,
            "extended_properties": 0,
            "indicate": 0,
            "authenticated_signed_writes": 0
          },
          "service_uuid": "ff12"
        }
      ]
    },
    {
      "service_uuid": "180a",
      "remote_id": "9C27C3BF-5C87-1C9E-5858-A44B44F763D9",
      "characteristics": [
        {
          "remote_id": "9C27C3BF-5C87-1C9E-5858-A44B44F763D9",
          "descriptors": [],
          "characteristic_uuid": "2a29",
          "properties": {
            "notify": 0,
            "indicate_encryption_required": 0,
            "write": 0,
            "read": 1,
            "write_without_response": 0,
            "notify_encryption_required": 0,
            "broadcast": 0,
            "extended_properties": 0,
[... 162 lines cut ...]
              "characteristic_uuid": "ff22",
              "descriptor_uuid": "2902",
              "service_uuid": "ff20"
            }
          ],
          "characteristic_uuid": "ff22",
          "properties": {
            "notify": 1,
            "indicate_encryption_required": 0,
            "write": 0,
            "read": 0,
            "write_without_response": 0,
            "notify_encryption_required": 0,
            "broadcast": 0,
            "extended_properties": 0,
            "indicate": 0,
            "authenticated_signed_writes": 0
          },
          "service_uuid": "ff20"
        }
      ]
    }
  ],
  "error_string": "success",
  "error_code": 0<…>

chipweinberger, 2025-01-24:
> Client Characteristic Configuration descriptors must be configured using setNotifyValue

use `setNotifyValue`, not `descriptor.write`

closed 2025-01-24 by chipweinberger