chipweinberger/flutter_blue_plus #1116: [Help]: Android not creating bond properly

Scottie-Fischer, 2025-01-24:
### Requirements

- [x] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

1.32.1

### Flutter Version

3.27.1

### What OS?

Android

### OS Version

Android 11 and Android 14

### Bluetooth Module

Nordic nRF52832

### What is your problem?

When reconnected to a device already "bonded" the (Android) app will still try to request security parameters (PM_EVT_CONN_SEC_PARAMS_REQ) which seems like the bond is never actually made and just a connection/pair is made. When I see the connection update in the gatt callback it shows the bond state being BONDED. 

### Logs

```shell
D/[FBP-Android](  601): [FBP]   rssi: -60
D/[FBP-Android](  601): [FBP]   status: GATT_SUCCESS (0)
D/[FBP-Android](  601): [FBP] onMethodCall: discoverServices
D/[FBP-Android](  601): [FBP] onServicesDiscovered:
D/[FBP-Android](  601): [FBP]   count: 3
D/[FBP-Android](  601): [FBP]   status: 0GATT_SUCCESS
D/[FBP-Android](  601): [FBP] onMethodCall: setNotifyValue
D/[FBP-Android](  601): [FBP] onDescriptorWrite:
D/[FBP-Android](  601): [FBP]   chr: 6e400003-b5a3-f393-e0a9-e50e24dcca9e
D/[FBP-Android](  601): [FBP]   desc: 2902
D/[FBP-Android](  601): [FBP]   status: GATT_SUCCESS (0)
D/[FBP-Android](  601): [FBP] onMethodCall: requestMtu
D/[FBP-Android](  601): [FBP] onMtuChanged:
D/[FBP-Android](  601): [FBP]   mtu: 247
D/[FBP-Android](  601): [FBP]   status: GATT_SUCCESS (0)
D/[FBP-Android](  601): [FBP] onMethodCall: requestConnectionPriority
D/[FBP-Android](  601): [FBP] onConnectionStateChange:disconnected
D/[FBP-Android](  601): [FBP]   status: LINK_SUPERVISION_TIMEOUT
D/[FBP-Android](  601): [FBP] autoconnect is true. Skipping gatt.close()
D/[FBP-Android](  601): [FBP] Bond State at disconnect: 12
D/[FBP-Android](  601): [FBP] onConnectionStateChange:connected
D/[FBP-Android](  601): [FBP]   status: SUCCESS
D/[FBP-Android](  601): [FBP] Device Type on connect: 2
D/[FBP-Android](  601): [FBP] onConnectionStateChange:disconnected
D/[FBP-Android](  601): [FBP]   status: REMOTE_USER_TERMINATED_CONNECTION
D/[FBP-Android](  601): [FBP] autoconnect is true. Skipping gatt.close()
D/[FBP-Android](  601): [FBP] Bond State at disconnect: 12

NRF LOGS:
Disconnected reason: 8
 0> <debug> nrf_ble_gatt: Updating data length to 251 on connection 0x0.
 0> <info> app: Connected
 0> <info> app: BLE_GAP_EVT_CONNECTED Connection interval : 0x27, 0x27.
 0> <debug> nrf_ble_gatt: Data length updated to 27 on connection 0x0.
 0> <debug> nrf_ble_gatt: max_rx_octets: 27
 0> <debug> nrf_ble_gatt: max_tx_octets: 27
 0> <debug> nrf_ble_gatt: max_rx_time: 328
 0> <debug> nrf_ble_gatt: max_tx_time: 328
 0> <info> app: Data Length update completed. data length 27
 0> <info> app: PM_EVT_CONN_SEC_PARAMS_REQ
 0> <info> app: Pair pin not active. Disconnect
 0> <info> app: Connection interval updated: 0x6, 0x6.
 0> <error> app: ERROR 8 [NRF_ERROR_INVALID_STATE] at C:\Users\scott.fischer\workspace\es2\BLE\Src\ble_app.c:227
 0> PC at: 0x000292AD
 0> <error> app: End of error report
```

chipweinberger, 2025-01-25:
sounds like an android / google bug

Scottie-Fischer, 2025-01-28:
Ya agreed. In case anyone else runs into this, iOS and Android seem to handle bonding differently with this event. iOS will remember the security parameters for previous connects, while Android will ask every time. There are other events (BLE_GAP events) you can use to reject bonds, but this is done on the peripheral side.

closed 2025-01-28 by Scottie-Fischer