chipweinberger/flutter_blue_plus #1117: [Help]:  Framework 'flutter_blue_plus' not found

bobosette, 2025-01-27:
### Requirements

- [x] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

1.35.2

### Flutter Version

3.27.3

### What OS?

iOS

### OS Version

17.5

### Bluetooth Module

none

### What is your problem?

Hi everybody.
From this morning when I try to build the application i've got this error: Framework 'flutter_blue_plus' not found

I tried:
- remove podfile.lock
- pod install
- flutter clean
- flutter pub get
- my podfile has the platform setted to 15.6

How can I fix?

Thanks

### Logs

```shell
Could not build the precompiled application for the device.
Error (Xcode): Framework 'flutter_blue_plus' not found

Error (Xcode): Linker command failed with exit code 1 (use -v to see invocation)
```

bobosette, 2025-01-27:
Just Fixed.

Go to ios/Runner.xcodeproj/project.pbxproj and modify all the flutter_blue_plus string with flutter_blue_plus_darwin

closed 2025-01-27 by bobosette

famasf1, 2025-01-29:
> Just Fixed.
> 
> Go to ios/Runner.xcodeproj/project.pbxproj and modify all the flutter_blue_plus string with flutter_blue_plus_darwin

Took me 2 days of delete/clearing cache and tearing my hair to figure this out. This should be part of migration guide.

bobosette, 2025-01-29:
Yes. It was very tricky

ekuleshov, 2025-01-29:
Here is the script I'm using to update iOS (or MacOS) cocoapod dependencies. Works every time.

```
flutter clean
flutter pub get
cd ios
pod repo update
pod update
pod install
```

bobosette, 2025-01-29:
I tried many time but it didn't work.
I had to manually modify project.pbxproj