chipweinberger/flutter_blue_plus #1120: [Help]: Remove bond behavior incorrect when device not connected

3ph, 2025-02-03:
### Requirements

- [x] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

1.35.2

### Flutter Version

3.27.1

### What OS?

Android

### OS Version

Android 13

### Bluetooth Module

not sure

### What is your problem?

This was originally reported in https://github.com/chipweinberger/flutter_blue_plus/issues/1018 but never resolved.

The issue is when the device is disconnected, `removeBond` throws exception. Removing bond shouldn't require the device to be connected. Moreover calling `removeBond` and immediate call `FlutterBluePlus.bondedDevices` will still return the unpaired device. It seems that unbonding process is not finished when the call throws.

Code:
```dart
Future<void> unbondDevice({required String remoteId}) async {
    final bondedDevices = await FlutterBluePlus.bondedDevices;

    final device =
        bondedDevices.firstWhereOrNull((d) => d.remoteId.str == remoteId);

    if (device != null) {
      try {
        await device.removeBond();
      } catch (e) {
        final bondState = device.prevBondState;
        print('!!! Bond state $bondState');
      }
    }

    await FlutterBluePlus.bondedDevices;
  }
```

### Logs

```shell
I/flutter (24392): [FBP] <getBondedDevices> args: null
D/[FBP-Android](24392): [FBP] onMethodCall: getBondedDevices
I/flutter (24392): [FBP] (getBondedDevices) result: {devices: [{remote_id: 00:0D:20:FE:32:5C, platform_name: xxx}]}
I/flutter (24392): [FBP] <removeBond> args: 00:0D:20:FE:32:5C
D/[FBP-Android](24392): [FBP] onMethodCall: removeBond
D/BluetoothAdapter(24392): isLeEnabled(): ON
I/BluetoothDevice(24392): removeBond() for device 00:0D:20:FE:32:5C called by pid: 24392 tid: 24392
I/flutter (24392): [FBP] (removeBond) result: true
I/flutter (24392): !!! Bond state BluetoothBondState.bonding
I/flutter (24392): [FBP] <getBondedDevices> args: null
D/[FBP-Android](24392): [FBP] onMethodCall: getBondedDevices
I/flutter (24392): [FBP] (getBondedDevices) result: {devices: [{remote_id: 00:0D:20:FE:32:5C, platform_name: xxx}]}
D/[FBP-Android](24392): [FBP] OnBondStateChanged: bond-none prev: bonded
I/flutter (24392): [FBP] [[ OnBondStateChanged ]] result: {bond_state: 0, remote_id: 00:0D:20:FE:32:5C, prev_state: 2}
[log] onChange(DevicesCubit, Change { currentState: DevicesState.detected(connectedDevice: null, devices: [Instance of 'RegisteredDevice']), nextState: DevicesState.detected(connectedDevice: null, devices: [Instance of 'RegisteredDevice']) })
```

chipweinberger, 2025-02-03:
are you sure you can remove a bond while disconnected? How will the device know to remove the bond as well?

if you're sure, try it, and open a pr.

chipweinberger, 2025-02-05:
merged https://github.com/chipweinberger/flutter_blue_plus/commit/f323b7bfab62d7c06fc8f1f3240a97395ffa55b3

closed 2025-02-05 by chipweinberger

3ph, 2025-02-06:
That seems to have fixed it! Thanks!