chipweinberger/flutter_blue_plus #1121: [Help]: connection succeeds but methods throw PlatformException "device is disconnected" in second connect cycle

ju-gem, 2025-02-06:
### Requirements

- [x] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.32.12 as well as in example app currently (29.01.) cloned from repo 

### Flutter Version

3.24.3

### What OS?

Android

### OS Version

14

### Bluetooth Module

various unknown (proprietary) modules 

### What is your problem?

Hello,

On one of my test devices (_Motorola Moto G34 5G, Android 14_), whenever I connect, disconnect, and connect again, the connect method returns successfully, but subsequent method calls like discoverServices() fail with PlatformException(discoverServices, device is disconnected, null, null). On a _Dogee N50, Android 13_, everything works as expected.

I reproduced this behavior with the example app by modifying the device page’s connect button to perform 

`await connectAndUpdateStream()`
`await disconnect()`
`await connectAndUpdateStream()`
`await discoverServices()` 

(same effect also with `connect(mtu:null)` - when requesting mtu, the PlatformExceptions already occurs for that request). 
I attached the logs from build to Exception (Note that the initial connect after scan works as usual, I then disconnected and then used the connect button with the modified behavoir).

I think this is the same problem as in [this sparsely explained log](https://github.com/chipweinberger/flutter_blue_plus/issues/492#issuecomment-2606351164).

### Logs

```shell
Launching lib\main.dart on moto g34 5G in debug mode...
Running Gradle task 'assembleDebug'...
You are applying Flutter's app_plugin_loader Gradle plugin imperatively using the apply script method, which is deprecated and will be removed in a future release. Migrate to applying Gradle plugins with the declarative plugins block: https://flutter.dev/to/flutter-gradle-plugin-apply

You are applying Flutter's main Gradle plugin imperatively using the apply script method, which is deprecated and will be removed in a future release. Migrate to applying Gradle plugins with the declarative plugins block: https://flutter.dev/to/flutter-gradle-plugin-apply

Warning: unable to detect project KGP version. Skipping version checking. 
This may be because you have applied KGP after the Flutter Gradle Plugin.
√ Built build\app\outputs\flutter-apk\app-debug.apk
Debug service listening on ws://127.0.0.1:57431/AuH6PHrQv7k=/ws
Syncing files to device moto g34 5G...
D/[FBP-Android](14725): [FBP] onMethodCall: flutterRestart
D/[FBP-Android](14725): [FBP] initializing BluetoothAdapter
D/CompatibilityChangeReporter(14725): Compat change id reported: 263076149; UID 10340; state: ENABLED
D/[FBP-Android](14725): [FBP] disconnectAllDevices(flutterRestart)
D/[FBP-Android](14725): [FBP] connectedPeripherals: 0
I/flutter (14725): [FBP] <setLogLevel> args: 5
D/[FBP-Android](14725): [FBP] onMethodCall: setLogLevel
I/flutter (14725): [FBP] (setLogLevel) result: true
I/flutter (14725): [FBP] <getAdapterState> args: null
D/[FBP-Android](14725): [FBP] onMethodCall: getAdapterState
[... 156 lines cut ...]
I/flutter (14725): [FBP] <connect> args: {remote_id: 00:04:CA:45:59:59, auto_connect: 0}
D/[FBP-Android](14725): [FBP] onMethodCall: connect
D/BluetoothGatt(14725): connect() - device: 00:04:CA:45:59:59, auto: false, eattSupport: false
D/BluetoothGatt(14725): registerApp()
D/BluetoothGatt(14725): registerApp() - UUID=9867c83a-8be8-4f90-8d29-9edd06c7a5d6
D/BluetoothGatt(14725): onClientRegistered() - status=0 clientIf=10
D/BluetoothGatt(14725): onClientConnectionState() - status=0 clientIf=10 device=00:04:CA:45:59:59
D/[FBP-Android](14725): [FBP] onConnectionStateChange:connected
D/[FBP-Android](14725): [FBP]   status: SUCCESS
D/BluetoothGatt(14725): onClientConnectionState() - status=0 clientIf=10 device=00:04:CA:45:59:59
D/[FBP-Android](14725): [FBP] onConnectionStateChange:connected
D/[FBP-Android](14725): [FBP]   status: SUCCESS
D/[FBP-Android](14725): [FBP] [unexpected connection] disconnecting now
D/BluetoothGatt(14725): cancelOpen() - device: 00:04:CA:45:59:59
D/[FBP-Android](14725): [FBP] onScanResult
D/BluetoothGatt(14725): close()
D/BluetoothGatt(14725): unregisterApp() - mClientIf=10
D/BluetoothGatt(14725): onClientConnectionState() - status=0 clientIf=10 device=00:04:CA:45:59:59
I/flutter (14725): [FBP] (connect) result: true
I/flutter (14725): [FBP] [[ OnConnectionStateChanged ]] result: {disconnect_reason_code: 0, disconnect_reason_string: SUCCESS, remote_id: 00:04:CA:45:59:59, connection_state: 1}
I/flutter (14725): [FBP] <discoverServices> args: 00:04:CA:45:59:59
D/[FBP-Android](14725): [FBP] onMethodCall: discoverServices
D/[FBP-Android](14725): [FBP] onScanResult
I/flutter (14725): PlatformException(discoverServices, device is disconnected, null, null)
```

ju-gem, 2025-02-07:
Frankly, I don't know how to fix it. I'd hoped you'd have an idea where it might come from or how to fix it.

chipweinberger, 2025-02-07:
sorry, i wont be able to fix it. 

but since you can easily reproduce it, just start digging into the code, work backwords from the exception.

chipweinberger, 2025-02-17:
closing. not enough info.

closed 2025-02-17 by chipweinberger

hezhipengzipp, 2025-03-27:
how to fix it,

Abhishek01039, 2026-09-17:
Hello @ju-gem 
Did you solve this issue?

I am also getting same error `[unexpected connection] disconnecting now` after connection successful.

Logs:
```

D/[FBP-Android]( 3956): [FBP] onConnectionStateChange:disconnected
D/[FBP-Android]( 3956): [FBP]   status: SUCCESS
D/BluetoothGatt( 3956): close()
D/[FBP-Android]( 3956): [FBP] onMethodCall: connect
D/BluetoothGatt( 3956): connect() - device: XX:XX:XX:XX:35:1B, auto: false
D/BluetoothGatt( 3956): registerApp()
D/BluetoothGatt( 3956): registerApp() - UUID=b8984ff1-5cdb-4b36-a87c-5e7dae2c0ffa
D/BluetoothGatt( 3956): onClientRegistered() - status=0 clientIf=193
D/BluetoothGatt( 3956): onClientConnectionState() - status=0 clientIf=193 connected=true device=XX:XX:XX:XX:35:1B
D/[FBP-Android]( 3956): [FBP] onConnectionStateChange:connected
D/[FBP-Android]( 3956): [FBP]   status: SUCCESS
D/BluetoothGatt( 3956): onClientConnectionState() - status=0 clientIf=193 connected=true device=XX:XX:XX:XX:35:1B
D/[FBP-Android]( 3956): [FBP] onConnectionStateChange:connected
D/[FBP-Android]( 3956): [FBP]   status: SUCCESS
D/[FBP-Android]( 3956): [FBP] [unexpected connection] disconnecting now
```

chipweinberger, 2026-09-17:
That's coming from this code

```
        // Android has an annoying edge case. If disconnect is called right as the connection is being
        // established, Android sometimes ignores the request to disconnect and completes the connection
        // anyway. To handle this case, we make sure the device is still in our currently connecting
        // devices map, otherwise kill the connection since the user was not expecting it to connect.
        private boolean handleUnexpectedConnectionEvents(BluetoothGatt gatt, int newState, String remoteId)
        {
            boolean unexpectedEvent = false;
            if (newState == BluetoothProfile.STATE_CONNECTED) {
                if (mCurrentlyConnectingDevices.get(remoteId) == null && mAutoConnected.get(remoteId) == null) {
                    log(LogLevel.DEBUG, "[unexpected connection] disconnecting now");
```

Did you call disconnect? If so, that's why this code is triggering a disconnect as soon as the connection finished.

What specific issue do you have?

Abhishek01039, 2026-09-21:
Hello @chipweinberger 

I am calling disconnect when my app goes to background. 
When app comes to foreground again then I am calling connect.

disconnect method calls in `didChangeAppLifecycleState`
```
  @override
  void didChangeAppLifecycleState(AppLifecycleState state) {
    switch (state) {
      case AppLifecycleState.resumed:
        connectToRememberedDevice(trigger: MixpanelHandshakeTrigger.appResume);
        break;
      case AppLifecycleState.paused:
      case AppLifecycleState.detached:
        if (!_disableDisconnectOnBackground) {
             _biaDevice?.disconnect();
        }
        break;
      case AppLifecycleState.inactive:
      case AppLifecycleState.hidden:
        break;
    }
  }
```

Connect function:
```
var connState = BluetoothConnectionState.disconnected;
      for (var attempt = 0; attempt < _connectAttempts; attempt++) {
        if (attempt > 0) {
          await Future<void>.delayed(_connectRetryBackoff * attempt);
        }

        connState = BluetoothConnectionState.disconnected;
        try {
          await _biaDevice!.connectAndUpdateStream(
            autoConnect: false,
            timeout: connectTimeout,
          );
          connState = await _biaDevice!.connectionState
              .firstWhere(
                (event) => event == BluetoothConnectionState.connected,
              )
              .timeout(
                connectTimeout,
                onTimeout: () => BluetoothConnectionState.disconnected,
              );
        } catch (e) {
          logError(
            'Connect to remembered device failed '
            '(attempt ${attempt + 1}/$_connectAttempts): $e',
          );
        }
        if (connState == BluetoothConnectionState.connected) break;
      }
      if (connState != BluetoothConnectionState.connected) {
        // Cancel the attempt before giving up: an uncancelled request stays
        // pending in the OS and the next attempt stacks on top of it.
        try {
          await _biaDevice!.disconnect(queue: false);
        } catch (e) {
          logError('Failed to cancel connect attempt: $e');
        }
        _telemetry.report(
          trigger: trigger,
          classification: null,
          result: ConnectResult.notConfigured,
          failure: HandshakeFailure.connectTimeout,
        );
        return ConnectResult.notConfigured;
      }
    }
```

chipweinberger, 2026-09-21:
> Disconnect on App Background

I would simply stop doing this! Is there a specific need to disconnect in the background?

chipweinberger, 2026-09-21:
It's also certainly possible there is a bug here. If you call connect, it should ideally cancel the pending "unexpected connection" if you do it in time.

Feel free to investigate further and open a PR.

Abhishek01039, 2026-09-21:
If I wait for few seconds while app is in background then it's connecting without any issue.
This issue happens when I move the app to background and open the app quickly.

chipweinberger, 2026-09-21:
Yes that makes sense. The disconnect can take awhile. Calling connect and disconnect quickly is not recommended. FBP should ideally handle this better though.

Abhishek01039, 2026-09-21:
So what do you suggest since I can't make disconnect async in `didChangeAppLifecycleState`?

chipweinberger, 2026-09-21:
I suggest leaving the connection active while the app is in the background. iOS and Android both support this.

Abhishek01039, 2026-09-21:
Even I move the app from background to foreground quickly. Logs still says disconnect happens successfully. and then for some reason `unexpected connection` happens which trigger disconnection.

Logs:
```
D/[FBP-Android](19975): [FBP] onMethodCall: disconnect
D/[FBP-Android](19975): [FBP] onConnectionStateChange:disconnected
D/[FBP-Android](19975): [FBP]   status: SUCCESS
D/[FBP-Android](19975): [FBP] onMethodCall: connect
D/[FBP-Android](19975): [FBP] onConnectionStateChange:connected
D/[FBP-Android](19975): [FBP]   status: SUCCESS
D/[FBP-Android](19975): [FBP] onConnectionStateChange:connected
D/[FBP-Android](19975): [FBP]   status: SUCCESS
D/[FBP-Android](19975): [FBP] [unexpected connection] disconnecting now
```

Do you know who could cause this? Does it happening from my code or from plugin?

chipweinberger, 2026-09-21:
Okay, I wont be able to dig into this deeper now. But it could be an issue with the plugin. Feel free to inspect the plugin code deeper.

chipweinberger, 2026-09-21:
I take that back. I'll look into this and push a fix today. Thanks for reporting this issue.

chipweinberger, 2026-09-21:
Pushed two fixes to master:

https://github.com/chipweinberger/flutter_blue_plus/commit/6f3960e5007f4b13f984c8fe08b8666e36b14def
https://github.com/chipweinberger/flutter_blue_plus/commit/fcf66c066a286bf9dc79516b64f25b7f32948b76

Please try out master and let me know if it's working for you.

```
dependencies:
  flutter_blue_plus:
    git:
      url: https://github.com/chipweinberger/flutter_blue_plus.git
      ref: master
      path: packages/flutter_blue_plus

dependency_overrides:
  flutter_blue_plus_android:
    git:
      url: https://github.com/chipweinberger/flutter_blue_plus.git
      ref: master
      path: packages/flutter_blue_plus_android
```

Abhishek01039, 2026-09-21:
Hello @chipweinberger 
Thanks for the quick changes, it works!

Thanks again!