chipweinberger/flutter_blue_plus #1122: [Help]: await FlutterBluePlus.systemDevices([...]) always returns empty list on iOS

itskck, 2025-02-11:
### Requirements

- [x] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.34.5

### Flutter Version

3.27.2

### What OS?

iOS

### OS Version

17.0+

### Bluetooth Module

Unknown

### What is your problem?

I have to get list of devices connected with the iPhone. On Android bondedDevices getter works fine, but on iOS I get empty list no matter what method I use(systemDevices, connectedDevices). I've double checked filters on systemDevices and they are correct.

### Logs

```shell
Code:
print('connectedDevices: ${FlutterBluePlus.connectedDevices}');
                print('systemDevices: ${await FlutterBluePlus.systemDevices([
                      Guid(BTConfig.customServiceUUID),
                    ])}');

Result:
flutter: connectedDevices: []
flutter: systemDevices: []
```

chipweinberger, 2025-02-12:
your issue is missing details on _why you expect it to not be empty_

chipweinberger, 2025-02-16:
not enough info

closed 2025-02-16 by chipweinberger