chipweinberger/flutter_blue_plus #1123: [Help]: `onValueReceived` not firing after the app is relaunched (either hot restart during development or relaunch in release mode)

Chiggy-Playz, 2025-02-15:
### Requirements

- [x] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.34.5

### Flutter Version

3.27.3

### What OS?

Android

### OS Version

15

### Bluetooth Module

NRF52480

### What is your problem?

I have a listener attached to a characteristic's `onValueReceived` stream. It works perfectly fine, but if the app is completely backout out from or quit and relaunched, the stream stops listening. The device connects, I send commands, commands are received by the device AND the device responds, but FBP doesnt fire the listener. I've tested using NRFConnect and can confirm my device is working as intended. I've attached logs below as well as a screen recording. My testing was done in the example app with minimal edits (editing what value is written, 0x00 or 0x01). You can see on the left on nrfconnect that the value is updated whenever i write, and it should be updated on the right in the same characteristic after subscribing. This works fine when I launch the app the first time. But when I back out, relaunch, it doesn't work. Device still works, value is updated on the left not on the right in the example app. 
PS: The logs below contain logs for both the first launch and the relaunch.

(Recording)[https://drive.google.com/file/d/1ke5t5JQCqYD03zVuptGsx3XfJoM2z5tV/view?usp=sharing] (Couldn't upload it here so uploaded on my google drive)

### Logs

```shell
I/flutter (21182): [FBP] <setLogLevel> args: 5
I/flutter (21182): [FBP] (setLogLevel) result: true
I/flutter (21182): [FBP] <getAdapterState> args: null
I/flutter (21182): [FBP] (getAdapterState) result: {adapter_state: 4}
I/flutter (21182): [FBP] <getSystemDevices> args: null
I/flutter (21182): [FBP] (getSystemDevices) result: {devices: []}
I/flutter (21182): [FBP] <startScan> args: {with_services: [], with_remote_ids: [], with_names: [], with_keywords: [], with_msd: [], with_service_data: [], continuous_updates: false, continuous_divisor: 1, android_legacy: false, android_scan_mode: 2, android_uses_fine_location: false}
I/flutter (21182): [FBP] (startScan) result: true
I/flutter (21182): [FBP] [[ OnScanResponse ]] result: {advertisements: [{rssi: -53, manufacturer_data: {6: 01092022202294f16c2f5af7cee12a8f811fe05d90870545ca5f94}, remote_id: 28:B2:99:5D:4E:ED, service_data: {}}]}
I/flutter (21182): [FBP] [[ OnScanResponse ]] result: {advertisements: [{rssi: -66, connectable: 1, manufacturer_data: {}, remote_id: C1:E6:0F:2A:87:58, platform_name: RV_Test, service_data: {}, adv_name: RV_Test, service_uuids: [12345678-1234-5678-1234-56789abcdef0]}]}
I/flutter (21182): [FBP] [[ OnScanResponse ]] result: {advertisements: [{rssi: -90, manufacturer_data: {6: 010f202219815b48a4531f544c168a7f8edfea7bf80f02d5bc9b0f}, remote_id: 30:99:39:EF:9D:E6, service_data: {}}]}
I/flutter (21182): [FBP] [[ OnScanResponse ]] result: {advertisements: [{rssi: -104, connectable: 1, manufacturer_data: {76: 10073d1f126c516068}, remote_id: 5E:3C:81:2F:FB:A8, tx_power_level: 7, service_data: {}}]}
I/flutter (21182): [FBP] [[ OnScanResponse ]] result: {advertisements: [{rssi: -88, connectable: 1, manufacturer_data: {17153: 0206da34bccbc6}, remote_id: 59:5D:25:56:A4:C9, platform_name: LE-Bose Flex SoundLink, tx_power_level: 10, service_data: {}, adv_name: LE-Bose Flex SoundLink, service_uuids: [febe]}]}
I/flutter (21182): [FBP] [[ OnScanResponse ]] result: {advertisements: [{rssi: -94, connectable: 1, manufacturer_data: {65535: 0301}, remote_id: B8:59:CE:C2:2E:AF, platform_name: JR_JioSTB-RFLSBLHH2087997, service_data: {}, adv_name: JR_JioSTB-RFLSBLHH2087997, service_uuids: [8980f798-0614-11ea-8d71-362b9e155667]}]}
I/flutter (21182): [FBP] [[ OnScanResponse ]] result: {advertisements: [{rssi: -94, manufacturer_data: {}, remote_id: 10:08:AB:7B:D6:2F, service_data: {fef3: 4a17235a585153113225968a4772cd94cda2e9a944dd865340f237}}]}
I/flutter (21182): [FBP] [[ OnScanResponse ]] result: {advertisements: [{rssi: -86, manufacturer_data: {76: 12020001}, remote_id: E3:17:10:FF:40:A0, service_data: {}}]}
I/flutter (21182): [FBP] [[ OnScanResponse ]] result: {advertisements: [{rssi: -89, connectable: 1, manufacturer_data: {117: 4204832011012d0000000000000000000000000000000000}, remote_id: 6C:70:CB:48:21:85, platform_name: Q-Series Soundbar, service_data: {}, adv_name: Q-Series Soundbar}]}
I/flutter (21182): [FBP] [[ OnScanResponse ]] result: {advertisements: [{rssi: -89, connectable: 1, manufacturer_data: {117: 42048301112407036c70cb48218500016e70cb4821840106}, remote_id: 6C:70:CB:48:21:85, platform_name: Q-Series Soundbar, service_data: {}, adv_name: Q-Series Soundbar}]}
I/flutter (21182): [FBP] <connect> args: {remote_id: C1:E6:0F:2A:87:58, auto_connect: 0}
I/flutter (21182): [FBP] (connect) result: true
I/flutter (21182): [FBP] [[ OnScanResponse ]] result: {advertisements: [{rssi: -90, connectable: 1, manufacturer_data: {117: 4204832011012d0000000000000000000000000000000000}, remote_id: 6C:70:CB:48:21:85, platform_name: Q-Series Soundbar, service_data: {}, adv_name: Q-Series Soundbar}]}
I/flutter (21182): [FBP] [[ OnConnectionStateChanged ]] result: {disconnect_reason_code: 0, disconnect_reason_string: SUCCESS, remote_id: C1:E6:0F:2A:87:58, connection_state: 1}
I/flutter (21182): [FBP] <readRssi> args: C1:E6:0F:2A:87:58
I/flutter (21182): [FBP] (readRssi) result: true
I/flutter (21182): [FBP] [[ OnReadRssi ]] result: {error_string: GATT_SUCCESS, rssi: -57, success: 1, remote_id: C1:E6:0F:2A:87:58, error_code: 0}
I/flutter (21182): [FBP] [[ OnScanResponse ]] result: {advertisements: [{rssi: -93, connectable: 1, manufacturer_data: {769: 0103485904380103485904380401014800000102030405060708090a0b0c0d0e0f}, remote_id: A4:C1:38:04:59:48, platform_name: telink_mesh1, service_data: {}, adv_name: telink_mesh1}]}
I/flutter (21182): [FBP] <discoverServices> args: C1:E6:0F:2A:87:58
I/flutter (21182): [FBP] (discoverServices) result: true
I/flutter (21182): [FBP] [[ OnDiscoveredServices ]] result: {
I/flutter (21182):   "error_string": "GATT_SUCCESS",
I/flutter (21182):   "success": 1,
I/flutter (21182):   "remote_id": "C1:E6:0F:2A:87:58",
[... 117 lines cut ...]
I/flutter (21182): [FBP] <setNotifyValue> args: {remote_id: C1:E6:0F:2A:87:58, service_uuid: 1801, characteristic_uuid: 2a05, force_indications: false, enable: true}
I/flutter (21182): [FBP] (setNotifyValue) result: true
I/flutter (21182): [FBP] [[ OnDescriptorWritten ]] result: {error_string: GATT_SUCCESS, service_uuid: 1801, success: 1, remote_id: C1:E6:0F:2A:87:58, descriptor_uuid: 2902, error_code: 0, characteristic_uuid: 2a05, value: 0200}
I/flutter (21182): [FBP] [[ OnScanResponse ]] result: {advertisements: [{rssi: -88, manufacturer_data: {76: 12020003}, remote_id: D7:7A:3D:7E:C1:8F, service_data: {}}]}
I/flutter (21182): [FBP] <setNotifyValue> args: {remote_id: C1:E6:0F:2A:87:58, service_uuid: 12345678-1234-5678-1234-56789abcdef0, characteristic_uuid: 12345678-1234-5678-1234-56789abcdef1, force_indications: false, enable: true}
I/flutter (21182): [FBP] (setNotifyValue) result: true
I/flutter (21182): [FBP] [[ OnScanResponse ]] result: {advertisements: [{rssi: -88, connectable: 1, manufacturer_data: {117: 42048301112407036c70cb48218500016e70cb4821840106}, remote_id: 6C:70:CB:48:21:85, platform_name: Q-Series Soundbar, service_data: {}, adv_name: Q-Series Soundbar}]}
I/flutter (21182): [FBP] [[ OnDescriptorWritten ]] result: {error_string: GATT_SUCCESS, service_uuid: 12345678-1234-5678-1234-56789abcdef0, success: 1, remote_id: C1:E6:0F:2A:87:58, descriptor_uuid: 2902, error_code: 0, characteristic_uuid: 12345678-1234-5678-1234-56789abcdef1, value: 0100}
I/flutter (21182): [FBP] <readCharacteristic> args: {remote_id: C1:E6:0F:2A:87:58, service_uuid: 12345678-1234-5678-1234-56789abcdef0, characteristic_uuid: 12345678-1234-5678-1234-56789abcdef1}
I/flutter (21182): [FBP] (readCharacteristic) result: true
I/flutter (21182): [FBP] [[ OnCharacteristicReceived ]] result: {error_string: GATT_SUCCESS, service_uuid: 12345678-1234-5678-1234-56789abcdef0, success: 1, remote_id: C1:E6:0F:2A:87:58, error_code: 0, characteristic_uuid: 12345678-1234-5678-1234-56789abcdef1, value: 01}
I/flutter (21182): [FBP] [[ OnScanResponse ]] result: {advertisements: [{rssi: -89, connectable: 1, manufacturer_data: {117: 4204832011012d0000000000000000000000000000000000}, remote_id: 6C:70:CB:48:21:85, platform_name: Q-Series Soundbar, service_data: {}, adv_name: Q-Series Soundbar}]}
[... cut ...]
I/flutter (21182): [FBP] <writeCharacteristic> args: {remote_id: C1:E6:0F:2A:87:58, service_uuid: 12345678-1234-5678-1234-56789abcdef0, characteristic_uuid: 12345678-1234-5678-1234-56789abcdef2, write_type: 0, allow_long_write: 0, value: 00}
I/flutter (21182): [FBP] (writeCharacteristic) result: true
I/flutter (21182): [FBP] [[ OnCharacteristicWritten ]] result: {error_string: GATT_SUCCESS, service_uuid: 12345678-1234-5678-1234-56789abcdef0, success: 1, remote_id: C1:E6:0F:2A:87:58, error_code: 0, characteristic_uuid: 12345678-1234-5678-1234-56789abcdef2, value: 00}
```

chipweinberger, 2025-02-15:
after reconnection you need to setNotifyValue again

Chiggy-Playz, 2025-02-15:
I do set it again in my own app. I also clicked the `subscribe` button in the example app which calls that function. The logs also mention that `setNotifyValue` was set true for that characteristic (`12345678-1234-5678-1234-56789abcdef1`).

chipweinberger, 2025-02-15:
ok. well i can tell you this works fine in my app.

Chiggy-Playz, 2025-02-15:
What can I do? How can I debug this? Can you point me a bit 😓 
I really need to get this bug fixed before my app's release.

chipweinberger, 2025-02-15:
1. try other phones
2. could be bug in your firmware

Chiggy-Playz, 2025-02-15:
Tested another device running android 14. Same issue. And firmware is not the issue since a service emulated using nrfconnect sending notifs is also not working. I'm really stuck...

chipweinberger, 2025-02-15:
> And firmware is not the issue since a service emulated using nrfconnect sending notifs is also not working.

?? this shows its probably a firmware issue.

closed 2025-02-16 by Chiggy-Playz