chipweinberger/flutter_blue_plus #1126: [Help]: Prioritize write operations

dlewis2017, 2025-02-18:
### Requirements

- [x] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

1.34.4

### Flutter Version

3.29.0

### What OS?

iOS

### OS Version

iPad OS 18.1.1

### Bluetooth Module

Nordic nrF5340

### What is your problem?

I'm wondering if there's a way to prioritize writes when streaming data? Currently, it seems when listening to a characteristic that is receiving a lot of data, it takes almost a minute for the write command I sent, to be sent successfully from iOS side of things.  Is there a mitigation I can do?

### Logs

```shell
Too many logs to post
```

chipweinberger, 2025-02-18:
good question but i'm unaware of anything that can be done.

closed 2025-02-24 by chipweinberger

dlewis2017, 2025-03-31:
Looking into this further, is there a way to setup writes to be on a separate method channel and/or separate iOS thread? It seems my operations are getting bogged down because everything is operating on the main thread. But if writes and reads could at least be separate, that would help a lot