chipweinberger/flutter_blue_plus #1127: [Help]: PlatformException(androidException, java.lang.IllegalArgumentException: Can only use lower 16 bits for requestCode

sumaro, 2025-02-19:
### Requirements

- [x] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

1.34.5

### Flutter Version

3.22.1

### What OS?

Android

### OS Version

android 13

### Bluetooth Module

android

### What is your problem?

When I changed MainActivity: FlutterActivity to MainActivity: FlutterFragmentActivity, an exception occurred after using turnOn

### Logs

```shell
PlatformException(androidException, java.lang.IllegalArgumentException: Can only use lower 16 bits for requestCode, java.lang.IllegalArgumentException: Can only use lower 16 bits for requestCode
I/flutter ( 3828): 	at androidx.fragment.app.FragmentActivity.checkForValidRequestCode(FragmentActivity.java:715)
I/flutter ( 3828): 	at androidx.fragment.app.FragmentActivity.startActivityForResult(FragmentActivity.java:660)
I/flutter ( 3828): 	at com.lib.flutter_blue_plus.FlutterBluePlusPlugin.lambda$onMethodCall$1$com-lib-flutter_blue_plus-FlutterBluePlusPlugin(FlutterBluePlusPlugin.java:439)
I/flutter ( 3828): 	at com.lib.flutter_blue_plus.FlutterBluePlusPlugin$$ExternalSyntheticLambda2.op(Unknown Source:4)
I/flutter ( 3828): 	at com.lib.flutter_blue_plus.FlutterBluePlusPlugin.ensurePermissions(FlutterBluePlusPlugin.java:1555)
I/flutter ( 3828): 	at com.lib.flutter_blue_plus.FlutterBluePlusPlugin.onMethodCall(FlutterBluePlusPlugin.java:424)
I/flutter ( 3828): 	at io.flutter.plugin.common.MethodChannel$IncomingMethodCallHandler.onMessage(MethodChannel.java:267)
```

chipweinberger, 2025-02-20:
FlutterFragmentActivity is not something we intend to support.

sumaro, 2025-02-25:
private final int enableBluetoothRequestCode = 1879842617;
enableBluetoothRequestCode this value is too large, can you set it smaller?

chipweinberger, 2025-02-25:
https://github.com/chipweinberger/flutter_blue_plus/commit/e67ae83d781b36056b61e038af525b798d8277b1