chipweinberger/flutter_blue_plus #1128: [Help]: On Web (Chrome), example is stuck at "Bluetooth Adapter is unknown"

panmari, 2025-02-21:
### Requirements

- [x] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.35.3

### Flutter Version

3.29.0

### What OS?

All

### OS Version

Windows 10, Chrome Stable

### Bluetooth Module

Onit Bluetooth Adapter

### What is your problem?

When starting the example app on WEB, I'm greeted with the "Bluetooth Adapter is unknown" screen, with no way to progress. 

### Logs

```shell
None
```

chipweinberger, 2025-02-21:
open PR, thanks

panmari, 2025-02-22:
Am I understanding correctly then that the example app is not yet expected to work for web? If so, I'm happy to chip in. Do you have a few pointers on what would need to be done?

ekuleshov, 2025-02-22:
I pushed a few fixes to main related to this.

@tnc1997 it seems like the `getAdapterState()` implementation wasn't updated when the "defaults" were added to the platform API. See my last commit there.

I also added some dummy values for the `webOptionalServices` parameter of the startScan() call in the example app (it is required for web and introduced in https://github.com/chipweinberger/flutter_blue_plus/pull/1124). Also had to ignore some odd errors in the `discoverServices()` implementation for web.

With those changes I see some services and their chars when running example in a web browser. @tnc1997 @kubatatami could you please review this if you have some time.

The example app could also use some improvements. E.g. to allow to enter "withServices" and "webOptionalServices" parameters in the UI to use when scanning for devices.

tnc1997, 2025-02-22:
> @tnc1997 it seems like the getAdapterState() implementation wasn't updated when the \"defaults\" were added to the platform API

The reason that it defaulted to unknown on web is that it is not currently possible to accurately determine the adapter state on web as far as I am aware.

ekuleshov, 2025-02-22:
> The reason that it defaulted to unknown on web is that it is not currently possible to accurately determine the adapter state on web as far as I am aware.

Right. We had similar issue with `isSupported()` call. So, changed implementation to use result of the `isSupported()` call to return "on" when true, and "unknown" otherwise.

panmari, 2025-02-23:
Thanks, that fixed the issue!

closed 2025-02-23 by panmari