chipweinberger/flutter_blue_plus #1132: [Help]: FlutterBluePlusException with android-code: 133

thisadee2897, 2025-02-24:
### Requirements

- [x] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

^1.35.3

### Flutter Version

sdks:   dart: ">=3.6.0 <4.0.0"   flutter: ">=3.27.0"

### What OS?

Android

### OS Version

7 ,11 ,15

### Bluetooth Module

Tablet Android 7.4

### What is your problem?

On Android 7, we encounter issues—specifically, a FlutterBluePlusException with android-code: 133 (ANDROID_SPECIFIC_ERROR)—while Android 11 and 14 run stably.
```dart
await device.connect()
```

```dart
import 'dart:async';
import 'package:device_info_plus/device_info_plus.dart';
import 'package:flutter/foundation.dart';
import 'package:permission_handler/permission_handler.dart';
import 'package:sunford_pos/export.dart';

class BluetoothScanNotifier extends StateNotifier<AsyncValue<List<BluetoothDevice>>> {
  BluetoothScanNotifier(this.ref) : super(const AsyncValue.loading()) {
    startScan();
  }

  final Ref ref;
  StreamSubscription<List<ScanResult>>? _scanSubscription;
  StreamSubscription<BluetoothConnectionState>? _subscription;
  Future<void> requestPermissions() async {
    final deviceInfo = DeviceInfoPlugin();
    final androidInfo = await deviceInfo.androidInfo;
    int sdkInt = androidInfo.version.sdkInt;

    Map<Permission, PermissionStatus> statuses;

    if (sdkInt >= 31) {
      statuses = await [
        Permission.bluetoothScan,
        Permission.bluetoothConnect,
      ].request();
    } else {
      statuses = await [Permission.location].request();
    }

![Image](https://github.com/user-attachments/assets/c01ee94f-2829-4204-811d-31ae0bc91c38)

    if (statuses.values.any((status) => status.isDenied || status.isPermanentlyDenied)) {
      if (kDebugMode) print("❌ Permission denied! ขออนุญาติเพิ่มเติม");
      // แจ้งให้ผู้ใช้ไปเปิด permission ใน settings
      await openAppSettings();
    }
  }

  Future<void> startScan() async {
    try {
      state = const AsyncValue.loading();
      await disconnectAllDevices();
      await requestPermissions();
      //clean up old devices and start scanning
      ref.read(listDevicesProvider.notifier).state = [];
      final _devices = <BluetoothDevice>[];
      state = const AsyncValue.loading();
      // List<BluetoothDevice> _devices = [];
      ref.read(listDevicesProvider.notifier).state = [];

      state = const AsyncValue.data([]);

      FlutterBluePlus.startScan(timeout: const Duration(seconds: 5));

      _scanSubscription?.cancel(); // ป้องกันการซ้ำซ้อน
      _scanSubscription = FlutterBluePlus.scanResults.listen((results) {
        for (ScanResult result in results) {
          if (!_devices.any((element) => element.remoteId == result.device.remoteId)) {
            List<Guid> uuids = result.advertisementData.serviceUuids;
            bool uuid1 = uuids.any((element) => element.toString().contains('ffe0'));
            bool uuid2 = uuids.any((element) => element.toString().contains('ffe1'));
            if (result.device.platformName.isNotEmpty && (uuid1 || uuid2)) {
              _devices.add(result.device);
              ref.read(listDevicesProvider.notifier).state = [..._devices];
            }
          }
        }
        state = AsyncValue.data([..._devices]);
      });
    } catch (e, stackTrace) {
      if (kDebugMode) print("Error-------------------------->: $e");
      state = AsyncValue.error(e, stackTrace);
    }
  }

//connect for Device
  Future<void> connectDevice(BluetoothDevice device) async {
    state = const AsyncValue.loading(); // กำลังเชื่อมต่อ
    try {
      // สั่งยกเลิกเชื่อมต่ออุปกรณ์ที่เชื่อมต่ออยู่ก่อนหน้า
      await disconnectDevice(device);
      // เชื่อมต่ออุปกรณ์ และ อ่านค่าจากอุปกรณ์ connectionState คือ สถานะการเชื่อมต่อ
      await Future.delayed(const Duration(seconds: 2));
      await device.connect().then((value) {
        device.connectionState.listen((_conNext) async {
          if (kDebugMode) print("🔵 Connection State: $_conNext");
          if (_conNext == BluetoothConnectionState.connected) {
            ref.read(selectedBluetoothDeviceProvider.notifier).state = device;
            // เก็บอุปกรณ์ที่เลือกไว้
            if (kDebugMode) print("✅ Connected to ${device.platformName}");
            await device.discoverServices().then((List<BluetoothService> v) {
              discover(v);
            });
            return;
            // เริ่มอ่านค่าจากอุปกรณ์;
          } else if (_conNext == BluetoothConnectionState.disconnected) {
            if (kDebugMode) print("⚠️ Disconnected, trying to reconnect...");
            return;
          }
        });
      });
    } catch (e, stackTrace) {
      if (kDebugMode) print("Error-------------------------->: $e");
      state = AsyncValue.error(e, stackTrace);
      return;
    }
  }

  //discoverServices
  void discover(List<BluetoothService> services) {
    try {
      BluetoothCharacteristic characteristic = services.last.characteristics.first;
      ref.read(characteristicProvider.notifier).state = characteristic;
      //อ่านแบบ stream เปิดให้อ่านค่าทุกครั้งที่มีการเปลี่ยนแปลง
      characteristic.setNotifyValue(true);
      characteristic.lastValueStream.listen((value) {
        Uint8List dataX = Uint8List.fromList(value);
        String text = String.fromCharCodes(dataX);
        String cleanedText = text.trim().replaceAll('+', '').replaceAll(' ', ''); // ลบ "+" และช่องว่าง
        double currentWeight = double.tryParse(cleanedText) ?? 0.0;
        ref.read(currentWeighProvider.notifier).currentWeigh = currentWeight;
        state = const AsyncValue.data([]);
      });
    } catch (e, stackTrace) {
      if (kDebugMode) print("Error-------------------------->: $e");
      state = AsyncValue.error(e, stackTrace);
    }
  }

//disconnect  all devices
  Future<void> disconnectAllDevices() async {
    final devices = ref.read(listDevicesProvider);
    for (final device in devices) {
      await device.disconnect();
    }
    ref.read(selectedBluetoothDeviceProvider.notifier).state = null;
    ref.read(currentWeighProvider.notifier).currentWeigh = 0;
  }

  //disconnect on device
  Future<void> disconnectDevice(BluetoothDevice device) async {
    final devices = ref.read(listDevicesProvider);
    for (final dv in devices) {
      if (dv != device) {
        await dv.disconnect();
      }
    }
    ref.read(selectedBluetoothDeviceProvider.notifier).state = null;
    ref.read(currentWeighProvider.notifier).currentWeigh = 0;
  }
}

final bluetoothScanProvider = StateNotifierProvider<BluetoothScanNotifier, AsyncValue<List<BluetoothDevice>>>(
  (ref) => BluetoothScanNotifier(ref),
);

final listDevicesProvider = StateProvider<List<BluetoothDevice>>((ref) => []);
final selectedBluetoothDeviceProvider = StateProvider<BluetoothDevice?>((ref) => null);

class BluetoothConnectionNotifier extends StateNotifier<bool> {
  BluetoothConnectionNotifier(this.ref) : super(false);

  final Ref ref;
  StreamSubscription<BluetoothConnectionState>? _subscription;

  Future<void> monitorConnection(BluetoothDevice device) async {
    _subscription?.cancel(); // ยกเลิกการฟังเก่าถ้ามี
    state = true; // ตั้งค่าให้เชื่อมต่ออยู่

    _subscription = device.connectionState.listen((connectionState) {
      if (connectionState == BluetoothConnectionState.disconnected) {
        state = false; // หลุดการเชื่อมต่อ
        // ref.read(connectedDeviceProvider.notifier).state = null;
        _showDisconnectedAlert();
      }
    });
  }

  void _showDisconnectedAlert() {
    final context = ref.read(scaffoldMessengerKeyProvider)?.context;
    if (context == null) return;

    ScaffoldMessenger.of(context).showSnackBar(
      SnackBar(
        content: const Text('เชื่อมต่อกับอุปกรณ์เสีย'),
        action: SnackBarAction(
          label: 'ปิด',
          onPressed: () {},
        ),
      ),
    );
  }

  Future<void> pushDataToDevice() async {
    final characteristic = ref.read(characteristicProvider);
    if (characteristic != null) {
      await characteristic.setNotifyValue(false);
    }
  }

  Future<void> setNotifyOnProvider() async {
    final characteristic = ref.read(characteristicProvider);
    if (characteristic != null) {
      await characteristic.setNotifyValue(true);
    }
  }

  @override
  void dispose() {
    _subscription?.cancel();
    super.dispose();
  }
}

final bluetoothConnectionProvider = StateNotifierProvider<BluetoothConnectionNotifier, bool>(
  (ref) => BluetoothConnectionNotifier(ref),
);

final scaffoldMessengerKeyProvider = Provider<ScaffoldMessengerState?>((ref) {
  final context = ref.watch(rootNavigatorKey).currentContext;
  return context != null ? ScaffoldMessenger.of(context) : null;
});

final characteristicProvider = StateProvider<BluetoothCharacteristic?>((ref) => null);
```

### Logs

```shell
FlutterBluePlusException with android-code: 133 (ANDROID_SPECIFIC_ERROR)
```

ekuleshov, 2025-02-24:
What's the full stackTrace for this error?

minhkimlong, 2025-02-25:
```
D/[FBP-Android](30398): [FBP]   status: GATT_SUCCESS (0)
I/flutter (30398): [FBP] (setNotifyValue) result: true
I/flutter (30398): [FBP] [[ OnDescriptorWritten ]] result: {error_string: GATT_SUCCESS, service_uuid: fff0, success: 1, remote_id: 44:1A:84:82:02:D4, descriptor_uuid: 2902, error_code: 0, characteristic_uuid: fff1, value: 0100}
D/BluetoothGatt(30398): onPhyUpdate() - status=0 address=44:1A:84:82:02:D4 txPhy=2 rxPhy=2
D/BluetoothGatt(30398): onClientConnectionState() - status=8 clientIf=6 device=44:1A:84:82:02:D4
D/[FBP-Android](30398): [FBP] onConnectionStateChange:disconnected
D/[FBP-Android](30398): [FBP]   status: LINK_SUPERVISION_TIMEOUT
D/BluetoothGatt(30398): close()
D/BluetoothGatt(30398): unregisterApp() - mClientIf=6
I/flutter (30398): [FBP] [[ OnConnectionStateChanged ]] result: {disconnect_reason_code: 8, disconnect_reason_string: LINK_SUPERVISION_TIMEOUT, remote_id: 44:1A:84:82:02:D4, connection_state: 0}
```

chipweinberger, 2025-02-25:
it means your ble peripheral is no longer responding.

closed 2025-02-25 by chipweinberger

minhkimlong, 2025-02-25:
I check that their existing Aromatech app (Xamarin) works well with the AromaDream diffuser. Only failed Google Pixel 7A (Android 13). Samsung S21/M53 works so well. @chipweinberger

minhkimlong, 2025-02-25:
```
I/flutter ( 1270): [FBP] <writeCharacteristic> result: true
I/flutter ( 1270): [FBP] [[ OnCharacteristicWritten ]] result: {error_string: GATT_SUCCESS, service_uuid: fff0, success: 1, remote_id: 44:1A:84:82:02:D4, error_code: 0, characteristic_uuid: fff2, value: a5aaac8d571707e902190b003102c5ccca}
I/flutter ( 1270): [FBP] [[ OnCharacteristicReceived ]] result: {error_string: GATT_SUCCESS, service_uuid: fff0, success: 1, remote_id: 44:1A:84:82:02:D4, error_code: 0, characteristic_uuid: fff1, value: a5aaac09571741434bc5ccca}
I/flutter ( 1270): [FBP] [[ OnCharacteristicReceived ]] result: {error_string: GATT_SUCCESS, service_uuid: fff0, success: 1, remote_id: 44:1A:84:82:02:D4, error_code: 0, characteristic_uuid: fff1, value: a5aaac35530a07e902190b003102100000001900}
I/flutter ( 1270): [FBP] [[ OnCharacteristicReceived ]] result: {error_string: GATT_SUCCESS, service_uuid: fff0, success: 1, remote_id: 44:1A:84:82:02:D4, error_code: 0, characteristic_uuid: fff1, value: 5a0700171e01441a84823d990000000000000000}
I/flutter ( 1270): [FBP] [[ OnCharacteristicReceived ]] result: {error_string: GATT_SUCCESS, service_uuid: fff0, success: 1, remote_id: 44:1A:84:82:02:D4, error_code: 0, characteristic_uuid: fff1, value: 0001000101c5ccca}
D/BluetoothGatt( 1270): onPhyUpdate() - status=0 address=44:1A:84:82:02:D4 txPhy=2 rxPhy=2
D/BluetoothGatt( 1270): onClientConnectionState() - status=8 clientIf=6 device=44:1A:84:82:02:D4
D/[FBP-Android]( 1270): [FBP] onConnectionStateChange:disconnected
D/[FBP-Android]( 1270): [FBP]   status: LINK_SUPERVISION_TIMEOUT
D/BluetoothGatt( 1270): close()
D/BluetoothGatt( 1270): unregisterApp() - mClientIf=6
I/flutter ( 1270): [FBP] [[ OnConnectionStateChanged ]] result: {disconnect_reason_code: 8, disconnect_reason_string: LINK_SUPERVISION_TIMEOUT, remote_id: 44:1A:84:82:02:D4, connection_state: 0}
```

minhkimlong, 2025-02-25:
I/flutter ( 1270): [FBP] <writeCharacteristic> result: true
I/flutter ( 1270): [FBP] [[ OnCharacteristicWritten ]] result: {error_string: GATT_SUCCESS, service_uuid: fff0, success: 1, remote_id: 44:1A:84:82:02:D4, error_code: 0, characteristic_uuid: fff2, value: a5aaac8d571707e902190b003102c5ccca}
I/flutter ( 1270): [FBP] [[ OnCharacteristicReceived ]] result: {error_string: GATT_SUCCESS, service_uuid: fff0, success: 1, remote_id: 44:1A:84:82:02:D4, error_code: 0, characteristic_uuid: fff1, value: a5aaac09571741434bc5ccca}
I/flutter ( 1270): [FBP] [[ OnCharacteristicReceived ]] result: {error_string: GATT_SUCCESS, service_uuid: fff0, success: 1, remote_id: 44:1A:84:82:02:D4, error_code: 0, characteristic_uuid: fff1, value: a5aaac35530a07e902190b003102100000001900}
I/flutter ( 1270): [FBP] [[ OnCharacteristicReceived ]] result: {error_string: GATT_SUCCESS, service_uuid: fff0, success: 1, remote_id: 44:1A:84:82:02:D4, error_code: 0, characteristic_uuid: fff1, value: 5a0700171e01441a84823d990000000000000000}
I/flutter ( 1270): [FBP] [[ OnCharacteristicReceived ]] result: {error_string: GATT_SUCCESS, service_uuid: fff0, success: 1, remote_id: 44:1A:84:82:02:D4, error_code: 0, characteristic_uuid: fff1, value: 0001000101c5ccca}
D/BluetoothGatt( 1270): onPhyUpdate() - status=0 address=44:1A:84:82:02:D4 txPhy=2 rxPhy=2
D/BluetoothGatt( 1270): onClientConnectionState() - status=8 clientIf=6 device=44:1A:84:82:02:D4
D/[FBP-Android]( 1270): [FBP] onConnectionStateChange:disconnected
D/[FBP-Android]( 1270): [FBP]   status: LINK_SUPERVISION_TIMEOUT
D/BluetoothGatt( 1270): close()
D/BluetoothGatt( 1270): unregisterApp() - mClientIf=6
I/flutter ( 1270): [FBP] [[ OnConnectionStateChanged ]] result: {disconnect_reason_code: 8, disconnect_reason_string: LINK_SUPERVISION_TIMEOUT, remote_id: 44:1A:84:82:02:D4, connection_state: 0}

minhkimlong, 2025-02-25:
This code will FlutterBluePlusException with android-code: 133
if (Platform.isAndroid) {
        setPreferredPhy(
          txPhy: 1,
          rxPhy: 1,
          option: PhyCoding.noPreferred,
        );
      }
@thisadee2897 @chipweinberger @ekuleshov

minhkimlong, 2025-02-25:
```
await device.connect();
if (Platform.isAndroid) {
setPreferredPhy(
txPhy: 1,
rxPhy: 1,
option: PhyCoding.noPreferred,
);
}
```
@thisadee2897

ekuleshov, 2025-02-25:
As @chipweinberger said, it is a firmware issue. Your peripheral is disconnecting when it is attempted to connect with parameters that hardware is unable to support.

As you noted, PHY or also when higher MTU is not supported by device.
I also put some fixes for Android 6 devices to handle the BLE connection mode.

minhkimlong, 2025-02-25:
Thank @ekuleshov