chipweinberger/flutter_blue_plus #1133: [Help]: Error while connecting to ble device on Android 14

sahil4-scrobits, 2025-02-24:
### Requirements

- [x] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.34.5

### Flutter Version

3.27.1

### What OS?

Android

### OS Version

14

### Bluetooth Module

ble 3.0

### What is your problem?

Getting this error few times:

On Android 7, we encounter issues—specifically, a FlutterBluePlusException with android-code: 133 (ANDROID_SPECIFIC_ERROR)—while Android 11 and 14 run stably.

````
Future<void> connectToDevice(BluetoothDevice device) async {
    await requestPermissions();
 // Ensure location permission is granted

    try {
      isConnecting = true;
      await stopScan();
      _updateState();

      // Connect to the device
      await device.connect(timeout: const Duration(seconds: 30));
      await Future.delayed(Duration(milliseconds: 500));
      // **Workaround for double pairing popup issue**
      if (Platform.isAndroid) {
        // bool isBonded = await device.prevB; // Check if already bonded
        // if (!isBonded) {
        await device.createBond(); // Proactively create bond
        // }
      }

      isConnecting = false;
      _updateState();
    } catch (e) {
      isConnecting = false;
      _handleError("Connect Error", e);
      _updateState();
    }
  }

### Logs

```shell
D/BluetoothGatt(11808): onClientConnectionState() - status=0 clientIf=6 device=XX:XX:XX:XX:C3:6E
D/[FBP-Android](11808): [FBP] onConnectionStateChange:connected
D/[FBP-Android](11808): [FBP]   status: SUCCESS
I/flutter (11808): Provider StateNotifierProvider<BluetoothService, BleModel>#1b976 updated from Instance of 'BleModel' to Instance of 'BleModel'
I/flutter (11808): widget updated
I/flutter (11808): nearby Device home
D/BluetoothGatt(11808): onConnectionUpdated() - Device=XX:XX:XX:XX:C3:6E interval=6 latency=0 timeout=500 status=0
D/BluetoothGatt(11808): onClientConnectionState() - status=22 clientIf=6 device=XX:XX:XX:XX:C3:6E
D/[FBP-Android](11808): [FBP] onConnectionStateChange:disconnected
D/[FBP-Android](11808): [FBP]   status: CONNECTION_TERMINATED_BY_LOCAL_HOST
D/BluetoothGatt(11808): close()
D/BluetoothGatt(11808): unregisterApp() - mClientIf=6
D/[FBP-Android](11808): [FBP] OnBondStateChanged: bond-none prev: bonded
I/flutter (11808): error while connect FlutterBluePlusException | discoverServices | fbp-code: 6 | device is not connected
```

chipweinberger, 2025-02-25:
your problem: CONNECTION_TERMINATED_BY_LOCAL_HOST

closed 2025-02-25 by chipweinberger

chipweinberger, 2025-02-25:
you'll need to access the firmware logs on your peripheral

Rupsero, 2025-03-06:
If the bonding gets completed during the service discovery, the connection stays stable.
If the bonding is completed before or after service discovery, I get the error:
D/BluetoothGatt(29769): onClientConnectionState() - status=22 clientIf=6 device=XX:XX:XX:XX:75:98
D/[FBP-Android](29769): [FBP] onConnectionStateChange:disconnected
D/[FBP-Android](29769): [FBP]   status: CONNECTION_TERMINATED_BY_LOCAL_HOST
as well...
Haven´t figured out a solution for this one...