chipweinberger/flutter_blue_plus #1135: [Help]:

NaynaAsodariya, 2025-02-26:
### Requirements

- [x] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

1.35.3

### Flutter Version

3.27.3

### What OS?

iOS

### OS Version

18.3.1

### Bluetooth Module

Not sure

### What is your problem?

Connect to device, discover services, subscribe to it's characteristic - works fine

Device added new characteristics to add some new features 
Disconnect from device and connect to device again, discover service remain unchanged
The characteristics added for new feature is not listing

------Check if discover service have new characteristics -------
var services =  await bluetoothDevice!.discoverServices();

 bool isDeviceHasNewCharacteristic = false;

 for (var service in services) {
      for (var characteristic in service.characteristics) {
        if (characteristic.serviceUuid == serviceID &&
            characteristic.characteristicUuid == characteristicId) {
          isDeviceHasNewCharacteristic = true;
          _subscribeToBandErrorLogs(characteristic);
        }
      }
    }

if(!isDeviceHasNewCharacteristic){
print("The services list is not updated with new characteristics");
}

I also have added code to reset the discoverService but this method never called

 device!.onServicesReset.listen((onData) async {
          _log.debug(
            "New service got and now subscribe to new service",
          );
          var services =  await bluetoothDevice!.discoverServices();
        });  

### Logs

```shell
-
```

chipweinberger, 2025-02-26:
logs needed

NaynaAsodariya, 2025-02-27:
device.discoverServices() is working correctly, but the issue occurs when the device was previously paired. After updating the device services, iOS caches the old services and does not update to the newly added services.

chipweinberger, 2025-02-28:
are you listening to onServicesReset?

```
// - uses the GAP Services Changed characteristic (0x2A05)
// - you must call discoverServices() again
device.onServicesReset.listen(() async {
    print("Services Reset");
    await device.discoverServices();
});
```

closed 2025-02-28 by chipweinberger