chipweinberger/flutter_blue_plus #1136: [Help]: How is each platform registered?

June1104, 2025-02-28:
### Requirements

- [x] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.35.3

### Flutter Version

3.24.1

### What OS?

Android

### OS Version

unimportant

### Bluetooth Module

unimportant

### What is your problem?

The question I asked has nothing to do with the function. I was confused when looking at the source code. This project defines a general interface package, namely flutter_blue_plus_platform_interface, which defines all the general methods. For example, when I call the FlutterBluePlus.startScan method, I click in to see that it is actually calling FlutterBluePlusPlatform.instance.startScan. FlutterBluePlusPlatform is an abstract class. If I use an Android phone, the actual call is the startScan method under the FlutterBluePlusAndroid class. I just want to know how the implementation class of this specific platform is registered. I see that there is a registerWith method under each class, such as:
`static void registerWith() {
FlutterBluePlusPlatform.instance = FlutterBluePlusAndroid();
},`
This code can clearly see that it corresponds to a specific class here, but I can't find where the registerWith method is called. This is my doubt. If you have time, can you give me an answer? Thank you.

### Logs

```shell
unimportant
```

ekuleshov, 2025-02-28:
@June1104 you can read more about this in the Flutter documentation at https://docs.flutter.dev/packages-and-plugins/developing-packages#federated-plugins

closed 2025-02-28 by ekuleshov

June1104, 2025-03-03:
OK, I'll read the document, thank you.