chipweinberger/flutter_blue_plus #1139: [Help]: Example app fails scanning on Android

panmari, 2025-03-02:
### Requirements

- [x] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

HEAD

### Flutter Version

3.29.0

### What OS?

Android

### OS Version

Baklava

### Bluetooth Module

Android Pixel 9 Pro

### What is your problem?

Clicking on "Scan" fails with `Invalid argument: Instance of 'Guid'`

I traced this back to the presence of `webOptionalServices`. I suggested a fix at #1137 to only allow setting it on web. Maybe there's another way of fixing it.

### Logs

```shell
Launching lib\main.dart on Pixel 9 Pro in debug mode...
√ Built build\app\outputs\flutter-apk\app-debug.apk
I/flutter (24379): [IMPORTANT:flutter/shell/platform/android/android_context_vk_impeller.cc(60)] Using the Impeller rendering backend (Vulkan).
Connecting to VM Service at ws://127.0.0.1:53404/qpt7mz2ES2w=/ws
Connected to the VM Service.
D/[FBP-Android](24379): [FBP] onMethodCall: flutterRestart
D/[FBP-Android](24379): [FBP] initializing BluetoothAdapter
D/[FBP-Android](24379): [FBP] disconnectAllDevices(flutterRestart)
D/[FBP-Android](24379): [FBP] connectedPeripherals: 0
I/flutter (24379): [FBP] <setLogLevel> args: 5
D/[FBP-Android](24379): [FBP] onMethodCall: setLogLevel
I/flutter (24379): [FBP] (setLogLevel) result: true
I/flutter (24379): [FBP] <getAdapterState> args: null
D/ProfileInstaller(24379): Installing profile for com.lib.flutter_blue_plus_example
D/[FBP-Android](24379): [FBP] onMethodCall: getAdapterState
I/flutter (24379): [FBP] (getAdapterState) result: {adapter_state: 4}
I/ImeTracker(24379): com.lib.flutter_blue_plus_example:6bfaa817: onCancelled at PHASE_CLIENT_ALREADY_HIDDEN
I/flutter (24379): [FBP] <getSystemDevices> args: null
D/[FBP-Android](24379): [FBP] onMethodCall: getSystemDevices
I/flutter (24379): [FBP] (getSystemDevices) result: {devices: []}
I/flutter (24379): [FBP] <startScan> args: {with_services: [], with_remote_ids: [], with_names: [], with_keywords: [], with_msd: [], with_service_data: [], continuous_updates: false, continuous_divisor: 1, android_legacy: false, android_scan_mode: 2, android_uses_fine_location: false, web_optional_services: [180f, 1800, 6e400001-b5a3-f393-e0a9-e50e24dcca9e]}
I/flutter (24379): Invalid argument: Instance of 'Guid'
```

chipweinberger, 2025-03-03:
Yes, webOptionalServices broke scanning on Android. @ekuleshov 

```
I/flutter (12733): [FBP] <startScan> args: {with_services: [], with_remote_ids: [], with_names: [], with_keywords: [], with_msd: [], with_service_data: [], continuous_updates: false, continuous_divisor: 1, android_legacy: false, android_scan_mode: 2, android_uses_fine_location: false, web_optional_services: [180f, 1800, 6e400001-b5a3-f393-e0a9-e50e24dcca9e]}
I/flutter (12733): Invalid argument: Instance of 'Guid'
I/flutter (12733): backtrace: #0      StandardMessageCodec.writeValue (package:flutter/src/services/message_codecs.dart:464:7)
I/flutter (12733): #1      StandardMessageCodec.writeValue (package:flutter/src/services/message_codecs.dart:454:9)
I/flutter (12733): #2      StandardMessageCodec.writeValue.<anonymous closure> (package:flutter/src/services/message_codecs.dart:461:9)
I/flutter (12733): #3      _LinkedHashMapMixin.forEach (dart:collection-patch/compact_hash.dart:633:13)
I/flutter (12733): #4      StandardMessageCodec.writeValue (package:flutter/src/services/message_codecs.dart:459:13)
I/flutter (12733): #5      StandardMethodCodec.encodeMethodCall (package:flutter/src/services/message_codecs.dart:602:18)
I/flutter (12733): #6      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:323:34)
I/flutter (12733): #7      MethodChannel.invokeMethod (package:flutter/src/services/platform_channel.dart:507:12)
I/flutter (12733): #8      FlutterBluePlusAndroid._invokeMethod (package:flutter_blue_plus_android/flutter_blue_plus_android.dart:408:37)
I/flutter (12733): <asynchronous suspension>
I/flutter (12733): #9      FlutterBluePlusAndroid.startScan (package:flutter_blue_plus_android/flu
```

ekuleshov, 2025-03-03:
> Yes, webOptionalServices broke scanning on Android. [@ekuleshov](https://github.com/ekuleshov)

I will take a look

chipweinberger, 2025-03-03:
fixed it

chipweinberger, 2025-03-03:
though in this case we could just skip it

closed 2025-03-03 by chipweinberger

panmari, 2025-03-03:
Thanks for the fix!