chipweinberger/flutter_blue_plus #1144: [Help]:

jagiro, 2025-03-05:
### Requirements

- [x] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

1.34.5

### Flutter Version

1.34.5

### What OS?

Android

### OS Version

Miui global 13.07

### Bluetooth Module

Custom

### What is your problem?

**App Context:**
Our application scans for custom BLE devices, connects to them, writes data, and then disconnects. This is a standard client-only BLE usage with no pairing required.

**Issue Description:**
We're encountering permission errors on Xiaomi Redmi 12C devices running MIUI Global 13.07. The errors don't appear in the app logs but are visible in the Android logcat:
```
CopyCaught a RuntimeException from the binder stub implementation.
java.lang.SecurityException: Need BLUETOOTH_ADMIN permission: uid 10316 does not have android.permission.BLUETOOTH_ADMIN.
at android.app.ContextImpl.enforce(ContextImpl.java:2189)
at android.app.ContextImpl.enforceCallingPermission(ContextImpl.java:2207)
at com.miui.whetstone.PowerKeeperPolicy.stopLeScan(PowerKeeperPolicy.java:1300)
at com.miui.whetstone.IPowerKeeperPolicy$Stub.onTransact(IPowerKeeperPolicy.java:287)
at android.os.Binder.execTransactInternal(Binder.java:1190)
at android.os.Binder.execTransact(Binder.java:1149)
```

**Implementation Details:**
Our AndroidManifest.xml includes all the permissions as specified in your readme
```
    <uses-permission android:name="android.permission.BLUETOOTH_SCAN" android:usesPermissionFlags="neverForLocation" />
    <uses-permission android:name="android.permission.BLUETOOTH_CONNECT" />
    <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />

    <!-- legacy for Android 11 or lower -->
    <uses-permission android:name="android.permission.BLUETOOTH" android:maxSdkVersion="30" />
    <uses-permission android:name="android.permission.BLUETOOTH_ADMIN" android:maxSdkVersion="30" />
```
We're properly requesting runtime permissions for bluetoothScan and bluetoothConnect

The UID (10316) mentioned in the error matches our app's UID in the logs, confirming it's related to our application

**Investigation:**
I consulted Claude AI about this issue, which suggested that this is likely related to MIUI's custom "PowerKeeper" battery optimization system. Even though MIUI 13 is based on Android 12, MIUI's custom implementation still requires the legacy BLUETOOTH_ADMIN permission without respecting the maxSdkVersion limits.
The error occurs when MIUI's PowerKeeperPolicy attempts to manage the BLE scanning process but lacks the BLUETOOTH_ADMIN permission that our app defined with android:maxSdkVersion="30" as recommended in the documentation.

### Logs

```shell
Caught a RuntimeException from the binder stub implementation.
java.lang.SecurityException: Need BLUETOOTH_ADMIN permission: uid 10316 does not have android.permission.BLUETOOTH_ADMIN.
at android.app.ContextImpl.enforce(ContextImpl.java:2189)
at android.app.ContextImpl.enforceCallingPermission(ContextImpl.java:2207)
at com.miui.whetstone.PowerKeeperPolicy.stopLeScan(PowerKeeperPolicy.java:1300)
at com.miui.whetstone.IPowerKeeperPolicy$Stub.onTransact(IPowerKeeperPolicy.java:287)
at android.os.Binder.execTransactInternal(Binder.java:1190)
at android.os.Binder.execTransact(Binder.java:1149)
```

chipweinberger, 2025-04-02:
We can't fix this.

Since the stack trace mentions com.miui.whetstone.PowerKeeperPolicy, it’s possible that the error is influenced by MIUI’s custom power management policies. While adding the required permission should resolve the issue for standard Android devices, be aware that some manufacturer-specific implementations might have additional restrictions or behaviors.

chipweinberger, 2025-04-02:
> I consulted Claude AI about this issue, which suggested that this is likely related to MIUI's custom "PowerKeeper" battery optimization system. Even though MIUI 13 is based on Android 12, MIUI's [...]
The error occurs when MIUI's PowerKeeperPolicy attempts to manage the BLE scanning process but lacks the BLUETOOTH_ADMIN permission that our app defined with android:maxSdkVersion="30" as recommended in the documentation.

So are you saying you removed `android:maxSdkVersion="30"` and it fixed it?

chipweinberger, 2025-04-21:
closing. no response.

closed 2025-04-21 by chipweinberger