chipweinberger/flutter_blue_plus #1151: [Help]: Scanning issue

PoonamKesarkar, 2025-03-12:
### Requirements

- [x] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.35.3

### Flutter Version

3.27.3

### What OS?

Android

### OS Version

11

### Bluetooth Module

Medical device

### What is your problem?

During scanning in ios swift code first time it receive 23 byte service data and 16 byte service data from same device. In flutter during scanning I have received only 16 byte service data every time. In android nRF application also showing 23 byte and 16 byte service data. 

### Logs

```shell
I/flutter (22586): [FBP] <startScan> args: {with_services: [], with_remote_ids: [], with_names: [], with_keywords: [], with_msd: [], with_service_data: [{service: fd19, data: , mask: }], continuous_updates: true, continuous_divisor: 1, android_legacy: false, android_scan_mode: 2, android_uses_fine_location: false}
I/flutter (22586): [FBP] [[ OnScanResponse ]] result: {advertisements: [{rssi: -36, connectable: 1, manufacturer_data: {}, remote_id: F7:33:75:C4:29:1B, service_data: {fd19: b0fb265fe596e7f9ae5885015c03ace1}}]}
I/flutter (22586): scanresult: ScanResult{device: BluetoothDevice{remoteId: F7:33:75:C4:29:1B, platformName: , services: null}, advertisementData: AdvertisementData{advName: , txPowerLevel: null, appearance: null, connectable: true, manufacturerData: {}, serviceData: {fd19: [176, 251, 38, 95, 229, 150, 231, 249, 174, 88, 133, 1, 92, 3, 172, 225]}, serviceUuids: []}, rssi: -36, timeStamp: 2025-03-12 17:10:12.831597}
I/flutter (22586): result of setState
D/[FBP-Android](22586): [FBP] onScanResult
I/flutter (22586): [FBP] [[ OnScanResponse ]] result: {advertisements: [{rssi: -35, connectable: 1, manufacturer_data: {}, remote_id: F7:33:75:C4:29:1B, service_data: {fd19: b0fb265fe596e7f9ae5885015c03ace1}}]}
I/flutter (22586): scanresult: ScanResult{device: BluetoothDevice{remoteId: F7:33:75:C4:29:1B, platformName: , services: null}, advertisementData: AdvertisementData{advName: , txPowerLevel: null, appearance: null, connectable: true, manufacturerData: {}, serviceData: {fd19: [176, 251, 38, 95, 229, 150, 231, 249, 174, 88, 133, 1, 92, 3, 172, 225]}, serviceUuids: []}, rssi: -35, timeStamp: 2025-03-12 17:10:14.102429}
I/flutter (22586): result of setState
D/[FBP-Android](22586): [FBP] onScanResult
I/flutter (22586): [FBP] [[ OnScanResponse ]] result: {advertisements: [{rssi: -36, connectable: 1, manufacturer_data: {}, remote_id: F7:33:75:C4:29:1B, service_data: {fd19: b0fb265fe596e7f9ae5885015c03ace1}}]}
I/flutter (22586): scanresult: ScanResult{device: BluetoothDevice{remoteId: F7:33:75:C4:29:1B, platformName: , services: null}, advertisementData: AdvertisementData{advName: , txPowerLevel: null, appearance: null, connectable: true, manufacturerData: {}, serviceData: {fd19: [176, 251, 38, 95, 229, 150, 231, 249, 174, 88, 133, 1, 92, 3, 172, 225]}, serviceUuids: []}, rssi: -36, timeStamp: 2025-03-12 17:10:16.671566}
```

tototo23, 2025-03-13:
I don't know if it's the same issue, but i'm also facing regression on iOS with flutter_blue_plus 1.35.X
For now i had to revert to 1.34.5 to receive correctly the scan/advertising data on iOS

chipweinberger, 2025-04-02:
@tototo23 what issue?

chipweinberger, 2025-04-02:
@tnc1997 anything you can think of? not much to go on here.

tnc1997, 2025-04-03:
> @tnc1997 anything you can think of? not much to go on here.

Hi @chipweinberger, after a brief glance over the description nothing comes to mind, would I be right in thinking that this appears to only affect iOS and not the other platforms?

chipweinberger, 2025-04-03:
okay. im going to close this. not enough info to reproduce.

i recommend just opening a PR if you want it fixed.

closed 2025-04-03 by chipweinberger