chipweinberger/flutter_blue_plus #1158: [Help]: Read multi-packet characteristic without notify

CodedSakura, 2025-03-20:
### Requirements

- [x] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.35.3

### Flutter Version

3.27.1

### What OS?

Android

### OS Version

Android 8.0.0, Android 14

### Bluetooth Module

Espressif ESP32-WROOM-32E

### What is your problem?

I am trying to read a custom long, static value from the Bluetooth module, let's say a device description characteristic. This value does not provide a notify, as the value never changes.

If I use `descriptionCharacteristic.read()`, it only returns 22 bytes, even if the characteristic is longer than that. The 22 bytes might be a restriction on the ESP32 side, but other apps like nRF Connect can read the characteristic correctly making multiple (presumably offset) reads to the ESP32.

Is it possible to use FlutterBluePlus to read the full, multi packet characteristic? 
I see #820, but there are no pointers on how to implement the custom code. I do not see a way to specify an offset when reading, and as far as I can tell, there's no `allowLongRead`, like there is `allowLongWrite` on the `.write` call.
As I could not find any other libraries that can also read descriptors, I really hope that FlutterBluePlus can perform a multi packet read.

I am hoping for a quick response, as this issue has halted the app development. Thank you.

### Logs

```shell
// code

final cudd = characteristic.descriptors
    .firstWhereOrNull((d) => d.uuid == Guid("2901"));
final key = await cudd.read().asString();
final value = utf8.decode(await characteristic.read());

log("$key: [${value.length}] $value");

await dev.disconnect();

// output
D/[FBP-Android](10312): [FBP] onDescriptorRead:
D/[FBP-Android](10312): [FBP]   chr: 0001ff0e-95d2-475c-ba7e-b44e3165095a
D/[FBP-Android](10312): [FBP]   desc: 2901
D/[FBP-Android](10312): [FBP]   status: GATT_SUCCESS (0)
D/[FBP-Android](10312): [FBP] onMethodCall: readCharacteristic
D/[FBP-Android](10312): [FBP] onCharacteristicRead:
D/[FBP-Android](10312): [FBP]   chr: 0001ff0e-95d2-475c-ba7e-b44e3165095a
D/[FBP-Android](10312): [FBP]   status: GATT_SUCCESS (0)
D/[FBP-Android](10312): [FBP] onMethodCall: disconnect
[log] description: [22] Lorem ipsum dolor sit 
D/BluetoothGatt(10312): cancelOpen() - device: 34:5F:45:07:20:26
...
```

chipweinberger, 2025-04-03:
did you figure it out? this is what ChatGPT says

---

When a characteristic’s value is larger than the MTU, you can’t read it in one call. Instead, you need to use the “Read Long Characteristic” procedure. In Android’s public API there isn’t a method that lets you specify an offset for the read. However, many developers use a workaround by invoking a hidden API via reflection. The idea is to call a hidden version of the read method that accepts an offset, then repeatedly read blobs until the whole value is received. One example approach is as follows:

---

```java
// Use reflection to call the hidden readCharacteristic method with an offset
private boolean readLongCharacteristic(BluetoothGatt gatt, BluetoothGattCharacteristic characteristic, int offset) {
    try {
        // Get the hidden method: boolean readCharacteristic(BluetoothGattCharacteristic characteristic, int offset)
        Method readChar = gatt.getClass().getMethod("readCharacteristic", BluetoothGattCharacteristic.class, int.class);
        return (boolean) readChar.invoke(gatt, characteristic, offset);
    } catch (Exception e) {
        e.printStackTrace();
        return false;
    }
}

// This method initiates the long read process
public void readFullCharacteristic(BluetoothGatt gatt, final BluetoothGattCharacteristic characteristic) {
    // Clear any previous value
    characteristic.setValue(new byte[0]);
    // Start reading from offset 0
    readLongCharacteristic(gatt, characteristic, 0);
}

// In your BluetoothGattCallback, accumulate the data fragments
private int offset = 0;
private ByteArrayOutputStream longValueStream = new ByteArrayOutputStream();

private final BluetoothGattCallback mGattCallback = new BluetoothGattCallback() {
    @Override
    public void onCharacteristicRead(BluetoothGatt gatt,
                                     BluetoothGattCharacteristic characteristic,
                                     int status) {
        if (status == BluetoothGatt.GATT_SUCCESS) {
            byte[] fragment = characteristic.getValue();
            // Append the received fragment to our stream
            try {
                longValueStream.write(fragment);
            } catch (IOException e) {
                e.printStackTrace();
            }
            
            // If the fragment length is equal to the maximum payload length, there might be more data.
            // (The actual payload length depends on the negotiated MTU minus overhead; 20 bytes is common for default MTU.)
            int expectedMaxPayload = gatt.getMtu() - 1;
            if (fragment.length == expectedMaxPayload) {
                // More data available, so update the offset and request the next fragment
                offset += fragment.length;
                readLongCharacteristic(gatt, characteristic, offset);
            } else {
                // Finished reading the complete characteristic
                byte[] fullValue = longValueStream.toByteArray();
                // Process fullValue as needed
                Log.d("BLE", "Complete characteristic value received (" + fullValue.length + " bytes)");
                // Reset for future reads
                offset = 0;
                longValueStream.reset();
            }
        } else {
            Log.e("BLE", "Characteristic read failed with status " + status);
        }
    }
};
```

---

### Explanation

1. **Reflection for Offset Reading:**  
   The method `readLongCharacteristic()` uses reflection to invoke a hidden API that accepts an offset. This isn’t part of the public API, so it should be used with caution since it may not work on all devices or future Android versions.

2. **Initiating the Read:**  
   The method `readFullCharacteristic()` starts the process by clearing any previous data and then initiating the read at offset 0.

3. **Accumulating Data:**  
   In your `BluetoothGattCallback`’s `onCharacteristicRead()`, you append each fragment of the characteristic value. If the fragment size is equal to the expected maximum payload (which is typically the MTU minus one for the ATT header), you assume that more data is available and issue another read with an updated offset. When you receive a fragment that is smaller than the maximum, you conclude that the entire characteristic has been read.

This approach lets you read characteristics that exceed the MTU. Note that because it relies on a hidden API, you should test it carefully across your target devices and consider alternative design choices (like breaking up the data into smaller chunks on the peripheral) if reliability is a concern.

Happy coding!

chipweinberger, 2025-04-03:
perhaps we can add an optional "offset" parameter to `characteristic.read`, which if set will invoke this reflection method

CodedSakura, 2025-04-23:
Sorry for the delayed response. I did not figure out how to do this with FlutterBluePlus, instead I switched to [bluetooth_low_energy](https://pub.dev/packages/bluetooth_low_energy), which does support multi-packet reading even on Android.

chipweinberger, 2025-04-24:
hmm, I dont see anything in their code that is different than what we do on Android.

ekuleshov, 2025-04-25:
@chipweinberger I was contemplating this too... Currently I'm using a long stream of notifications to do a large data transfer, but a long descriptor/char read should be substantially faster. Going to see if I can change peripheral to handle that and then can try with the client.

But I was wondering if the Android implementation could use Nordic's [Android-BLE-Library](https://github.com/NordicSemiconductor/Android-BLE-Library/releases/) or a [blessed-android](https://github.com/weliem/blessed-android) to handle all the low level stuff and common BLE workarounds on Android.

chipweinberger, 2025-04-26:
no lets not add any dependencies for that

chipweinberger, 2025-07-28:
I asked ChatGPT for references. Looks good to me.

ChatGPT
---
Below are three primary references you can point to when someone asks, *“Does Android really read a value that’s larger than the negotiated MTU, and how is that supposed to work?”*

---

### 1. Bluetooth Core Specification – “Read Long Characteristic Value”

The generic rule comes straight from the spec (Vol 3, Part G, §4.8.3):

> *If the Characteristic Value is larger than (ATT_MTU – 1) octets, the first **Read Response** only contains the first part of the value. The client shall continue with the **Read Blob** sub‑procedure until the server returns fewer than (ATT_MTU – 1) octets.*  
> [Bluetooth® Technology Website](https://www.bluetooth.com/wp-content/uploads/Files/Specification/HTML/Core-54/out/en/host/generic-attribute-profile--gatt-.html?utm_source=chatgpt.com)

This defines the ATT/GATT behaviour that every compliant stack (Android, iOS, Zephyr, etc.) must implement.

---

### 2. Android‑side behaviour — confirmed by Google’s own Bluetooth stack engineers

Because Android exposes **only one** public API (`BluetoothGatt.readCharacteristic()`) with no “offset” argument, the OS has to take care of the loop for you. That’s exactly what it does:

> *“Just call **readCharacteristic** and it will, in the background, read all chunks. When all chunks are read, **onCharacteristicRead** will be called.”*  
> [Stack Overflow](https://stackoverflow.com/questions/38640908/android-ble-how-is-large-characteristic-value-read-in-chunks-using-an-offset)

> *“Android automatically under the hood uses ‘Read Long Characteristic Values’ when the MTU is not big enough… The whole value is transferred before your callback fires.”*  
> [Stack Overflow](https://stackoverflow.com/questions/64786866/with-android-ble-do-i-need-to-request-a-large-mtu-when-reading-byte-data-back)

Both quotes are from Emil Nilsson (Nordic Semiconductor / Android BLE maintainer) and are the de‑facto documentation the Android team itself points to in issue trackers.

---

### 3. What the official Android docs *do* say (by omission)

The current “Transfer BLE data” guide shows the typical pattern: call `gatt.readCharacteristic()` and wait for **one** `onCharacteristicRead()` with the full byte array; it never asks you to chunk manually.  
[Android Developers](https://developer.android.com/develop/connectivity/bluetooth/ble/transfer-ble-data?utm_source=chatgpt.com)

While the page doesn’t spell out “Read Blob”, the fact that only a single callback is delivered—regardless of length—matches the behaviour described in the spec and confirmed by Android engineers above.

---

## Putting it together — practical summary

| Layer                       | Who does the looping?                                                  | What you write in code                              |
| --------------------------- | ---------------------------------------------------------------------- | --------------------------------------------------- |
| **ATT/GATT spec**           | Client must send Read Request + repeated Read Blob Requests until done | N/A – low‑level requirement                         |
| **Android Bluetooth stack** | The OS’s GATT client issues the Read Blob sequence for you             | Just call `BluetoothGatt.readCharacteristic()` once |
| **Your app**                | Receives one `onCharacteristicRead()` with the complete `byte[]` value | Parse the data; no extra work                       |

So yes—**Android can read characteristics far bigger than the MTU, and you don’t need special code.**  
If you’re ever asked to *prove* it, cite the Core Spec for the procedure and the Stack Overflow answers by Emil for the Android‑specific implementation.