chipweinberger/flutter_blue_plus #1160: [Help]: MTU  is always 23

AndyAls, 2025-03-28:
### Requirements

- [x] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

1.35.2

### Flutter Version

3.27.4

### What OS?

Android

### OS Version

anroid 15

### Bluetooth Module

xiaomi 15

### What is your problem?

If the Bluetooth module firmware has a default MTU of 512 and does not support negotiation to modify the MTU value, the obtained MTU is always 23.  

### Logs

```shell
This should be a consistently reproducible issue.
```

AndyAls, 2025-03-28:
If you call the connect function with mtu set to null, even for Bluetooth modules that support MTU negotiation, the MTU value will consistently return ​23. The MTU value I obtained using nRF Connect is 512.

sukhi-23, 2025-04-01:
Have you resolved this issue? Because I am facing this issue for Android 15

chipweinberger, 2025-04-03:
> If you call the connect function with mtu set to null, even for Bluetooth modules that support MTU negotiation, the MTU value will consistently return ​23

This is expected behavior. If you don't request a specific MTU, the default is 23.

If you specific a higher MTU, it will do the negotiation process and use the negotiated value.

closed 2025-04-03 by chipweinberger