chipweinberger/flutter_blue_plus #1161: [Help]: iOS app crashes when disconnecting from Raspberry Pi's Bluetooth connection

SevenCho, 2025-03-29:
### Requirements

- [x] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.35.3 

### Flutter Version

3.27.1

### What OS?

iOS

### OS Version

iOS 15.7 / 18.3.2

### Bluetooth Module

OrangePi 20U5622

### What is your problem?

It works fine with versions 1.33.6 and earlier. 
With later versions, the app crashes when Bluetooth is disconnected, and the stack trace shows that -[CBService invalidate] is called continuously until the app crashes.

[image]

### Logs

```shell
[FBP-iOS] handleMethodCall: stopScan
flutter: [FBP] \^[[1;30m(stopScan)\^[[0m result: \^[[1;33mtr<…>
* thread #1, queue = 'com.apple.main-thread', stop reason = EXC_BAD_ACCESS (code=2, address=0x16d7cfff0)
    frame #0: 0x00000001afb56f50 CoreBluetooth`-[CBService invalidate] + 8
CoreBluetooth`-[CBService invalidate]:
->  0x1afb56f50 <+8>:  stp    x28, x27, [sp, #0x190]
    0x1afb56f54 <+12>: stp    x26, x25, [sp, #0x1a0]
    0x1afb56f58 <+16>: stp    x24, x23, [sp, #0x1b0]
    0x1afb56f5c <+20>: stp    x22, x21, [sp, #0x1c0]
Target 0: (Runner) stopped.
```

chipweinberger, 2025-04-03:
can you please find the specific commit that broke it?

```
dependencies:
  flutter_blue_plus:
    git:
      url: https://github.com/chipweinberger/flutter_blue_plus.git
      ref: a1b2c3d4e5f6g7h8i9j0  # replace with your commit hash
```

and then ideally find the specific line and open a PR. Thanks

SevenCho, 2025-04-03:
The following commit caused problems, and the specific causes are still under investigation.

[[Refactor] replace secondaryServiceUuid with primaryServiceUuid](https://github.com/chipweinberger/flutter_blue_plus/commit/2064c18e40d1a01e2bf1e382c746acc588cc029f)

SevenCho, 2025-04-03:
> can you please find the specific commit that broke it?
>
> ```
> dependencies:
>   flutter_blue_plus:
>     git:
>       url: https://github.com/chipweinberger/flutter_blue_plus.git
>       ref: [...]

The following commit caused problems, and the specific causes are still under investigation.

[[Refactor] replace secondaryServiceUuid with primaryServiceUuid](https://github.com/chipweinberger/flutter_blue_plus/commit/2064c18e40d1a01e2bf1e382c746acc588cc029f)

SevenCho, 2025-04-12:
[image]

So far, I found that the problem caused by calling **discoverIncludedServices** in **peripheral:didDiscoverServices:**;

I don't quite understand why the **discoverIncludedServices** function needs to be called here?

Is it better to provide a separate function for users who need it?

chipweinberger, 2025-04-12:
I'm not sure the right solution.

SevenCho, 2025-04-12:
> I'm not sure the right solution.

I personally think that **discoverIncludedServices** is the same as **discoverServices**, and can provide an interface for us to call separately?

chipweinberger, 2025-04-12:
theyre not the same

chipweinberger, 2025-04-12:
you can look at the previous code to see how we used to find included services. try asking AI too.

SevenCho, 2025-04-12:
> you can look at the previous code to see how we used to find included services. try asking AI too.

I only took a cursory look at the source code, and I'm not sure why **discoverIncludedServices** needs to be called in **peripheral:didDiscoverServices:**;
I haven't encountered any need to use **discoverIncludedServices** to scan services in the projects I've worked on so far, and iOS native **discoverIncludedServices** can be called separately when needed;
It may be that I misunderstood, or it may be a problem with OrangePi's device Bluetooth. 

Anyway, thank you for such a great project.

chipweinberger, 2025-04-12:
you'll need to learn more about bluetooth to understand included services

chipweinberger, 2025-04-21:
you should also report this issue to Apple

chipweinberger, 2025-05-22:
i've reported the issue 

https://feedbackassistant.apple.com/feedback/17673241