chipweinberger/flutter_blue_plus #1163: [Help]: unrecognized selector sent to instance

Nicoeevee, 2025-04-01:
### Requirements

- [x] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.35.3

### Flutter Version

Flutter 3.29.2

### What OS?

iOS

### OS Version

iOS 18.4 macOS 15.3.2 iOS 17.5.1 

### Bluetooth Module

IDK

### What is your problem?

Cant write any data on iOS or macOS

### Logs

```shell
[+2408 ms] flutter: [FBP] \^[[1;30m<connect>\^[[0m args: \^[[1;35m{remote_id: 7963C65C-006D-3201-1700-EACF06137407, auto_connect: <…>
[   +3 ms] flutter: [FBP] \^[[1;30m(connect)\^[[0m result: \^[[1;33mtr<…>
[+1087 ms] flutter: [FBP] \^[[1;30m[[ OnConnectionStateChanged ]]\^[[0m result: \^[[1;33m{remote_id: 7963C65C-006D-3201-1700-EACF06137407, connection_state: 1, disconnect_reason_code: null, disconnect_reason_string: nul<…>
[        ] flutter: [FBP] \^[[1;30m<discoverServices>\^[[0m args: \^[[1;35m7963C65C-006D-3201-1700-EACF061374<…>
[        ] flutter: [FBP] \^[[1;30m(discoverServices)\^[[0m result: \^[[1;33mtr<…>
[  +11 ms] flutter: [FBP] \^[[1;30m[[ OnMtuChanged ]]\^[[0m result: \^[[1;33m{remote_id: 7963C65C-006D-3201-1700-EACF06137407, success: 1, error_string: success, error_code: 0, mtu: 2<…>
[ +136 ms] flutter: [FBP] \^[[1;30m[[ OnMtuChanged ]]\^[[0m result: \^[[1;33m{remote_id: 7963C65C-006D-3201-1700-EACF06137407, success: 1, error_string: success, error_code: 0, mtu: 51<…>
[+1065 ms] flutter: [FBP] \^[[1;30m[[ OnDiscoveredServices ]]\^[[0m result: \^[[1;33m{
[        ]   "remote_id": "7963C65C-006D-3201-1700-EACF06137407",
[        ]   "success": 1,
[        ]   "services": [
[        ]     {
[        ]       "service_uuid": "180a",
[        ]       "remote_id": "7963C65C-006D-3201-1700-EACF06137407",
[        ]       "characteristics": [
[        ]         {
[        ]           "remote_id": "7963C65C-006D-3201-1700-EACF06137407",
[        ]           "descriptors": [],
[        ]           "characteristic_uuid": "2a29",
[        ]           "properties": {
[        ]             "notify": 0,
[        ]             "indicate_encryption_required": 0,
[        ]             "write": 0,
[        ]             "read": 1,
[        ]             "write_without_response": 0,
[        ]             "notify_encryption_required": 0,
[        ]             "broadcast": 0,
[        ]             "extended_properties": 0,
[        ]             "indicate": 0,
[        ]             "authenticated_signed_writes": 0
[        ]           },
[        ]           "service_uuid": "180a"
[        ]         },
[        ]         {
[        ]           "remote_id": "7963C65C-006D-3201-1700-EACF06137407",
[... 110 lines cut ...]
[        ] 14  CoreFoundation                      0x000000018ae6fcd8 CFRunLoopRunSpecific + 608
[        ] 15  GraphicsServices                    0x00000001cfd201a8 GSEventRunModal + 164
[        ] 16  UIKitCore                           0x000000018d4a890c 1741FA37-4E53-371E-8DAE-D611AAB0043D + 4237580
[        ] 17  UIKitCore                           0x000000018d55c9d0 UIApplicationMain + 340
[        ] 18  UIKitCore                           0x000000018d6d6384 1741FA37-4E53-371E-8DAE-D611AAB0043D + 6521732
[        ] 19  Runner.debug.dylib                  0x000000010447c584 $sSo21UIApplicationDelegateP5UIKitE4mainyyFZ + 120
[        ] 20  Runner.debug.dylib                  0x000000010447c4fc $s6Runner11AppDelegateC5$mainyyFZ + 44
[        ] 21  Runner.debug.dylib                  0x000000010447c600 __debug_main_executable_dylib_entry_point + 28
[        ] 22  dyld                                0x00000001ae521e4c 71846EAC-EE65-3697-BF7D-790B6A07DCDB + 249420}, null)
[  +71 ms] flutter: backtrace: #0      StandardMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:652:7)
[        ] #1      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:370:18)
[        ] <asynchronous suspension>
[        ] #2      FlutterBluePlusDarwin._invokeMethod (package:flutter_blue_plus_darwin/flutter_blue_plus_darwin.dart:285:17)
[        ] <asynchronous suspension>
[        ] #3      FlutterBluePlusDarwin.writeCharacteristic (package:flutter_blue_plus_darwin/flutter_blue_plus_darwin.dart:252:12)
[        ] <asynchronous suspension>
[        ] #4      FlutterBluePlus._invokeMethod (package:flutter_blue_plus/src/flutter_blue_plus.dart:583:14)
[        ] <asynchronous suspension>
[        ] #5      BluetoothCharacteristic.write (package:flutter_blue_plus/src/bluetooth_characteristic.dart:196:7)
[        ] <asynchronous suspension>
[        ] #6      _CharacteristicTileState.onWritePressed (package:flutter_blue_plus_example/widgets/characteristic_tile.dart:64:7)
[        ] <asynchronous suspension>
[        ] #7      _CharacteristicTileState.buildWriteButton.<anonymous closure> (package:flutter_blue_plus_example/widgets/characteristic_tile.dart:120:11)
[        ] <asynchronous suspension>
```

Nicoeevee, 2025-04-01:
maybe https://github.com/chipweinberger/flutter_blue_plus/pull/611

dev-dylan, 2025-04-17:
> [image]
> 
> args[@"value"] 的类型从 NSData 变成 FlutterStandardTypedData 了，这里没有做兼容处理。不知道 flutter 是在哪个版本修改了这块逻辑

看了下项目源码，只有在 [flutter_blue_plus_darwin 4.0.0](https://github.com/chipweinberger/flutter_blue_plus/commit/f27c9b4aaaf4ecfd6aacf4f320b011bdd34726e0) 中使用的是 NSData，在 flutter_blue_plus_darwin 3.0.0 版本中使用的是 NSString *value = args[@"value"]。

在 flutter_blue_plus 1.35.3 中引用的版本是 flutter_blue_plus_darwin ^3.0.0，理论来上不应该有问题。可能是执行 pub get 出问题导致的，建议执行 flutter pub clean 后重新 flutter pub get应该可以解决问题。

joshuadutton, 2025-04-17:
I ran into this too on both iOS and macos. This happens for me on 1.35.4 but not when I downgrade to 1.35.3.

William-Geng, 2025-04-19:
@dev-dylan  Yes can confirm, locking flutter_blue_plus_darwin to 3.0.0 would solve the issue for iOS

or if you are on 1.35.4 or using flutter_blue_plus_darwin, you need to change @dev-dylan's highlighted line to 
`FlutterStandardTypedData *typedData = args[@"value"];
  NSData    *value              = [typedData data];`

chipweinberger, 2025-04-21:
please open a PR.

it probably regressed in this commit: https://github.com/chipweinberger/flutter_blue_plus/commit/b9e1a0d6f271d2537e9e9c598dae53947b6d9f89

@ekuleshov

ekuleshov, 2025-04-21:
I'll check later tonight

ekuleshov, 2025-04-21:
PR #1186

closed 2025-04-21 by chipweinberger

chipweinberger, 2025-04-21:
fixed in 4.0.1