chipweinberger/flutter_blue_plus #1164: [Help]: turnOn does not throw early when denied

Conrad598, 2025-04-02:
### Requirements

- [x] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.35.3

### Flutter Version

3.29.2

### What OS?

Android

### OS Version

Android 12

### Bluetooth Module

N/A

### What is your problem?

The `Future` returned by [`turnOn`](https://pub.dev/documentation/flutter_blue_plus/latest/flutter_blue_plus/FlutterBluePlus/turnOn.html) does not complete immediately with an exception when user denies the request. It seems the feature described by #692 no longer works. This is reproducible in the example app. The `SnackBar` [here](https://github.com/chipweinberger/flutter_blue_plus/blob/5fc128ed6ff6b1bc038ec9f91a6d9789c2cebc53/packages/flutter_blue_plus/example/lib/screens/bluetooth_off_screen.dart#L41) only shows up after the timeout when the request is denied. 

From my quick look at the android implementation, it seems like the result of [`activityBinding.getActivity().startActivityForResult(enableBtIntent, enableBluetoothRequestCode)`](https://github.com/chipweinberger/flutter_blue_plus/blob/5fc128ed6ff6b1bc038ec9f91a6d9789c2cebc53/packages/flutter_blue_plus_android/android/src/main/java/com/lib/flutter_blue_plus/FlutterBluePlusPlugin.java#L444C24-L444C88) is being ignored. The method [`OnTurnOnResponse`](https://github.com/chipweinberger/flutter_blue_plus/blob/5fc128ed6ff6b1bc038ec9f91a6d9789c2cebc53/packages/flutter_blue_plus_android/android/src/main/java/com/lib/flutter_blue_plus/FlutterBluePlusPlugin.java#L1524) which is meant to handle the result does appear not exist.

### Logs

```shell
Launching lib\main.dart on SM G981U in debug mode...
Running Gradle task 'assembleDebug'...                             19.2s
√ Built build\app\outputs\flutter-apk\app-debug.apk
Installing build\app\outputs\flutter-apk\app-debug.apk...           3.5s
I/flutter (17978): [IMPORTANT:flutter/shell/platform/android/android_context_vk_impeller.cc(60)] Using the Impeller rendering backend (Vulkan).
I/flutter (17978): [FBP] <setLogLevel> args: 5
I/flutter (17978): [FBP] (setLogLevel) result: true
I/flutter (17978): [FBP] <getAdapterState> args: null
I/flutter (17978): [FBP] (getAdapterState) result: {adapter_state: 6}
Syncing files to device SM G981U...                                338ms

Flutter run key commands.
r Hot reload.
R Hot restart.
h List all available interactive commands.
d Detach (terminate "flutter run" but leave application running).
c Clear the screen
q Quit (terminate the application on the device).

A Dart VM Service on SM G981U is available at: http://127.0.0.1:52622/VJeGDZqBbtU=/
The Flutter DevTools debugger and profiler on SM G981U is available at: http://127.0.0.1:9102?uri=http://127.0.0.1:52622/VJeGDZqBbtU=/
I/flutter (17978): [FBP] <turnOn> args: null
D/[FBP-Android](17978): [FBP] onMethodCall: turnOn
I/flutter (17978): [FBP] (turnOn) result: true
I/ViewRootImpl@4b41e5e[MainActivity](17978): MSG_WINDOW_FOCUS_CHANGED 0 1
I/ViewRootImpl@4b41e5e[MainActivity](17978): stopped(false) old=false
I/DecorView(17978): notifyKeepScreenOnChanged: keepScreenOn=false
I/flutter (17978): [FBP] [[ OnTurnOnResponse ]] result: {user_accepted: false}
I/ViewRootImpl@4b41e5e[MainActivity](17978): Relayout returned: old=(0,0,1080,2400) new=(0,0,1080,2400) req=(1080,2400)0 dur=5 res=0x1 s={true -5476376650277282080} ch=false fn=2
I/ViewRootImpl@4b41e5e[MainActivity](17978): updateBoundsLayer: t = android.view.SurfaceControl$Transaction@68f26b3 sc = Surface(name=Bounds for - com.lib.flutter_blue_plus_example/com.lib.flutter_blue_plus_example.MainActivity@0)/@0xa303470 frame = 2
I/ViewRootImpl@4b41e5e[MainActivity](17978): mWNT: t = android.view.SurfaceControl$Transaction@68f26b3 fN = 2 android.view.ViewRootImpl.prepareSurfaces:2752 android.view.ViewRootImpl.performTraversals:3998 android.view.ViewRootImpl.doTraversal:2893
I/ViewRootImpl@4b41e5e[MainActivity](17978): mWNT: merge t to BBQ
I/ViewRootImpl@4b41e5e[MainActivity](17978): MSG_WINDOW_FOCUS_CHANGED 1 1
D/InputMethodManager(17978): startInputInner - Id : 0
I/InputMethodManager(17978): startInputInner - mService.startInputOrWindowGainedFocus
D/ProfileInstaller(17978): Installing profile for com.lib.flutter_blue_plus_example
I/ue_plus_exampl(17978): Thread[2,tid=17980,WaitingInMainSignalCatcherLoop,Thread*=0xb40000738ff35820,peer=0x14080560,"Signal Catcher"]: reacting to signal 10
I/ue_plus_exampl(17978):
I/ue_plus_exampl(17978): SIGUSR1 forcing GC (no HPROF) and profile save
I/ue_plus_exampl(17978): Explicit concurrent copying GC freed 2266KB AllocSpace bytes, 3(60KB) LOS objects, 68% free, 2774KB/8918KB, paused 112us,28us total 15.247ms
D/InputTransport(17978): Input channel destroyed: 'ClientS', fd=142
D/[secipm](17978): mSecIpmManager setProfileLength com.lib.flutter_blue_plus_example profile:2623
I/flutter (17978): FlutterBluePlusException | turnOn | fbp-code: 1 | Timed out after 60s
I/flutter (17978): backtrace: #0      FutureTimeout.fbpTimeout.<anonymous closure> (package:flutter_blue_plus/src/utils.dart:18:7)
I/flutter (17978): #1      _RootZone.run (dart:async/zone.dart:1843:54)
I/flutter (17978): #2      Future.timeout.<anonymous closure> (dart:async/future_impl.dart:1036:34)
I/flutter (17978): <asynchronous suspension>
I/flutter (17978): #3      FlutterBluePlus.turnOn (package:flutter_blue_plus/src/flutter_blue_plus.dart:129:7)
I/flutter (17978): <asynchronous suspension>
I/flutter (17978): #4      BluetoothOffScreen.buildTurnOnButton.<anonymous closure> (package:flutter_blue_plus_example/screens/bluetooth_off_screen.dart:38:15)
I/flutter (17978): <asynchronous suspension>
```

chipweinberger, 2025-04-02:
please open a PR

chipweinberger, 2025-04-02:
@tnc1997 I imagine the platform interface change broke this?

PR #1166 (2025-04-04): [Refactor] Add on turn on response stream
Closes #1164
closed 2025-04-04 by chipweinberger