chipweinberger/flutter_blue_plus #1165: [Feature]: Improved Timeout API

Conrad598, 2025-04-02:
### FlutterBluePlus Version

1.35.3

### Flutter Version

N/A

### What OS?

All

### OS Version

N/A

### Bluetooth Module

N/A

### What is your feature request?

It would be great if the public API used `Duration?` for all of the methods with timeouts like [`turnOn`](https://pub.dev/documentation/flutter_blue_plus/latest/flutter_blue_plus/FlutterBluePlus/turnOn.html), [`write`](https://pub.dev/documentation/flutter_blue_plus/latest/flutter_blue_plus/BluetoothCharacteristic/write.html), [`discoverServices`](https://pub.dev/documentation/flutter_blue_plus/latest/flutter_blue_plus/BluetoothDevice/discoverServices.html), etc. It is nullable to support no timeout. I believe this is the idiomatic way of handling timeouts in dart.

This is generally how timeouts work in official dart packages. For example [Socket.connect](https://api.dart.dev/dart-io/Socket/connect.html).

If you want sensible default timeouts while supporting nullable, that could be achieved with:

```dart
Future<void> turnOn({Duration? timeout = const Duration(seconds: 30)}) {
}
```

### Logs

```shell
N/A
```

chipweinberger, 2025-04-02:
yes i've considered this too, but its a breaking change. It's not planned.

closed 2025-04-02 by chipweinberger

shingohu, 2025-05-09:
@chipweinberger This feature improvement is very important, especially in controlling write timeouts. Currently, it can only be controlled up to seconds, but we need millimeter level control For example, we use BLE to transmit voice data every 20ms. When the BLE device is kept at a possible distance from the phone (about 10 meters), it may cause some data transmission to timeout and fail. If it fails, the voice data packet can be directly discarded. However, since the current minimum timeout time can only be set to 1 second, it means that subsequent data needs to wait for 1 second before transmission, resulting in a delay. If the timeout time can be set to 20ms, even if it fails, there will be no greater delay Please reconsider the improvement of this feature