chipweinberger/flutter_blue_plus #1172: [Feature]: Create Release for Bug Fixes

Conrad598, 2025-04-10:
### FlutterBluePlus Version

1.35.3

### Flutter Version

N/A

### What OS?

All

### OS Version

N/A

### Bluetooth Module

N/A

### What is your feature request?

I would like to kindly request a new release of `flutter_blue_plus`. There have been a bunch of great fixes since the last release which users would benefit from. (#1166, #1153, and others)

In the meantime, I'm using `flutter_blue_plus` as a [git package](https://dart.dev/tools/pub/dependencies#git-packages), but I couldn't get it to work without [`dependency_overrides`](https://dart.dev/tools/pub/dependencies#dependency-overrides) (This probably has the potential to break something).
```yaml
dependency_overrides:
  flutter_blue_plus:
    git:
      url: https://github.com/chipweinberger/flutter_blue_plus
      path: packages/flutter_blue_plus
      ref: 1fc104716649b1463fad91f32bc1a48d8fcaf4c1
  flutter_blue_plus_android:
    git:
      url: https://github.com/chipweinberger/flutter_blue_plus
      path: packages/flutter_blue_plus_android
      ref: 1fc104716649b1463fad91f32bc1a48d8fcaf4c1
  flutter_blue_plus_platform_interface:
    git:
      url: https://github.com/chipweinberger/flutter_blue_plus
      path: packages/flutter_blue_plus_platform_interface
      ref: 1fc104716649b1463fad91f32bc1a48d8fcaf4c1
  flutter_blue_plus_web:
    git:
      url: https://github.com/chipweinberger/flutter_blue_plus
      path: packages/flutter_blue_plus_web
      ref: 1fc104716649b1463fad91f32bc1a48d8fcaf4c1
```

### Logs

```shell
N/A
```

chipweinberger, 2025-04-11:
@tnc1997 wanna handle this?

tnc1997, 2025-04-17:
https://pub.dev/packages/flutter_blue_plus/versions/1.35.4