chipweinberger/flutter_blue_plus #1176: [Help]: Why does iOS pop up 'Allow finding devices on local network'? I found that other similar apps don't pop up this, but they can still scan and connect devices

topjx, 2025-04-17:
### Requirements

- [x] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

1.31.0

### Flutter Version

3.24.5

### What OS?

iOS

### OS Version

18.3.1

### Bluetooth Module

i do not know

### What is your problem?

Why does iOS pop up 'Allow finding devices on local network'? I found that other similar apps don't pop up this, but they can still scan and connect devices

<dict>
    <key>NSBluetoothAlwaysUsageDescription</key>
    <string>This app needs Bluetooth to function</string>

This is the only one in the info.plist

Ways to apply for permission
Permission.bluetooth.request()

### Logs

```shell
not have
```

topjx, 2025-04-17:
After release, it may not display

chipweinberger, 2025-04-21:
this only happens in DEBUG builds, in order to enable debugging.

closed 2025-04-21 by chipweinberger