chipweinberger/flutter_blue_plus #1177: [Help]: URGENT: Exception when building for android after upgrading to 1.35.4

CordMemescape, 2025-04-17:
### Requirements

- [x] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

1.35.4

### Flutter Version

Flutter (Channel stable, 3.29.1, on macOS 15.4.1 24E263 darwin-arm64, locale en-GB)

### What OS?

All

### OS Version

Xcode 16.3  /  Android SDK version 34.0.0

### Bluetooth Module

Espressif ESP-32

### What is your problem?

After upgrading to FlutterBluePlus 1.35.4 from 1.35.3, when building for android, I receive the error shown in the logs. Building for macOS and iOS works fine.

Looking at the issue in Android Studio, commenting out lines 2610-2615 in `FlutterBluePlusPlkugin.java` seems to resolve the first issue.

```
        // Manufacturer Specific Data
        HashMap<Integer, byte[]> manufDataB = new HashMap<>();
        if (manufData != null) {
            for (Map.Entry<Integer, byte[]> entry : manufData.entrySet()) {
                manufDataB.put(entry.getKey(), entry.getValue());
            }
        }
```

Renaming variable `manufData` to `manufDataB` on line 2659 seems to resolve the second issue.

I still need to test the resulting build to confirm the above but suggest this as a starting point.

### Logs

```shell
Running flutter build appbundle

/Users/<redacted>/.pub-cache/hosted/pub.dev/flutter_blue_plus_android-4.0.0/android/src/main/java/com/lib/flutter_blue_plus/FlutterBluePlusPlugin.java:1614: warning: [deprecation] LOCATION_MODE in Secure has been deprecated
                Settings.Secure.LOCATION_MODE, Settings.Secure.LOCATION_MODE_OFF);
                               ^
/Users/<redacted>/.pub-cache/hosted/pub.dev/flutter_blue_plus_android-4.0.0/android/src/main/java/com/lib/flutter_blue_plus/FlutterBluePlusPlugin.java:2611: error: cannot find symbol
        if (manufData != null) {
            ^
  symbol:   variable manufData
  location: class FlutterBluePlusPlugin
/Users/<redacted>/.pub-cache/hosted/pub.dev/flutter_blue_plus_android-4.0.0/android/src/main/java/com/lib/flutter_blue_plus/FlutterBluePlusPlugin.java:2612: error: cannot find symbol
            for (Map.Entry<Integer, byte[]> entry : manufData.entrySet()) {
                                                    ^
  symbol:   variable manufData
  location: class FlutterBluePlusPlugin
/Users/<redacted>/.pub-cache/hosted/pub.dev/flutter_blue_plus_android-4.0.0/android/src/main/java/com/lib/flutter_blue_plus/FlutterBluePlusPlugin.java:2618: error: variable manufDataB is already defined in method bmScanAdvertisement(BluetoothDevice,ScanResult)
        HashMap<Integer, byte[]> manufDataB = new HashMap<>();
                                 ^
/Users/<redacted>/.pub-cache/hosted/pub.dev/flutter_blue_plus_android-4.0.0/android/src/main/java/com/lib/flutter_blue_plus/FlutterBluePlusPlugin.java:2659: error: cannot find symbol
        if (manufData != null)           {map.put("manufacturer_data", manufDataB);}
            ^
  symbol:   variable manufData
  location: class FlutterBluePlusPlugin
4 errors
1 warning

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':flutter_blue_plus_android:compileReleaseJavaWithJavac'.
> Compilation failed; see the compiler output below.
  /Users/<redacted>/.pub-cache/hosted/pub.dev/flutter_blue_plus_android-4.0.0/android/src/main/java/com/lib/flutter_blue_plus/FlutterBluePlusPlugin.java:1614: warning: [deprecation] LOCATION_MODE in Secure has been deprecated
                  Settings.Secure.LOCATION_MODE, Settings.Secure.LOCATION_MODE_OFF);
                                 ^
  /Users/<redacted>/.pub-cache/hosted/pub.dev/flutter_blue_plus_android-4.0.0/android/src/main/java/com/lib/flutter_blue_plus/FlutterBluePlusPlugin.java:2611: error: cannot find symbol
          if (manufData != null) {
              ^
    symbol:   variable manufData
    location: class FlutterBluePlusPlugin
  /Users/<redacted>/.pub-cache/hosted/pub.dev/flutter_blue_plus_android-4.0.0/android/src/main/java/com/lib/flutter_blue_plus/FlutterBluePlusPlugin.java:2612: error: cannot find symbol
              for (Map.Entry<Integer, byte[]> entry : manufData.entrySet()) {
                                                      ^
    symbol:   variable manufData
    location: class FlutterBluePlusPlugin
  /Users/<redacted>/.pub-cache/hosted/pub.dev/flutter_blue_plus_android-4.0.0/android/src/main/java/com/lib/flutter_blue_plus/FlutterBluePlusPlugin.java:2618: error: variable manufDataB is already defined in method bmScanAdvertisement(BluetoothDevice,ScanResult)
          HashMap<Integer, byte[]> manufDataB = new HashMap<>();
                                   ^
  /Users/<redacted>/.pub-cache/hosted/pub.dev/flutter_blue_plus_android-4.0.0/android/src/main/java/com/lib/flutter_blue_plus/FlutterBluePlusPlugin.java:2659: error: cannot find symbol
          if (manufData != null)           {map.put("manufacturer_data", manufDataB);}
              ^
    symbol:   variable manufData
    location: class FlutterBluePlusPlugin
  4 errors
  1 warning

* Try:
> Check your code and dependencies to fix the compilation error(s)
> Run with --scan to get full insights.

BUILD FAILED in 1s
Running Gradle task 'bundleRelease'...                           1,992ms
Gradle task bundleRelease failed with exit code 1
```

lumisoft, 2025-04-17:
Exactly the same issue, I tried to upgrade flutter to : 

Flutter 3.29.3 • channel stable • https://github.com/flutter/flutter.git
Framework • revision ea121f8859 (il y a 6 jours) • 2025-04-11 19:10:07 +0000
Engine • revision cf56914b32
Tools • Dart 3.7.2 • DevTools 2.42.3

And still the same issue.

CordMemescape, 2025-04-17:
I have tested this on two android devices and can confirm that, after implementing the suggested fixes, my app still works.

mehul-hirpara-vl, 2025-04-17:
My build automation started failing with this change.

```
/Users/mehul.hirpara/.pub-cache/hosted/pub.dev/flutter_blue_plus_android-4.0.0/android/src/main/java/com/lib/flutter_blue_plus/FlutterBluePlusPlugin.java:1614: warning: [deprecation] LOCATION_MODE in Secure has been deprecated
                Settings.Secure.LOCATION_MODE, Settings.Secure.LOCATION_MODE_OFF);
                               ^
/Users/mehul.hirpara/.pub-cache/hosted/pub.dev/flutter_blue_plus_android-4.0.0/android/src/main/java/com/lib/flutter_blue_plus/FlutterBluePlusPlugin.java:2611: error: cannot find symbol
        if (manufData != null) {
            ^
  symbol:   variable manufData
  location: class FlutterBluePlusPlugin
/Users/mehul.hirpara/.pub-cache/hosted/pub.dev/flutter_blue_plus_android-4.0.0/android/src/main/java/com/lib/flutter_blue_plus/FlutterBluePlusPlugin.java:2612: error: cannot find symbol
            for (Map.Entry<Integer, byte[]> entry : manufData.entrySet()) {
                                                    ^
  symbol:   variable manufData
  location: class FlutterBluePlusPlugin
/Users/mehul.hirpara/.pub-cache/hosted/pub.dev/flutter_blue_plus_android-4.0.0/android/src/main/java/com/lib/flutter_blue_plus/FlutterBluePlusPlugin.java:2618: error: variable manufDataB is already defined in method bmScanAdvertisement(BluetoothDevice,ScanResult)
        HashMap<Integer, byte[]> manufDataB = new HashMap<>();
                                 ^
/Users/mehul.hirpara/.pub-cache/hosted/pub.dev/flutter_blue_plus_android-4.0.0/android/src/main/java/com/lib/flutter_blue_plus/FlutterBluePlusPlugin.java:2659: error: cannot find symbol
        if (manufData != null)           {map.put("manufacturer_data", manufDataB);}
            ^
  symbol:   variable manufData
  location: class FlutterBluePlusPlugin
4 errors
1 warning

FAILURE: Build failed with an exception.
```

liaowei12138, 2025-04-17:
I just found this problem. Can rolling back the Dart SDK solve it?

mehul-hirpara-vl, 2025-04-17:
I am able to resolve errors with the following changes:

```
if (manufData != null) {
if (manufDataB != null) {
```

Removed extra declaration
```
HashMap<Integer, byte[]> manufDataB = new HashMap<>();   <--- Remove this
if (rawMsd != null && rawMsd.length >= 2) {
```

```
if (manufData != null)           {map.put("manufacturer_data", manufDataB);}
if (manufDataB != null)           {map.put("manufacturer_data", manufDataB);}
```

liaowei12138, 2025-04-17:
> I am able to resolve errors with the following changes:
>
> ```
> if (manufData != null) {
> if (manufDataB != null) {
> ```
>
> Removed extra declaration
>
> ```
> HashMap<Integer, byte[]> manufDataB = new [...]
Yes, just as I thought, it's solved. thanks!

chipweinberger, 2025-04-18:
@tnc1997 yes we need to re-release. I bugged up android.

tnc1997, 2025-04-18:
> and I am using an older version like flutter_blue_plus: ^1.31.17

That version constant syntax means that your application will use the latest 1.y.z version.

If you do not wish to get the latest changes, including small breaking changes, then you should pin to a specific version of this package.

`flutter_blue_plus: 1.31.17`

mehul-hirpara-vl, 2025-04-18:
ah! I was misguided by some documents. Thanks for the clarity, I'll make changes accordingly.

William-Geng, 2025-04-18:
The above change will allow you to build - but manufacturer data parsing seems to be broken though

MrCsabaToth, 2025-04-19:
> The above change will allow you to build - but manufacturer data parsing seems to be broken though this was the manufacturer data for the same device before update `{1047: [2, 21, 105, 179, 6, 240, 185, 224, 17, 235, 133, 41, 2, 66, 172, 19, 0, 3, 0, 99, 0, 2, 192, 14, 120, 93, 19, 176, 161, 171, 79, 220, 177, 4, 93, 238, 111, 106, 149, 141, 54, 94]}`, this is after update: `{1047: [2, 21, 105, 179, 6, 240, 185, 224, 17, 235, 133, 41, 2, 66, 172, 19, 0, 3, 0, 99, 0, 2, 192, 23, 4, 14, 120, 93, 19, 176, 161, 171, 79, 220, 177, 4, 93, 238, 111, 106, 149, 141, 54, 94]}`,
> 
> extra two byte [23, 4] got inserted, after ibeacon tx power, perhaps a header that wasn't there before?

Is that manufacturer data appended together from multiple MSDs? There were changes around that I think.

Darshan-Dhanke, 2025-04-19:
@chipweinberger, Any ETA on a fix for the manufData compilation issue in flutter_blue_plus_android v4.0.0? Just checking if a patch is coming soon or if we should stick to any older version for now??

ekuleshov, 2025-04-19:
@tnc1997 @chipweinberger perhaps we should update the readme about the direct use of federated plugins...

tnc1997, 2025-04-19:
> perhaps we should update the readme about the direct use of federated plugins

#1182

Darshan-Dhanke, 2025-04-20:
@tnc1997 @ekuleshov-idexx Made the recommended changes from the README, including setting minSdkVersion to 21, but the error still persists. Any ideas?

Maikol130, 2025-04-20:
For me worked adding the flutter_blue_plus_android to the pubspec.yaml to the version 3.0.0:

  flutter_blue_plus: ^1.34.5
  flutter_blue_plus_android: ^3.0.0

with this we force flutter to use the previous version in the meantime is solved

Darshan-Dhanke, 2025-04-20:
@Maikol130, how could you add dependent packages.. Getting the following error on flutter pub get

Resolving dependencies... 
Because flutter_blue_plus_android ^3.0.0 depends on flutter_blue_plus_platform_interface ^3.0.0 and flutter_blue_plus 1.35.4 depends on
  flutter_blue_plus_platform_interface ^4.0.0, flutter_blue_plus_android ^3.0.0 is incompatible with flutter_blue_plus 1.35.4.
So, because <flutter-project> depends on both flutter_blue_plus 1.35.4 and flutter_blue_plus_android 3.0.0, version solving failed.

You can try one of the following suggestions to make the pubspec resolve:
* Try upgrading your constraint on flutter_blue_plus_android: flutter pub add flutter_blue_plus_android:^4.0.0
* Consider downgrading your constraint on flutter_blue_plus: flutter pub add flutter_blue_plus:^1.35.3
Failed to update packages.

MrCsabaToth, 2025-04-20:
> [@Maikol130](https://github.com/Maikol130), how could you add dependent packages.. Getting the following error on flutter pub get
> 
> Resolving dependencies... Because flutter_blue_plus_android ^3.0.0 depends on flutter_blue_plus_platform_interface ^3.0.0 and flutter_blue_plus 1.35.4 depends on flutter_blue_plus_platform_interface ^4.0.0, flutter_blue_plus_android ^3.0.0 is incompatible with flutter_blue_plus 1.35.4. So, because depends on both flutter_blue_plus 1.35.4 and flutter_blue_plus_android 3.0.0, version solving failed.
> 
> You can try one of the following suggestions to make the pubspec resolve:
> 
>     * Try upgrading your constraint on flutter_blue_plus_android: flutter pub add flutter_blue_plus_android:^4.0.0
> 
>     * Consider downgrading your constraint on flutter_blue_plus: flutter pub add flutter_blue_plus:^1.35.3
>       Failed to update packages.

Try to override the version in the yaml `dependency_overrides:` section.

chipweinberger, 2025-04-21:
fixed in `flutter_blue_plus_android: 4.0.1`. 

update your packages again

closed 2025-04-21 by chipweinberger

ekuleshov, 2025-04-21:
> fixed in `flutter_blue_plus_android: 4.0.1`.
> 
> update your packages again

@chipweinberger could you please bump the endorsed android plugin version in the federated root `flutter_blue_plus` plugin, and release `flutter_blue_plus` with the version bump

ArchangelXu, 2025-04-21:
> fixed in `flutter_blue_plus_android: 4.0.1`.
> 
> update your packages again

* What went wrong:
Execution failed for task ':flutter_blue_plus_android:compileDebugJavaWithJavac'.
> Compilation failed; see the compiler output below.
  D:\Work\pub_cache\hosted\pub.dev\flutter_blue_plus_android-4.0.1\android\src\main\java\com\lib\flutter_blue_plus\FlutterBluePlusPlugin.java:1614: warning: [deprecation] Secure:LOCATION_MODE deprecated
                  Settings.Secure.LOCATION_MODE, Settings.Secure.LOCATION_MODE_OFF);
                                 ^
  D:\Work\pub_cache\hosted\pub.dev\flutter_blue_plus_android-4.0.1\android\src\main\java\com\lib\flutter_blue_plus\FlutterBluePlusPlugin.java:2651: error: symbol not found
          if (manufData != null)           {map.put("manufacturer_data", manufDataB);}
              ^
    symbol:   var manufData
    location: class FlutterBluePlusPlugin
  1 个warning
  1 个error

not completely fixed

tnc1997, 2025-04-21:
> could you please bump the endorsed android plugin version in the federated root `flutter_blue_plus`

Just a heads up for reference that the latest 4.y.z of `flutter_blue_plus_android` should be installed when [installing](https://docs.flutter.dev/packages-and-plugins/using-packages#adding-a-package-dependency-to-an-app-using-flutter-pub-add) or [upgrading](https://docs.flutter.dev/release/upgrade#upgrading-packages) `flutter_blue_plus` because we use semantic versioning for the [implementations](https://github.com/chipweinberger/flutter_blue_plus/blob/master/packages/flutter_blue_plus/README.md#versioning).

> When you add a dependency to your package, you'll sometimes want to specify a range of versions to allow. How do you know what range to pick? You need to be forward compatible, so ideally the range encompasses future versions that haven't been released yet. But how do you know your package is going to work with some new version that doesn't even exist yet?
> 
> To solve that, you need to agree on what a version number means. Imagine that the developers of a package you depend on say, "If we make any backwards incompatible change, then we promise to increment the major version number." If you trust them, then if you know your package works with 2.3.5 of theirs, you can rely on it working all the way up to 3.0.0.
> 
> To make this work, then, we need to come up with that set of promises. Fortunately, other smart people have done the work of figuring this all out and named it semantic versioning.

https://dart.dev/tools/pub/versioning#semantic-versions

ekuleshov, 2025-04-22:
> The same for me, only working combination for me was after downgrading to the following combination:
> ` flutter_blue_plus: ^1.34.5`

Leave only this ^^^ dependency. Remove any dependency overrides.
Then run `flutter pub upgrade`

tnc1997, 2025-04-22:
@lumisoft are you able to reproduce the issue with `flutter_blue_plus_android` version 4.0.2 as I believe that there was an additional patch made after version 4.0.1 was published?

```
PS C:\Users\Thomas.Clark\Documents\hello_world> flutter pub upgrade
Resolving dependencies...
Downloading packages...
> flutter_blue_plus_android 4.0.2 (was 4.0.1)
Changed 1 dependency!
```

lumisoft, 2025-04-22:
Thanks to your advice of `flutter pub upgrade` , I was was able to use latest version. I removed totally my .pub-cache/ and tried again `flutter_blue_plus: ^1.35.4` which is using flutter_blue_plus_android 4.0.2 and it is working too now.